jgromes/RadioLib #524: [SX1276] Request to add setCrcFiltering(bool crcOn = true)

matthias-bs, 2022-05-22:
For my application, I have to disable CRC filtering in FSK packet mode.

As a workaround, I changed SX127x.cpp as follows:
In `SX127x::configFSK()`:
`state = _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_PACKET_CONFIG_1,[...] | RADIOLIB_SX127X_CRC_ON | [...], 7, 0);`
replaced by
`state = _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_PACKET_CONFIG_1,[...] | RADIOLIB_SX127X_CRC_OFF | [...], 7, 0);`

The CC1101 module has a function `setCrcFiltering(bool crcOn)` for this purpose. It would be great if the SX127x had the same function.

matthias-bs, 2022-05-23:
Added proposed solution:
Pull request [[SX1276] Request to add setCrcFiltering()](https://github.com/jgromes/RadioLib/pull/525)

matthias-bs, 2022-05-23:
Added proposed solution:
Pull request [[SX127x] added setCrcFiltering()](https://github.com/jgromes/RadioLib/pull/527)
(This time changed from a fresh fork. =:-) )

jgromes, 2022-05-24:
Resolved by merging #527.

closed 2022-05-24 by jgromes

matthias-bs, 2022-05-24:
Thanks a lot!