jgromes/RadioLib #529: [MORSE] Use out of band tone to make a space.

n9wxu, 2022-05-26:
The current morse implementation puts the radio in standby to create a space.  This stops the carrier and creates a "static" burst in a typical FM receiver.  This can be easily tested with the 127x and a Baofeng handheld.

Instead of using standby, set the tone to a frequency that is out of band for the audio filters.  This will keep the carrier live and the receiver quiet.

I tested with tone(1) and it is much more pleasant to hear than the original configuration.

jgromes, 2022-05-26:
This is possible, however, introduces a couple of issues. For example: "out of band for the audio filters" can mean different things on different radios. If the tone is not moved enough, it won't make the space. If it's moved too far, you could be transmitting in a band where you're not supposed to.

n9wxu, 2022-05-26:
On the SX1276 using DIO2 to modulate the FSK, if you modulate at 1HZ instead of 1KHZ, then you will create an audio tone that is outside of the audio bandwidth of the receiver which will produce silence.  I don't think you will create RF frequencies outside of the frequency deviation set during initialization.

jgromes, 2022-05-27:
Ah, I see - you're using the AFSK Morse - RadioLib can also transmit Morse in FSK mode, which must then be demodulated by SSB recevier (not FM) to be audible.

I'll have to investigate the impact of this suggestion to the AFSK, FSK and OOK (demodulated as AM) modes, before implementing.

n9wxu, 2022-05-27:
Exactly.  I used the FSK mode but I need to make a rocket tracking transmitter that uses standard FM equipment for the ground station.  The AFSK mode works very well for this but the static bursts in the spaces were making it unpleasant to hunt the signal.

jgromes, 2022-05-28:
I tested this with all three Morse options (FSK, AFSK and OOK). This indeed really improves performance in AFSK mode. There also seems to be no impact in FSK or OOK modes, so I implmeneted this - thanks for the suggestion!

I also tried playing around a little bit with the other modes, and found out that when a similar thing is done in Hellschreiber, it will actually create a text with inverted color, so I added that as well. There's probably no real use for it, I just find that really funny.

closed 2022-05-28 by jgromes