jgromes/RadioLib #533: With LLCC68,when use radio.begin() get error -707?

bjfhskj, 2022-06-03:
MCU:esp32  , Wireless module:LLCC68 ; Arduino IDE version :1.8.15
when radio.begin() get error -707
and radio.begin(868.0, 250.0, 7, 5, 0x34, 22, 10, 0, false); when send mesage get erro timeout.

`
// include the library
#include <RadioLib.h>

#define ESP32
#define TX_EN
//#define RX_EN

#ifdef ESP32
#pragma config board="ESP32 Wrover Module"
#pragma config board="ESP32 Wrover Module"
//#pragma config port="4"
#endif

#ifdef TX_EN
void tx_routine (void);
void init_tx    (void);
#endif

#ifdef RX_EN
void rx_routine (void);
void init_rx    (void);
// flag to indicate that a packet was received
volatile bool receivedFlag = false;
// disable interrupt when it's not needed
volatile bool enableInterrupt = true;
void setFlag(void);
#endif

#ifdef ESP32
#pragma config board="ESP32 Wrover Module"
#define NSS_PIN  5
#define DIO1_PIN 21
#define DIO2_PIN 39 // not connected
#define BUSY_PIN 22
#define SCLK_PIN 18
#define MISO_PIN 19
#define MOSI_PIN 23
// #define NSS_PIN   5
#define RST_PIN  4
// GPIOs to control E22 RX_EN and TX_EN
#define RADIO_RXEN -1
#define RADIO_TXEN -1
#define FREQ   868
#endif

// SX1262 has the following connections:
SX1262 radio = new Module(NSS_PIN, DIO1_PIN, DIO2_PIN, BUSY_PIN);

#ifdef TX_EN
void init_tx(void) {
  if (radio.setFrequency(868) == RADIOLIB_ERR_INVALID_FREQUENCY) {
    Serial.println(F("Selected frequency is invalid for this module!"));
    while (true);
  }
  // set bandwidth to 250 kHz
  if (radio.setBandwidth(250.0) == RADIOLIB_ERR_INVALID_BANDWIDTH) {
    Serial.println(F("Selected bandwidth is invalid for this module!"));
    while (true);
  }
  // set spreading factor to 10
  if (radio.setSpreadingFactor(10) == RADIOLIB_ERR_INVALID_SPREADING_FACTOR) {
    Serial.println(F("Selected spreading factor is invalid for this module!"));
    while (true);
  }
  // set coding rate to 6
  if (radio.setCodingRate(6) == RADIOLIB_ERR_INVALID_CODING_RATE) {
    Serial.println(F("Selected coding rate is invalid for this module!"));
    while (true);
  }
  // set LoRa sync word to 0xAB
  if (radio.setSyncWord(0x34) != RADIOLIB_ERR_NONE) {
    Serial.println(F("Unable to set sync word!"));
    while (true);
  }
  // set output power to 10 dBm (accepted range is -17 - 22 dBm)
  if (radio.setOutputPower(22) == RADIOLIB_ERR_INVALID_OUTPUT_POWER) {
    Serial.println(F("Selected output power is invalid for this module!"));
    while (true);
  }
  // set over current protection limit to 80 mA (accepted range is 45 - 240 mA)
  // NOTE: set value to 0 to disable overcurrent protection
  if (radio.setCurrentLimit(80) == RADIOLIB_ERR_INVALID_CURRENT_LIMIT) {
    Serial.println(F("Selected current limit is invalid for this module!"));
    while (true);
  }
  // set LoRa preamble length to 15 symbols (accepted range is 0 - 65535)
  if (radio.setPreambleLength(15) == RADIOLIB_ERR_INVALID_PREAMBLE_LENGTH) {
    Serial.println(F("Selected preamble length is invalid for this module!"));
    while (true);
  }
  // disable CRC
  if (radio.setCRC(false) == RADIOLIB_ERR_INVALID_CRC_CONFIGURATION) {
    Serial.println(F("Selected CRC is invalid for this module!"));
    while (true);
  }
  // Some SX126x modules have TCXO (temperature compensated crystal
  // oscillator). To configure TCXO reference voltage,
  // the following method can be used.
  if (radio.setTCXO(3.3) == RADIOLIB_ERR_INVALID_TCXO_VOLTAGE) {
    Serial.println(F("Selected TCXO voltage is invalid for this module!"));
    while (true);
  }
  // Some SX126x modules use DIO2 as RF switch. To enable
  // this feature, the following method can be used.
  // NOTE: As long as DIO2 is configured to control RF switch,
  //       it can't be used as interrupt pin!
  if (radio.setDio2AsRfSwitch() != RADIOLIB_ERR_NONE) {
    Serial.println(F("Failed to set DIO2 as RF switch!"));
    while (true);
  }
  Serial.println(F("All settings succesfully changed!"));
}
#endif

#ifdef RX_EN
void init_rx(void) {
  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);

  init_tx();

  // start listening for LoRa packets
  Serial.print(F("[SX1262] Starting to listen ... "));
  int state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

// this function is called when a complete packet
// is received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void) {
  // check if the interrupt is enabled
  if (!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}
#endif

void setup() {
  Serial.begin(115200);

#ifdef ESP32
  SPI.begin(SCLK_PIN, MISO_PIN, MOSI_PIN, NSS_PIN);
#endif
  pinMode(RST_PIN, OUTPUT);
  digitalWrite(RST_PIN, LOW);
  delay(100);
  digitalWrite(RST_PIN, HIGH);
  // initialize SX1268 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
// int state = radio.begin();
   int state = radio.begin(868.0, 250.0, 7, 5, 0x34, 22, 10, 0, false);
  //
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  Serial.print(F("[SX1268] Initializing ... "));
  // carrier frequency:           915.0 MHz
  // bandwidth:                   500.0 kHz
  // spreading factor:            6
  // coding rate:                 5
  // sync word:                   0x34 (public network/LoRaWAN)
  // output power:                2 dBm
  // preamble length:             20 symbols
  // you can also change the settings at runtime
  // and check if the configuration was changed successfully
  // set carrier frequency to 433.5 MHz
  

#ifdef TX_EN
  init_tx();
#endif

#ifdef RX_EN
  init_rx();
#endif
}

#ifdef RX_EN
void rx_routine (void) {
  if (receivedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = radio.readData(str);

    // you can also read received data as byte array
    /*
      byte byteArr[8];
      int state = radio.readData(byteArr, 8);
    */

    if (state == RADIOLIB_ERR_NONE) {
      // packet was successfully received
      Serial.println(F("[SX1262] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1262] Data:\t\t"));
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1262] RSSI:\t\t"));
      String RSSI_str = String(radio.getRSSI());
      Serial.print(RSSI_str);
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1262] SNR:\t\t"));
      String SNR_str = String(radio.getSNR());
      Serial.print(SNR_str);
      Serial.println(F(" dB"));

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }

    // put module back to listen mode
    radio.startReceive();

    // we're ready to receive more packets,
    // enable interrupt service routine
    enableInterrupt = true;
  }
}
#endif

#ifdef TX_EN
int num = 0;
char cstr[16];

void tx_routine (void) {
  Serial.print(F("[SX1262] Transmitting packet ... "));
  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  num = num + 1;
  itoa(num, cstr, 10);
  int state = radio.transmit(cstr);
  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */
  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));
    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));
  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));
  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));
  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }
  // wait for a second before transmitting again
  delay(1000);
  if (num == 999) {
    num = 0;
  }

}
#endif
//
void loop() {
#ifdef TX_EN
  tx_routine();
#endif
#ifdef RX_EN
  rx_routine();
#endif
}`

jgromes, 2022-06-03:
First of all, you have not provided the information that's called for in the issue template - mainly the [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode) output.

Next, I noticed that you're calling SPI.begin. I'm not sure why you're doing that, please stick to the example code for the first startup.

Also, you're setting configuration twice for some reason.

TL;DR you seem to be making things unnecessarily complicated on yourself. Start with the examples, and only add functionality you really need.

bjfhskj, 2022-06-04:
1. I use SPI.begin for set the pin for ESP32,
2.  The first time ,I use radio.begin, the setting configuration only once,but get error -707,so I modfy it radio.begin(868.0, 250.0, 7, 5, 0x34, 22, 10, 0, false),and this make the the setting configuration twice .
3. Debug message:
[SX1262] Initializing ... success!
[SX1268] Initializing ... All settings succesfully changed!
[SX1262] Transmitting packet ... timeout!
[SX1262] Transmitting packet ... timeout!
[SX1262] Transmitting packet ... timeout!
[SX1262] Transmitting packet ... timeout!

jgromes, 2022-06-05:
1. In that case, you have to provide a reference of your chosen SPI instance (HSPI or VSPI) to the `Module` constructor, as shown in couple other issues (e.g. this one: #422). Otherwise, RadioLib has no way to figure out that you already called `SPI.begin()` and will do so too, potentially changing something.

2. What I meant is why are you calling `begin(868.0, 250.0, 7, 5, 0x34, 22, 10, 0, false)`, and then configuring the same parameters again. It doesn't change anything, it's just unnecessary.

3. That's jus the standard output. Follow the link I posted, enable the debug, recompile and reupload the sketch and then post the output.

bjfhskj, 2022-06-18:
I follow the issues,it work.

closed 2022-06-18 by bjfhskj