jgromes/RadioLib #535: SX127x transmit.ino is not working with ESP8266

choudharyanil244, 2022-06-11:
I am using ESP8266 12E NodeMcu module, Lora RFM95. First to check whether the Lora module is working, i tried with arduino uno r3. It works successfully. Then i Connect the Lora module as below 

Lora             |       ESP8266
------------------------------------
MISO           |      SDO / MISO
MOSI           |      SD1/MOSI
CLK              |      CLK
NSS             |      CMD/CS / GPIO11
RESET          |       SD2/GPIO9
DIO0            |       D0/GPIO16
DIO1            |       SD3/GPIO10

Please find code below for reference.

I am using Arduino IDE for uploading the code to ESP8266 NodeMCU. I am sure code is uploading i test it by uploading the blink led example code. 

I want to interface Lora module with ESP8266 using HSPI. Just for testing i am using SPI. Please help me to solve this problem and let me know how can i interface Lora with HSPI of ESP8266.

Code : 
/*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

#define FREQ 868
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(11, 16, 9, 10);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}

[image]

Getting in Serial monitor : 
ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00045c20
~ld
[SX1278] Initializing ... 

The WDT is continuously reset please help me how to solve it. And interface with HSPI...

jgromes, 2022-06-12:
Please follow the [module not working template](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md) - especially, it is important to post the [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode) output.

Regarding your comment on using HSPI, there's a page about that [on the wiki](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup).

jgromes, 2022-06-25:
Closed due to inactivity.

closed 2022-06-25 by jgromes