jgromes/RadioLib #536: SX1272 not found on Arduino DUE

EtienneLescarbeault, 2022-06-13:
Hi, 
I am currently doing an embedded system project involving few sensors a data transmission. The project consists of:
- An emitter, an Arduino DUE with SX1272 
- A receiver, an Arduino UNO with SX1272.

While initializing the module on the UNO, I get no problem. The device seems to be properly listening.
However, with the second module on the DUE, the initialization fails with error code -2, indicating that no device could be found.
The wiring on my UNO was very simple, the module fitted on top right away and the same thing went for the DUE, where both pin numbers are the same. I suspect the wiring was not an issue...

To get things started, since I am pretty new to Arduinos, I tried to run the following example which caused the failure:

**Sketch that is causing the module fail**

```c++
/*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

The exact same sketch works on the UNO. I also swapped the modules, but without success...

**Hardware setup**
For the wiring, it was pretty straightforward. I simply attached the module to the DUE, just like I did for the UNO:

[image]
[image]

**Debug mode output**
[image]

**Additional info (please complete):**
 - MCU: Arduino DUE 
 - Link to Arduino core: https://github.com/arduino/ArduinoCore-sam
 - Wireless module type: SX1272
 - Arduino IDE version: 1.8.19
 - Library version: 5.1.2

Truly yours,

E.

jgromes, 2022-06-13:
> The wiring on my UNO was very simple, the module fitted on top right away and the same thing went for the DUE, where both pin numbers are the same

Are you sure the pins (especially the SPI pins) are the same on Uno and Due? Looking at a couple different pinout diagrams on Google, I don't think that's the case. E.g. this one, which seems to suggest SPI is only available on the central 6-pin connector:

http://www.robgray.com/temp/Due-pinout.svg

> Also, I tried to build an SX1272 radio object instead of the SX1278 provided by default while creating the Module().

You should use the one that matches the hardware - if your shield has SX1272 on it, that'S the one you have to select.

EtienneLescarbeault, 2022-06-13:
Thanks for your quick response. It was indeed a wiring problem. I simply reordered my wires to fit the central connector on the DUE. 

Thanks again for your time,

E.

closed 2022-06-13 by EtienneLescarbeault