jgromes/RadioLib #544: SX1280, DIO2 as RF switch control

01Space, 2022-07-08:
Hi,
Can you add a code that uses dio2 pin to control RF switch?

thank you!

jgromes, 2022-07-08:
You mean something like this? https://github.com/jgromes/RadioLib/blob/8ae092ac8123716db58dda01af1486a87aab2964/src/modules/SX126x/SX126x.cpp#L1296-L1304

That's already implemented, and enabled by default. If your DIO2 is tied to RF switch, it will be switched automatically.

closed 2022-07-08 by jgromes

01Space, 2022-07-08:
The code of sx126x has this function, but I can't find this function in the code of sx1280

jgromes, 2022-07-08:
Ah, I though you were talking about the SX126x - sorry.

For SX128x, it seems like this functionality is not supported by the device.

01Space, 2022-07-08:
thank you! I consider adding a pin of MCU to control PA switch