jgromes/RadioLib #550: 9600 FSK... center freq or default mark/space freqs?

K4KDR, 2022-07-25:
Hello!  It has been a while since I bothered you.

I am attempting to help a Project Team that is having success with 1k2 FSK from an SX1268 module using RadioLib but we are not able to decode anything they send if the baud rate is changed to 9600.    It's my understanding that they are generating test packets with the only change being:

`int state = radio.beginFSK(434.0, 9.6);`

... instead of the default '1.2'.

When viewing the spectrum of their 1k2 packets, I see clear mark/space frequencies (perhaps not the correct term, exactly) that allow us to specify custom 2FSK tones in Direwolf to decode those 1k2 packets.

When viewing their 9k6 packet audio, I do NOT see distinct 2FSK tones, just a center peak around 6500 Hz.   So, I can't use the same trick to provide Direwolf with a pair of non-standard frequencies to use to decode the 9k6 packets.

Do you know what we need to do differently to decode these 9k6 packets?  Is there some default center freq or pair of 2FSK freqs in the audio spectrum that are used for 9k6?   Or, do we need to do something different entirely in the transmit code when moving up to 9600?

Thanks as always!

-Scott,  K4KDR

jgromes, 2022-07-25:
Hi Scott!

There should be no changes needed, the bit rate is just passed to the transceiver module.

Couple of thoughts:

1. Changing just the bit rate won't change the frequency deviation of the signal, the default is +-5 kHz. It should be fine, but it's a thing to keep in mind.

2. What does the receiving end look like? Since you're processing as audio, maybe some audio filter got in the way? Does it happen at e.g. 4k8?

K4KDR, 2022-07-25:
Thanks very much for the reply!  

I did question whether the team had provided adequate audio output bandwidth for 9k6 when creating the sample file and was assured that they did.   Since we have had success with decoding their 1k2 packets, for maximum compatibility with Amateur ground stations, I believe that with a launch date quickly approaching the decision has been made to stay with that baud rate.  

Unfortunately I don't have a 4k8 sample file and I suspect there is very little time remaining for testing before integration with the launch vehicle.   Several other downlink modes are reported to be fine; the late testing & decision about the 2FSK beacon is probably considered a 'bonus' mode at this point.

Also, in my research I think I saw a comment that payload size was limited (64 bytes?) with the 9k6 2FSK mode.  Did I understand that correctly?  I've lost that reference so I'm not sure if it was accurate.

Thanks again for the information & help!

jgromes, 2022-08-07:
Seeing as the launch unfortunately failed, I'm closing this now. @K4KDR please let the team know to reopen in case they wish to follow-up. Thanks!

closed 2022-08-07 by jgromes