jgromes/RadioLib #551: Getting -705 when begin()

GilbertLau, 2022-07-27:
I am trying to init LORA on a TTGO T-Beam ESP32 LORA.   However, I keep getting -705.   Is there something I am missing?

#define SCK           5   // GPIO5  -- SX1278's SCK
#define MISO         19   // GPIO19 -- SX1278's MISnO
#define MOSI         27   // GPIO27 -- SX1278's MOSI
#define SS           18   // GPIO18 -- SX1278's CS
#define RST          23   // GPIO14 -- SX1278's RESET
#define DI0          26   // GPIO26 -- SX1278's IRQ(Interrupt Request)
#define BUSY_PIN     32

SPIClass newSPI(VSPI);
SX1268 radio = new Module(SS, RADIO_DIO1_PIN, RST, BUSY_PIN, newSPI);

newSPI.begin(SCK, MISO, MOSI, SS);
int state = radio.begin(433.0F);

**Hardware setup**
LILYGO® TTGO T-Beam V1.1 ESP32 LORA

**Debug mode output**
M       SX126x
CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

CMD     80
DATW    0       0

jgromes, 2022-07-27:
Based on the debug output, the SPI is always returning 0x00 - that probably means there's an issue with the hardware. I would check if the pins are actually correct.

jgromes, 2022-08-11:
Closed due to inactivity.

closed 2022-08-11 by jgromes