jgromes/RadioLib #553: SSTV examples with RFM98 and ESP32-S just generates a carrier.

DystopiaNow, 2022-07-31:
**Describe the bug**
SSTV examples not working  with ESP32-S and RFM98 at 434 Mhz . Tried varying the correction factor and changing modes (Wrasse, Martin1). Only a narrow (1khz) continuous carrier is generated for a period of time at the desired frequency. Other modes (LoRa, FSK, CW) work perfectly. 

**To Reproduce**
SSTV examples  (both AFSK and non-AFSK examples generate the same issue).

Modified module.cpp to initialize SPI  custom pins: 
#if defined(RADIOLIB_BUILD_ARDUINO)
void Module::SPIbegin() {
  _spi->begin(14,12,13,15);
}
///////
Used these ESP32 pins for initializing:
SX1278 radio = new Module(15, 2, 4, 5);

**Expected behavior**
There should be some visible in waterfall and audible SSTV-like signal.

**Additional info (please complete):**
 - MCU: ESP32-S
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type RFM98 433MHZ
 - Arduino IDE version  [e.g. 1.8.13]
 - Library version [e.g. 5.0.2]

jgromes, 2022-08-02:
Couple of thoughts:

1. What's the difference between ESP32-S and ESP32? In debug mode, RadioLib will print the detected platform into console during compilation.
2. You wrote you're using RFM98, but the code you showed seems to reference SX1278. While the two are very similar, they're not identical (they have different frequency range).

Also, this:

void Module::SPIbegin() {
_spi->begin(14,12,13,15);
}

is not the correct way to initialize custom SPI pins. The correct way is described here: https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup.

jgromes, 2022-08-04:
@DystopiaNow I tried with ESP32 (actually Heltec LoRa V2 board) and managed to produce audible SSTV signal, decoded by MMSSTV as the expected test image. Playing around a bit with the correction factor, 0.995 seems to be working OK.

TL;DR I cannot reproduce the issue, check the points in my previous comment.

DystopiaNow, 2022-08-05:
@jgromes Im using HSPI pins in a custom design. ESP32-S is the "common" ESP32 series (the ones that are used in Heltec boards) I tried initializing the module as you describe in https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup defining  SPIClass spi(HSPI);  but I've got the No Radio is found error (code -2). 

Moreover, in the SSTV examples, if I try to initialize the radio as RFM98, I get the following error in the Arduino console (in short):
no matching function for call to 'RFM96::beginFSK()'

jgromes, 2022-08-05:
> defining SPIClass spi(HSPI); but I've got the No Radio is found error (code -2).

That suggests an issue in the wiring or pinout.

>  if I try to initialize the radio as RFM98, I get the following error in the Arduino console (in short):
no matching function for call to 'RFM96::beginFSK()'

Looking at the current code, class RFM96 does have a beginFSK method - which version of the library are you using? There's a reason this information is listed as "please complete" in the template.

DystopiaNow, 2022-08-05:
Library version is 5.0.2. 
Arduino IDE version 1.8.13

jgromes, 2022-08-05:
There's no version 5.0.2 - do you mean 5.2.0? In any case, I would suggest updating to 5.3.0.

DystopiaNow, 2022-08-05:
Will check that again, since I updated it using the Arduino IDE (moving to Platformio soon).

DystopiaNow, 2022-08-07:
@jgromes A mistake of mine, I'm using the 5.3.0 version of Radiolib.

This is the error code I've got when trying to compile initializing the module in the SSTV examples with RFM98 instead of SX1278:

In file included from C:\\Arduino\libraries\RadioLib\src/TypeDef.h:4:0,
                 from C:\...\Arduino\libraries\RadioLib\src/RadioLib.h:38,
                 from C:\...\AppData\Local\Temp\arduino_modified_sketch_529990\SSTV_Transmit.ino:36:
C:\...\Arduino\libraries\RadioLib\src/BuildOpt.h:182:57: note: #pragma message: RADIOLIB_PLATFORM: ESP32
     #define RADIOLIB_PLATFORM                           "ESP32"
                                                         ^
C:\...\Arduino\libraries\RadioLib\src/RadioLib.h:50:41: note: in expansion of macro 'RADIOLIB_PLATFORM'
   #pragma message "RADIOLIB_PLATFORM: " RADIOLIB_PLATFORM
                                         ^
C:\...\AppData\Local\Temp\arduino_modified_sketch_529990\SSTV_Transmit.ino: In function 'void setup()':
SSTV_Transmit:95:30: error: no matching function for call to 'RFM96::beginFSK()'
   int state = radio.beginFSK();
                              ^
In file included from C:\...\Arduino\libraries\RadioLib\src/RadioLib.h:65:0,
                 from C:\...\AppData\Local\Temp\arduino_modified_sketch_529990\SSTV_Transmit.ino:36:
C:\...\Arduino\libraries\RadioLib\src/modules/RFM9x/RFM96.h:80:13: note: candidate: int16_t RFM96::beginFSK(float, float, float, float, int8_t, uint16_t, bool)
     int16_t beginFSK(float freq, float br, float freqDev, float rxBw, int8_t power, uint16_t preambleLength, bool enableOOK);
             ^
C:\...\Arduino\libraries\RadioLib\src/modules/RFM9x/RFM96.h:80:13: note:   candidate expects 7 arguments, 0 provided
exit status 1
no matching function for call to 'RFM96::beginFSK()'

DystopiaNow, 2022-08-07:
This is the output I've got from RX-SSTV using the AFSK SSTV example and ESP32, with the module configured as SX1278 (I'm using GPIO19 from ESP32 for AFSK)
[image]

jgromes, 2022-08-09:
I was able to reproduce the exact same grey/color strips, however, only with AFSK. Non-AFSK works fine for me, in both Martin1 and Scottie1 modes.

There was actually an issue in the tone() emulation for ESP32, which caused issues when tones were being changed quickly (as is the case for SSTV). I tested Wrasse, Scottie1 and Martin1 in both AFSK and non-AFSK and everything seems to be working.

@DystopiaNow can you check with the latest fix?

DystopiaNow, 2022-08-12:
I tried the last fix and could generate the desired bars with Martin1 in AFSK (thanks a lot for that).

 I'm a bit confused with Non-AFSK...does it requiere any other pin other than SPI/DIO0/RST/DIO1 pins? I'm using HSPI from ESP32, and GPIOs 2,4,5 for DIO0, RST and DIO1. I don't know if  some issue with that pin selection could affect the generation of the SSTV signal under FSK.   But, for example, RTTY under FSK, with the same pins and module, works fine.

jgromes, 2022-08-12:
What's the output in FSK mode? And how are you receiving? Keeping in mind in FSK you can't use FM demodulation, you have to use SSB and manually tune it so that the audio frequencies are correct.

DystopiaNow, 2022-08-13:
This is the type of signal I'm getting with just FSK, tried in SSB with no luck, since  these are just long and monotonous tones changing slowly, one "tone" at a time (they last a few seconds each one)
[image]

jgromes, 2022-08-13:
Unfortunately I still can't replicate that issue - could you post the entire sketch?

DystopiaNow, 2022-08-19:
Im using RFM98, but the code wont compile if I try to use that class instead of SX1278, so I use SX1278. Compiling for "ESP32 Dev Module" with default settings, with Arduino IDE 1.8.13.

`// include the library
#include <RadioLib.h>
SPIClass spi(HSPI);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(15, 2, 4, 5,spi, spiSettings);

SSTVClient sstv(&radio);

// test "image" - actually just a single 320px line
// will be sent over and over again, to create vertical color stripes at the receiver
uint32_t line[320] = {
  // black
  0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000,
  0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000, 0x000000,

  // blue
  0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF,
  0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF, 0x0000FF,

  // green
  0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00,
  0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00, 0x00FF00,

  // cyan
  0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF,
  0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF, 0x00FFFF,

  // red
  0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000,
  0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000, 0xFF0000,

  // magenta
  0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF,
  0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF, 0xFF00FF,

  // yellow
  0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00,
  0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00, 0xFFFF00,

  // white
  0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF,
  0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF, 0xFFFFFF
};

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  spi.begin();
  int state = radio.beginFSK();
 //   int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  Serial.print(F("[SSTV] Initializing ... "));

  state = sstv.begin(434.0, Wrasse, 0.95);
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // to help tune the receiver, SSTVClient can send
  // continuous beep at the frequency corresponding to
  // 1900 Hz in upper sideband (aka USB) modulation
  // (SSTV header "leader tone")
  /*
    sstv.idle();
    while(true);
*/
}

void loop() {
  // send picture with 8 color stripes
  Serial.print(F("[SSTV] Sending test picture ... "));

  // send synchronization header first
  sstv.sendHeader();

  // send all picture lines
  for(uint16_t i = 0; i < sstv.getPictureHeight(); i++) {
    sstv.sendLine(line);
  }

  // turn off transmitter
  radio.standby();

  Serial.println(F("done!"));

  delay(30000);
}`

jgromes, 2022-08-24:
I fixed the issue with RFM9x not building (the beginFSK method aws missing the default arguments) - @DystopiaNow I would suggest trying with the RFM98 class now. That being said, I'm still unable to replicate this issue using SX1278 - I don't have RFM9x on hand to test with.

DystopiaNow, 2022-08-24:
Tried again with 2-FSK example and RFM98 class, but still, I only get something like this, with two tones changing (tried hearing with USB/LSB but it doesn't look like SSTV) 
[image]

On the other hand, I've been experimenting with Martin2/1 and Scottie1-2. But still I get distorted images (trying to transmit a BMP) on these modes. Heres an example with Martin 1 
[image]

With Wrasse, the quality seems to be good: 
[image]

jgromes, 2022-08-25:
Unfortunately I cannot replicate this issue - the FSK mode is working fine on my Heltec LoRa v2 setup. I don't have an RFM98 to test with unfortuntely.

Regarding the image quality, I'm not sure what could be causing the issues with Scottie and Martin (possibly some slight issue with timing - Wrasse has quite long pixel length, 734 us compared to the 200 - 400 us in Scottie/Martin).

Though honestly, I'm actually quite surprised it's working this well, considering it's being transmitted by a digital transciever IC.

closed 2022-08-25 by jgromes