jgromes/RadioLib #555: LLCC68&ESP8266-Receive-Error State-707

qhfCode, 2022-08-08:
- Describe the bug:

Using Ai-Thinker Ra-01SC (LLCC68) to do point-to-point lora communication, the sender uses ESP32-WROOM to send SHT31 sensor temperature and humidity data, and the receiver uses ESP8266 to receive the data, observing the serial port found that the received data did not match, and there was a -707 error.

- To Reproduce

Transmitter using RadioLib SX126x Transmit Example.
The transmitted packets are shown in the following figure:
  //read temperature and humidity
  float t = sht31.readTemperature();
  float h = sht31.readHumidity();
  char tim[32];
  sprintf(tim, "%f", t); //tim save temp
  int state = radio.transmit(tim);
  char humi[32];
  sprintf(humi, "%f", h); // humi save humi
  int state1 = radio.transmit(humi);
Example changes:
int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 22, 8, 1.6, false);

 Receiver using RadioLib SX126x Receive Example.
Example changes:
int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 22, 8, 1.6, false);

- Expected behavior

The serial port should have the correct received data and should not have a -707 error

Additional info (please complete):
[image]
[image]

jgromes, 2022-08-10:
> int state = radio.transmit(stringOne);
delay(3000);
Serial.println("next");
int state1 = radio.transmit(stringTwo);
delay(10000);
>
> Receiver using RadioLib SX126x Receive Example(ESP8266):
>
> Add delay(10000) to the end of the loop;

That doesn't seem correct. On the transmitter side, you're sending a packet, then waiting 3 seconds, then sending another one, and then waiting 10 seconds. On the receiver side, you're trying to receive in blocking mode, and then waiting 10 seconds regardless of the result (I'm guessing here, you haven't shown your code).

I would suggest to start simple, with unchanged examples (only editing the wiring to match your hardware). Confirm your hardware works first. If the error is still there, then we can solve further. Only after the basics work think about how to transfer useful data. It's probably a lot easier to measure both variables first and send the values in a single packet, rather than trying to synchronize multiple transmissions.

qhfCode, 2022-08-12:
First,Thank you for the timely response from the library owner.Following the library owner's advice to try from simple, it finally worked.

Here are some of the suggestions I made in debugging LLCC68 for others to refer to:
1.Check your wiring a few more times to make sure it is 100% correct.
2.LLCC68 does not have TCXO, so the TCXO power control setting should be set to 0. For example,  int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 22, 8, 0, false);

There are still some minor problems:
1.There are many timeouts on the receiving end;Is there a solution to this problem?（See the appendix for a demonstration of the problem）

Appendix:
1.Sender:Sample code run results[No changes except for the wiring pins (based on your own board's wiring pins) and TCXO to 0]
[image]
2.Receiving side:Sample code run results[No changes except for the wiring pins (based on your own board's wiring pins) and TCXO to 0]
[image]
3.Sender:Sending sensor data:[No changes except for the wiring pins (based on your own board's wiring pins) and TCXO to 0]
Delay (1000) is added between each packet
[image]
[image]
4.Receiving side:Sample code run results[No changes except for the wiring pins (based on your own board's wiring pins) and TCXO to 0]
[image]

jgromes, 2022-08-13:
Glad you got it working!

> 1.There are many timeouts on the receiving end;Is there a solution to this problem?

Since you are using blocking reception, timeouts are to be expected - the device listens for a while, adn if no packet is received, it returns timeout. There's an interrupt-driven version that triggers an interrupt each time a packet is received.

closed 2022-08-13 by jgromes