jgromes/RadioLib #556: LoRa module receive fake data continuously even without any transmission

fabrizioPucci, 2022-08-10:
**Expected behavior**
Receiving data only on transmission and not countinuously.

**Screenshots**
See the output (it goes on forever), it is the same with or without the explicit settings:
 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
[SX1278] Initializing ... success!
All settings successfully changed!
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			
[SX1278] RSSI:			-164.00 dBm
[SX1278] SNR:			0.00 dB
[SX1278] Frequency error:	0.00 Hz
[SX1278] Waiting for incoming transmission ... success!

**Additional info:**
 - MCU:                           [Esp32C3 ] Esp32C3 DevModule  
 - Link to Arduino core:  [??]
 - Wireless module type:[Ra 02 Sx1278 433MHz LoRa-02] with dedicated antenna 
 - Arduino IDE version:   [1.8.16]
 - Library version:            [5.2.0]
 
**HARDWARE***
Board Esp32 M5Stack  + Ra 02 Sx1278 433MHz LoRa-02 bought. on Amazon as "best choice"  to avoid fake board.
Esp32 Board :[https://shop.m5stack.com/collections/m5-controllers/products/m5stamp-c3-mate-with-pin-headers]
Lora Board :[https://www.amazon.it/gp/product/B07RD2JV7Y/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1]
Antennas:[https://www.amazon.it/gp/product/B07QWRTN3F/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1]

**Other info**
The receiver receives data even without any transmitter. 
I tested it with or without the explicit Settings (see your example)
I tried using different SyncWord
My only change to your code are:[https://github.com/jgromes/RadioLib/blob/master/examples/SX127x/SX127x_Receive/SX127x_Receive.ino]

- explicit SPI configuration SPI.begin(SCK, MISO, MOSI, SS);    (it works fine)
- SX1278 radio = new Module (5,9,8);                                       (no DIO1 on my LoRa module)

fabrizioPucci, 2022-08-10:
Also using byteArray gives me the same result:

byte byteArr[8];
memset(byteArr, 0, 8);
int state = radio.receive(byteArr, 8);
String str=	String((char*)byteArr);

fabrizioPucci, 2022-08-10:
I have tested also with an older board RA 01 with the same Sx1278 and a soldered spring antenna with the same behaviour.

jgromes, 2022-08-10:
Couple of thoughts: 

> explicit SPI configuration SPI.begin(SCK, MISO, MOSI, SS); (it works fine)
> SX1278 radio = new Module (5,9,8); (no DIO1 on my LoRa module)

If you initialized the SPI yourself, you need to use the proper Module constructor. See this wiki page: https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup.

> (no DIO1 on my LoRa module)

The one you posted seems to have a DIO1, so I'm not sure what you're talking about. The absence of DIO1 is not an issue in itself, however, you can no longer use the blocking receive and you have to use the interrupt-driven method.

> MCU: [Esp32C3 ] Esp32C3 DevModule

That's not very descriptive. What are you selecting in the Board menu in Arduino IDE?

Overall, this entire thing looks like a bad wiring somewhere on DIO0, since that's the pin that gets triggered when packet is received.

fabrizioPucci, 2022-08-11:
Thank you for your fast answer.

Sorry, You are right, the Ra 02 has the DIO1 pin I forgot to use it because I first used the module RA 01 that doesn't have it.

Anyway for the general cabling if I use the library Arduino LoRa the boards work fine (even if the RSSI is low -130dBm at 1m). 
I would like to experiment using your library.

I will try connecting also the DIO1.

For the board, I add my Arduino configuration but it works fine for other projects and for the same with other libraries.

For the SPI I will try to configure the SPI as you suggest (I tried but I had some problems) but just a question, why if I do not configure the SPI your library report an error on the LoRa device configuration and if I use my configuration it configure the device without any problem and also set all the parameters without any error?

Again, thank you for your help and sorry for my mistakes.

[image]

fabrizioPucci, 2022-08-11:
Really thank you for your help. 
As you correctly have diagnosed, the main problem was on the DIO0 pin it is not having a good contact.
Strangely the Arduino-LoRa library doesn't use it and I didn't check. My fault.
On the other side, my SPI configuration is working fine, now, I am correctly receiving the data.

My payload is less than 80 bytes @1Hz
so I tried with:

radio.setFrequency(433)
radio.setBandwidth(125.0)
radio.setSpreadingFactor(12) 
radio.setCodingRate(6) 
radio.setSyncWord(0x14) 
radio.setOutputPower(17)
radio.setCurrentLimit(80)
radio.setPreambleLength(15)   
radio.setGain(1)

 

[SX1278] Data Length:	12
[SX1278] Data       :	Hello World!
[SX1278] RSSI       :	-56.00 dBm
[SX1278] SNR       :	13.00 dB
[SX1278] Frequency error:	2329.94 Hz
Data received!
[SX1278] Data Length:	12
[SX1278] Data       :	Hello World!
[SX1278] RSSI       :	-57.00 dBm
[SX1278] SNR       :	10.75 dB
[SX1278] Frequency error:	2338.32 Hz

by the way,  Is it the frequency error a problem?

Again,
 thank you

fabrizioPucci, 2022-08-11:
Last update, 
I changed the frequency in the transmitter to reduce the frequency error that now is around 20 Hz.
Thank you

jgromes, 2022-08-11:
I'm glad you got it working! To answer your questions (I hope I got them all).

> For the board, I add my Arduino configuration but it works fine for other projects and for the same with other libraries.

Thanks for that, I just wanted to check if the ESP32C3 is a supported platform, though it looks like it is.

> why if I do not configure the SPI your library report an error on the LoRa device configuration and if I use my configuration it configure the device without any problem and also set all the parameters without any error?

The fact that it works (or seems to be working) without the recommeded configuration method is probably just a coincidence. What you're probably doing right now is calling SPI.begin() twice. Once from the library (since by default, it assumes you are using a default SPI setup), and once in your sketch. On ESP32, that might be fine - on other platforms, it might not.

> by the way, Is it the frequency error a problem?

LoRa can tolerate up to +-25% bandwidth of frequency offset between the transmitter and receiver (in most cases). In your case of 125 kHz bandwidth, the maximum offset it can cope with is about 31 kHz. There's no point in playing with frequency to get a low frequency error - especially since the module you're using is most likely using a cheap crystal oscillator as its frequency reference, which are very prone to frequency drift with temperature and time.

closed 2022-08-11 by jgromes