jgromes/RadioLib #557: disable direct reception, TRNG feature (for Arduino/AVR)

jg1uaa, 2022-08-15:
Hi,

Recently I tried my [LoRaKissTNC-SX1280 project](https://github.com/jg1uaa/LoRaKissTNC/tree/jg1uaa-sx1280) and I found it could not receive transmitted text. RadlioLib-4.4.2 works good, but 4.5.0 and later occurs that problem.

The code is running on LGT8F328P, a compatible of ATmega328P.

Direct reception support is added between 4.4.2 and 4.5.0 (https://github.com/jgromes/RadioLib/commit/7690298d25b6dd85c1ba641980b357a335c15a3e) and the problem is relieved that the size of uint8_t _buffer[] changed from RADIOLIB_STATIC_ARRARY_SIZE to zero.

I think this buffer is allocated statically and it is a great impact for small-memory platform. So I suggest that adding "RADIOLIB_EXCEPT_DIRECT_RECEPTION" to BuildOpt.h, to disable this feature.

And, adding "RADIOLIB_EXCEPT_TRNG" is worth considering. It reduces a bit(4-byte) memory usage but saving more code space.

jgromes, 2022-08-15:
The buffer is 256 bytes long, so for ATmega328P, that's 12.5% of the available RAM. So that's quite a large impact for smaller platforms, which I did not realize.

Adding exclusion macro is probably the only option here, decrasing the size of the buffer doesn't really make much sense since it will severly limit utility of direct reception.

On the other hand, I'm not sure how much of an impact is excluding TRNG going to have - there's no such large buffer there.

jgromes, 2022-08-15:
@jg1uaa I added the macro `RADIOLIB_EXCLUDE_DIRECT_RECEIVE` which you can use in your project. However, it's disabled by default and I would like to keep it like that in the library itself.

jg1uaa, 2022-08-16:
@jgromes I put `RADIOLIB_EXCLUDE_DIRECT_RECEIVE` into BuildOpt.h and my problem has solved. Thanks.

And I tried to reduce randBuff[4] -> [0] in PhysicalLayer.cpp, free memory looks no different. Maybe no need to exclude TRNG feature.

closed 2022-08-16 by jg1uaa