jgromes/RadioLib #566: radio.receive(str) blocks forever on Adafruit Feather 32u4 RFM96 LoRa

a-schneider-fmi, 2022-09-09:
Using the receive example program adapted to [Adafruit Feather 32u4 RFM96 LoRa](https://www.adafruit.com/product/3079) (complete code see below), `radio.receive(str)` hangs forever if no messages are available. If another boards actually sends messages with the transmit example, it works and messages are received correctly. The interrupt example works at all times, i.e. independent of whether a message is available or not. This might be related to #277 but has slightly different behaviour.

My full code:
```
#include <RadioLib.h>

/* Pins for Adafruit Feather32u4 RFM9x */
#define RFM96_CS 8
#define RFM96_RST 4
#define RFM96_INT 7

// Radio settings
#define RFM96_FREQ 434.0 // radio carrier frequency in MHz
#define RFM96_BW 125 // LoRa link bandwidth in kHz
#define RFM96_SF 12 // LoRa link spreading factor
#define RFM96_CR 5 // LoRa link coding rate denominator
#define RFM96_POWER 17 // Transmission output power in dBm

RFM96 radio = new Module(RFM96_CS, RFM96_INT, RFM96_RST);

void setup() {
  Serial.begin(115200);
  while (! Serial); // Wait until Serial is ready.
  Serial.println(F("LoRa test starting."));

  // Initialize RFM96.
  Serial.print(F("[RFM96] Initializing ... "));
  int state = radio.begin(RFM96_FREQ, RFM96_BW, RFM96_SF, RFM96_CR, RADIOLIB_SX127X_SYNC_WORD, RFM96_POWER);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(1);
  }
}

void loop() {
  Serial.print(F("[RFM96] Waiting for incoming transmission ... "));
  String str;
  int state = radio.receive(str);
  if (state == RADIOLIB_ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));
    // print the data of the packet
    Serial.print(F("[RFM96] Data:\t\t"));
    Serial.println(str);
    // print RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[RFM96] RSSI:\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));
  } else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));
  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));
  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  delay(500);
}
```

The output with debug mode enabled is:
```
LoRa test starting.
[RFM96] Initializing ... R	42	12	
SX127x not found! (1 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (2 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (3 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (4 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (5 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (6 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (7 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (8 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (9 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
R	42	12	
SX127x not found! (10 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0012, expected 0x0011
No SX127x found!
R	42	12	
M	SX127x
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	24	5	
W	24	0	
R	24	0	
R	1	9	
R	1	9	
R	1	9	
W	1	8	
R	1	8	
R	1	8	
W	1	88	
R	1	88	
R	1	88	
R	1	88	
W	1	89	
R	1	89	
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	39	12	
W	39	12	
R	39	12	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	B	2B	
W	B	23	
R	B	23	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	20	0	
W	20	0	
R	20	0	
R	21	8	
W	21	8	
R	21	8	
M	SX1278
M	RFM96
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	72	
R	1D	72	
Symbol length: 0.01 ms
R	26	4	
W	26	4	
R	26	4	
R	24	0	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	6	6C	
W	6	6C	
R	6	6C	
R	7	80	
W	7	80	
R	7	80	
R	8	0	
W	8	0	
R	8	0	
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	72	
R	1D	72	
R	1E	70	
W	1E	C0	
R	1E	C0	
R	31	C3	
W	31	C3	
R	31	C3	
R	37	A	
W	37	A	
R	37	A	
Symbol length: 32.77 ms
R	26	4	
W	26	C	
R	26	C	
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	72	
R	1D	72	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	9	4F	
W	9	CF	
R	9	CF	
R	9	CF	
W	9	FF	
R	9	FF	
R	4D	84	
W	4D	84	
R	4D	84	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	26	C	
W	26	C	
R	26	C	
success!
[RFM96] Waiting for incoming transmission ... R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	24	0	
R	40	0	
W	40	0	
R	40	0	
R	1	89	
W	12	FF	
R	F	0	
W	F	0	
R	F	0	
R	D	0	
W	D	0	
R	D	0	
R	1	89	
R	1	89	
W	1	8E	
R	1	8E	
```

It actually claims that the SX127x is not found. Since the module is included in the PCB I can exclude a wiring error. And it is surprising since the function returns success. Moreover, transmission works successfully if another module sends. Why does it hang if no packet is available?

a-schneider-fmi, 2022-09-09:
I get the same behaviour with an [Adafruit RFM96W LoRa Radio Transceiver Breakout](https://www.adafruit.com/product/3073) connected to a Sparkfun Artemis board ([SparkFun Artemis Global Tracker](https://www.sparkfun.com/products/18712)).

jgromes, 2022-09-09:
The "No SX127x found" message can be ignored - it's caused by the fact that there seem to be two possible values of the version register, the official 0x11 and the unofficial 0x12, you seem to ahve the latter. There should be no impact on function.

Regarding the forever block, SX127x in LoRa mode uses DIO1 to sense Rx timeout. Since your code doesn't seem to be using the pin, I assume it's not present on the board.

a-schneider-fmi, 2022-09-10:
Does this mean the hanging is intended behaviour? To avoid the block, I tried to check for incoming messages with `radio.available()`, but it always returns 0 even if a message is available (I added it directly before `radio.receive(str)`, and it returned 0 even if `receive()` returns a message). Is this also intended?
I'll check DIO1.

jgromes, 2022-09-10:
The `available()` method only works with direct mode reception, not in packet mode.

I would call it "expected" behavior - a logical consequence, but slightly beyond intended. You need DIO1 for blocking reception. But if you don't have it, you can still use the interrupt version.

a-schneider-fmi, 2022-09-11:
Would it be possible to provide an `available()` method for packet mode (may be named differently)?
It would be great if the forever block feature would be better documented.

jgromes, 2022-09-11:
`available()` for packet mode is already implemented - it's called `getPacketLength()`, and will return the length of the last received buffer since `startReceive()` was called.

The infinite block is not really a feature. Thinking of it more, I think it would be better to extend the `receive()` method so that when no DIO1 is provided (as was in your case), it will use the same timeout logic as FSK (500% of the expected time-on-air based on the expected packet length).

a-schneider-fmi, 2022-09-12:
`getPacketLength()` does not work as expected: sometimes, it shows a value > 0 although there is no packet available (thus the program hangs), sometimes keeps giving 0 although packets are sent (and received).

a-schneider-fmi, 2022-09-12:
Although the radio's DIO1 pin is not connected to the microcontroller on the Feather LoRa PCB, I have now manually added such a connection and added the respective GPIO as fourth parameter to the constructor of `Module`. With that modification, timeouts work as expected.

jgromes, 2022-09-18:
@a-schneider-fmi I added a software timeout so that when no DIO1 is provided (set to `RADIOLIB_NC`), the timeout of 100 LoRa symbols will be handled by software.

closed 2022-09-18 by jgromes