jgromes/RadioLib #571: High Power consumption after transmit with E19-915M30S ( SX1276 915MHz 1W SMD Wireless Module)

davehummel, 2022-09-16:
When using a powerful amplified radio like E19-915M30S (
SX1276 915MHz 1W SMD Wireless Module), I have found large power consumption after using the asynchronous `startTransmit(buffer, len)` function, continuing permanently until I switch the radio to receive mode.  Even though transmission is complete, the high power consumption (~500 mA at 5v) continues until startReceive() is called.

I believe this is because the TX Enable pin is held high even after transmission is complete.  Adding an Arduino digitalWrite to LOW for that pin during the transmit complete interupt bound function solves the high power consumption problem and does not seem to cause any other problems.  I suspect this behavior should be added to RadioLib internal logic on completion (when using TX enable pin).  

This was discovered using LORA mode.

jgromes, 2022-09-16:
You're right, when using RF switch control from the library, the tx enable will be held high after transmission is done. Unfortunately at that point the control is at the user level, not at the library level, there's no "finishTransmit" method.

However, there's the `standby` method, which disables the RF switch altogether. Rather than adding a new method which basically does the same, I would prefer to amend the examples so that `standby` is called from the example sketch.

Also I think this will be the case for any module that supports `startTransmit`, not just SX127x.

davehummel, 2022-09-16:
That makes sense to me, thanks for the fast response.  
Would calling `standby` send any SPI commands?  I would call `standby` from an interrupt bound function and have other highly active devices using the SPI bus, causing potential instability.

jgromes, 2022-09-16:
`standby` does send an SPI command (namely to set the transceiver to ... well, standby mode). However, I don't think it needs to be done directly from the interrupt service routine. I would just use it to set a flag (as shown in the examples), and then do all the time-consuming stuff (including standby and disableing TX enable) in the main loop.

davehummel, 2022-09-16:
I am running this library in a somewhat complicated scheduling environment to sustain high display framerates and input responsiveness (remote control handheld device).  While I could schedule a follow-up event to then invoke standby safely, that is somewhat heavy handed.  Perhaps this is why finishTransmit might be worth having?  Do you see any problems with using `digitalWrite` (I know your documentation warns against doing just that)?

jgromes, 2022-09-16:
> Do you see any problems with using digitalWrite (I know your documentation warns against doing just that)?

I don't know your platform, or how it implements `digitalWrite`, but it should be safe to call from ISR, since it's usually very simple function. Where exactly does RadioLib documentation warn against `digitalWrite`?

An argument could be made that since user gave the control over the RF switch to the library, it should be the library job to clean up after transmission is over. From that point of view, it makes sense to add a `finishTransmit` method that would perform the cleanup, much like `readData` does in async Rx (after `startReceive`). However, adding it will take a bit of time, since that change needs to be propagated to all modules, all relevant examples and the `PhysicalLayer` abstraction layer.

In your specific case, it's probably easiest to just disable the TX enable pin manually ;)

davehummel, 2022-09-16:
Sounds good, I'll manually control the pin for now.  I really appreciate your work on this library!  [Its been great to work with for my project](https://photos.app.goo.gl/fmFsAj41bGGtGwra7)

I saw the warning in the class documentation, but I think it may be in more than one place.  https://jgromes.github.io/RadioLib/class_module.html#a0ecbb4e1e98094c1296b1e823dc14703  
`When using automatic RF switch control, DO NOT change the pin mode of rxEn or txEn from Arduino sketch!`

jgromes, 2022-09-17:
> I saw the warning in the class documentation

Ah, I remember now. Another argument in favor of `finishTransmit`. I'll implement it and close this afterwards.

> Its been great to work with for my project

Glad it's been useful, the project looks pretty cool!

jgromes, 2022-09-18:
@davehummel I added the above mentioned `finishTransmit`, which will do proper cleanup for any specific radio module. However, FYI usually that includes clearing the IRQ flags, which needs an SPI transaction.

closed 2022-09-18 by jgromes