jgromes/RadioLib #572: Issue with the receiving example code on SX1262 #48 (still a problem)

MrSniffer, 2022-09-17:
I am having the same issue as described formerly, with both my new purchased CubeCell-Plus units. Somehow it thinks it is in FSK mode at startup. Note I don't want to use any of the LoRaWAN code which I gather is embedded. Here is as simple a sketch as I can make; it fails 100% with error code -20. I am stymied!

/*
   RadioLib SX1262 CubeCell Startup Test
*/

// include the library
#include <RadioLib.h>

SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

  void setup() {
  Serial.begin(115200);

  // initialize with explicit settings
  
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(923.3,500.0,8,6,0x12,17,32,0,false);
  
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
}

jgromes, 2022-09-17:
This doesn't seem related to #48 - instead I would suggest to check #522.

Also:

1. Does this still happen for `begin()` method without arguments?
2. Please post the [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode) output

MrSniffer, 2022-09-17:
1. I removed the arguments from begin, e.g. now have 
   Serial.print(F("[SX1262] Initializing ... "));
  //int state = radio.begin(923.3,500.0,8,6,0x12,17,32,0,false);
  int state = radio.begin();
I now get randomly three results, namely success, code = -20 and code = -707.
2. I edited the BuildOpt.h, rebuild and get no debugging output. I'm using Arduino IDE 2.0. What else do I need to do to force a rebuild w debugging on?

Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... failed, code -707
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... success!
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... success!
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... failed, code -20
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... success!

jgromes, 2022-09-17:
We really need that debug output to move forward. Did you enable both debug and verbose level?

Regarding the configuration in `begin`, three parameters that often cause problems are frequency (out of range frequency will cause calibration to fail), setting TCXO voltage for XTAL module (or vice versa) and regulator configuration (LDO vs DC/DC). But again, it's hard to guess what is the issue without the debug output.

MrSniffer, 2022-09-17:
Ooops. I only uncommented VERBOSE. Now both and here's the output, using just radio.begin();:

Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... M     SX126x
CMD     80
DATW    0       AA

CMD     80
DATW    0       A2

CMD     80
DATW    0       A2

CMD     8F
DATW    0       A2      0       A2

CMD     8A
DATW    1       A2

CMD     93
DATW    20      A2

CMD     88
DATW    3       A2      16      A2      A       A2      0       A2      0      A2       0       A2      0       A2

CMD     2
DATW    43      A2      FF      A2

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0      A2       0       A2      0       A2      0       A2

CMD     89
DATW    7F      A2

CMD     80
DATW    0       AA

CMD     17
DATR    0       A2      0       A2      0       0

CMD     97
DATW    0       A2      0       A2      1       A2      40      A2

CMD     11
DATR    0       A2      0       A2

failed, code -20

P.S. I tried adding the delay referred to in 
SX126x/8x SPI communication issue on some platforms](https://github.com/jgromes/RadioLib/issues/158
but this didn't improve things.

P.P.S. rebuilding using Arduino IDE 2.0 gave the following errors (not happy with "'s around CubeCell??)

c:\Users\hsaal\Documents\Meshtastic\libraries\RadioLib\src/BuildOpt.h:684:57: note: #pragma message: RADIOLIB_PLATFORM: CubeCell
     #define RADIOLIB_PLATFORM                           "CubeCell"
                                                         ^
c:\Users\hsaal\Documents\Meshtastic\libraries\RadioLib\src/RadioLib.h:50:41: note: in expansion of macro 'RADIOLIB_PLATFORM'
   #pragma message "RADIOLIB_PLATFORM: " RADIOLIB_PLATFORM

jgromes, 2022-09-18:
Looks like it's failing after command 0x89 - that's RADIOLIB_SX126X_CMD_CALIBRATE. Could you try the default arguments for begin, only setting tcxoVoltage to e.g. 1.6?

MrSniffer, 2022-09-18:
Using radio.begin(434.0,125.0,9,7,0x12,10,8,1.6,false); it still randomly succeeds or gives code=-20.
Debug output sample:
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... M     SX126x
CMD     80
DATW    0       AA

CMD     80
DATW    0       A2

CMD     80
DATW    0       A2

CMD     8F
DATW    0       A2      0       A2

CMD     8A
DATW    1       A2

CMD     93
DATW    20      A2

CMD     88
DATW    3       A2      16      A2      A       A2      0       A2      0      A2       0       A2      0       A2

CMD     2
DATW    43      A2      FF      A2

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0      A2       0       A2      0       A2      0       A2

CMD     89
DATW    7F      A2

CMD     80
DATW    0       AA

CMD     17
DATR    0       A2      0       A2      0       0

CMD     97
DATW    0       A2      0       A2      1       A2      40      A2

CMD     11
DATR    0       A2      0       A2

failed, code -20

Meanwhile, I've tried compiling and running several examples from Heltec for the CubeCell. They don't fail....on these same devices, so it seems very unlikely that there is some hardware problem. Attached is one example sketch to peruse.
[CubeCell_LoRaSender.ino.txt](https://github.com/jgromes/RadioLib/files/9593356/CubeCell_LoRaSender.ino.txt)

MrSniffer, 2022-09-18:
Attached are what I think are the relevant low level files used in the Heltec examples for the CubeCell. Perhaps useful for comparison.
[radio.zip](https://github.com/jgromes/RadioLib/files/9593401/radio.zip)

jgromes, 2022-09-18:
> the relevant low level files used in the Heltec examples for the CubeCell

As far as I can tell, that library (as well as most others) do not actually check return codes from the radio module. So even if it was failing in the same way, there would be no way to know.

> Using radio.begin(434.0,125.0,9,7,0x12,10,8,1.6,false); it still randomly succeeds or gives code=-20

I would try the following:

1. Set TCXO voltage to 0. That's what you started with, if it gives consistent -20 error, it would indicate the module has TCXO as opposed to XTAL (I wasn't able to find this information on Heltec website).

2. If the above is true, try the other TCXO voltages.

3. Try to change the regulator mode (last argument in begin set to true). Again, no information on Heltec site about the regulator type.

MrSniffer, 2022-09-18:
I tracked down a schematic of the board (attached). Looks like there is a XTAL. So does that mean playing w TXCO voltage is irrelevant?

Meanwhile, let me try a functional sketch, IGNORING the initialization code if == -20, and report back.
[Heltec HTCC-AB02A_SchematicDiagram.pdf](https://github.com/jgromes/RadioLib/files/9595095/Heltec.HTCC-AB02A_SchematicDiagram.pdf)

MrSniffer, 2022-09-18:
Proceeding following a code -20 didn't bypass the failure. Doh.
For now, I'll loop until code==0. Setting the TXCO voltage = 1.6 seems to improve chances of success (even with a xtal!?!)

jgromes, 2022-09-18:
> tracked down a schematic of the board (attached). Looks like there is a XTAL.

Actually, that looks like a TCXO (based on the connection to DIO3, which is used as the reference voltage supply, and only being connected to the XTA pin.

Also, it looks like the device is using a DC/DC supply, so at least that is cleared up.

MrSniffer, 2022-09-20:
I've getting to use my CubeCell by specifying TXCO voltage of 1.6 explicitly in the radio.begin parameter. Still fails periodically. My startup code hard reboots if it needs to. Ugh.

int state;
  do {
    state = radio.begin(923.3,500.0,9,6,0x12,17,32,1.6,false);
    if (state == RADIOLIB_ERR_WRONG_MODEM) {
      Serial.println("Oooops..."); Serial.println(" ");
      Serial.flush();
      delay(1000);
      CySoftwareReset();
    }
  }
    while (state != 0 || state == RADIOLIB_ERR_WRONG_MODEM);

jgromes, 2022-09-21:
I don't see much reason for this behavior, sorry. At the same time I don't have the CubeCell module to check for myself. Final though - the periodic nature of the failure is somewhat strange, perhaps an issue with the power supply?

So far, this doesn't seem like an issue in the library itself (possibly in the integration with CubeCell, though I don't exactly see how), so I'm going to close this for now. Feel free to reopen if new information becomes available.

closed 2022-09-21 by jgromes

MrSniffer, 2022-09-21:
I agree, since I have a (ugly) workaround. Others should chime in if they have issues. 
P.S. Is there any potential benefit for me to play around with a different TXCO voltage than 1.60? Higher or lower? By how much??