jgromes/RadioLib #583: RadioLib 5.4.0 fails initialization for CubeCell

MrSniffer, 2022-10-03:
Code runs OK with 5.3.0.
SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE); 
.....
state = radio.begin(923.3,500.0,9,6,0x12,17,PREAMBLE_LENGTH,1.6,false);

Fails with 5.4.0; state = -707 (RADIOLIB_ERR_SPI_CMD_FAILED)

rjmcdougall, 2022-10-04:
also fails with -707 on lillygo t-echo sx1262

rjmcdougall, 2022-10-04:
06:04:57 9 Starting meshradio init...
06:04:57 9 Set radio: region=US, name=testchan, config=0, ch=55, power=30
06:04:57 9 Radio myRegion->freqStart / myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
06:04:57 9 Radio myRegion->numChannels: 104
06:04:57 9 Radio channel_num: 55
06:04:57 9 Radio frequency: 915.875000
06:04:57 9 Slot time: 42 msec
06:04:57 9 Set radio: final power level=22
M       SX126x
M       SX126x reset
CMD     80
DATW    0       AA

CMD     80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

M       SX126x standby
CMD     80
DATW    0       A2

CMD     C0
DATR    0       22      0       22

M       SX126x config
M       SX126x setBufferBaseAddress
CMD     8F
DATW    0       A2      0       A2

CMD     C0
DATR    0       22      0       22

M       SX126x modem
CMD     8A
DATW    1       A2

CMD     C0
DATR    0       22      0       22

M       SX126x fallback
CMD     93
DATW    20      A2

CMD     C0
DATR    0       22      0       22

M       SX126x cad
CMD     88
DATW    3       A2      16      A2      A       A2      0       A2      0       A2      0       A2      0       A2

CMD     C0
DATR    0       22      0       22

M       SX126x irq
CMD     2
DATW    43      A2      FF      A2

CMD     C0
DATR    0       22      0       22

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2

CMD     C0
DATR    0       22      0       22

M       SX126x calibrate
CMD     89
DATW    7F      A2

CMD     C0
DATR    0       2A      0       0

06:04:57 9 SX126x init result -707

jgromes, 2022-10-04:
@MrSniffer could you please post the debug output as well so that we know if this is the same issue?

@rjmcdougall it looks like the failure is occuring after the calibration command - it could be caused by the newly added SPI state verification (#575).

MrSniffer, 2022-10-04:
Happy to post it, but can't any sooner than Friday.

jgromes, 2022-10-04:
@MrSniffer thank you, it fails in the exact same spot. 

The fact that it does not fail on Arduino Uno is strange, but I guess than can be exaplin by Uno being a comparatively slow platform.

It looks like it's caused by performing command verification during an ongoing calibration. In the latest commit (170ce9752bf7ccf78713d60fb1a0b93f9ba3e3d2), I changed this to skip verification in this one case.

@MrSniffer @rjmcdougall could you try this hotfix on your boards?

rjmcdougall, 2022-10-04:
Oh interesting -- yes I will try this

MrSniffer, 2022-10-04:
In a bit I will try this with a CubeCell and also a LilyGo TBeam V1.1, both w SX1262s.

jgromes, 2022-10-04:
Thank you for the confirmation @rjmcdougall, I will publish a hotfix release ASAP

MrSniffer, 2022-10-04:
Not quite there, I think. Instead of failing 100%, it is maybe 10% now. Here is the debug output:
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1262] Initializing ... M     SX126x
CMD     80
DATW    0       AA

CMD     80
DATW    0       A2

CMD     C0
DATR    0       A2      0       22

CMD     80
DATW    0       A2

CMD     C0
DATR    0       A2      0       22

CMD     8F
DATW    0       A2      0       A2

CMD     C0
DATR    0       A2      0       22

CMD     8A
DATW    1       A2

CMD     C0
DATR    0       A2      0       22

CMD     93
DATW    20      A2

CMD     C0
DATR    0       A2      0       22

CMD     88
DATW    3       A2      16      A2      A       A2      0       A2      0       A2      0       A2      0       A2

CMD     C0
DATR    0       A2      0       22

CMD     2
DATW    43      A2      FF      A2

CMD     C0
DATR    0       A2      0       22

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2

CMD     C0
DATR    0       A2      0       22

CMD     89
DATW    7F      A2

CMD     80
DATW    0       AA

CMD     17
DATR    0       AA      0       0       0       0

CMD     97
DATW    0       AA      0       3F      1       0       40      20

failed, code -707

MrSniffer, 2022-10-04:
P.S. I am still getting occasional initialization errors with code -20 w 5.4.0 + this hotfix.

MrSniffer, 2022-10-04:
I just tested multiple times on the LilyGo TBeam V1.1 SX1262 and no -707 errors found. Still having CubeCell failures. I can try to work around by re-initializing in the interim, as I do for the -20 error.

MrSniffer, 2022-10-04:
T-Beam now exhibiting random failures, e.g.
Bad startReceive(). Err=-705 RADIOLIB_ERR_SPI_CMD_TIMEOUT
Still lurking problems, this one not at startup hence rather hard to get debug log.

jgromes, 2022-10-04:
Reopened until I have more time to test with ESP32.

jgromes, 2022-10-05:
@MrSniffer I was not able to replicate your issues directly, however, in two cases I was able to get my SX126x/ESP32 into an unstable state.

The first case was by introducing a slight error in the configuration - for example, attempting to use TCXO on a module with XTAL, which consisntely yielded -707 error on startup.

The second case was by using insufficient power source. For example, by powering an SX1268 module from a 3.3V pin of NodeMCU 32S board connected only to a USB port, I was able to get semi-regular -2/-20 error codes. When using a dedicated LDO, there were no issues transmitting or receiving on ESP32. I would suggest to examine the board(s) you are using, and what is the power source for the SX1268.

closed 2022-10-05 by jgromes

MrSniffer, 2022-10-06:
FYI: I am using the LilyGo T-Beam V1.1 1262 model. It surely has a TCXO. I am using a voltage of 1.6 (default) but I have no clue what it ought to be. Where would I find that spec?

jgromes, 2022-10-06:
@MrSniffer you will have to contact the manufacturer of that module.

jgromes, 2022-10-06:
@MrSniffer I fixed an issue in the receive method that caused calls to `receive()` to end with error -705 when there was a timeout event. Not sure if this is one of the issues you had, but wanted to let you know.

MrSniffer, 2022-10-06:
I have queried LilyGo to find out what TCXO voltage they designed for. The Semtech SX1262 data sheet says it should be a minimum of 1.6V, typical 1.7V, up to a max of 3.3V (using more power!)