jgromes/RadioLib #586: Getting -2 and -20 errors constantly on E22-400M30S

EliasA97, 2022-10-05:
I recently started getting these errors when I start my sketch. 

-------------------------------------------------------------------------------
[SX1262] Initializing ... M	SX126x
CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	80	
DATW	0	80	

CMD	8F	
DATW	0	80	0	80	

CMD	8A	
DATW	1	80	

CMD	93	
DATW	20	A0	

CMD	88	
DATW	3	82	16	80	A	80	0	80	0	80	0	80	0	80	

CMD	2	
DATW	43	82	FF	A2	

CMD	8	
DATW	0	80	0	80	0	80	0	80	0	80	0	80	0	80	0	80	

CMD	89	
DATW	7F	A2	

CMD	80	
DATW	0	AA	

CMD	17	
DATR	0	A0	0	0	0	0	

CMD	97	
DATW	0	0	0	80	1	80	40	80	

failed, code -2

-------------------------------------------------------------------------------

The code is the following:

-------------------------------------------------------------------------

#include <RadioLib.h>
#include "AESLib.h"

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(10, 2, 3, 9); // NSS, DIO1, NRST, BUSY

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

// or using CubeCell
//SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

  uint8_t key[] = {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15};
  char data[20]; //16 chars, 17 bytes including zero terminator

void setup() {
  Serial.begin(115200);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(434.0, 250.0, 12, 7, 0x12, 10-12, 16, 1.6, false); // frequency, bandwidth, spreading factor, coding rate, sync word, output power, preamble length, tcxo ref voltage, ldo regulator mode
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

    if (radio.setDio2AsRfSwitch() != RADIOLIB_ERR_NONE) {
    Serial.println(F("Failed to set DIO2 as RF switch!"));
  }

  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1262] Starting to listen ... "));
  state = radio.startReceive();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // if needed, 'listen' mode can be disabled by calling
  // any of the following methods:
  //
  // radio.standby()
  // radio.sleep()
  // radio.transmit();
  // radio.receive();
  // radio.readData();
  // radio.scanChannel();
}

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

// this function is called when a complete packet
// is received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!

void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
  // check if the flag is set
  if(receivedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
  
    int state = radio.readData(data, 20);

      
    

    if (state == RADIOLIB_ERR_NONE) {
      // packet was successfully received
      Serial.println(F("---------------------------------------------"));
      Serial.println(F("[SX1262] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1262] Data: "));
      aes128_dec_single(key, data);
      //Serial.print("decrypted: ");
      Serial.println(data);
      
    //  for (byte i=0; i<20; i++) {
    //    Serial.print(data[i]&0xFF,HEX);
    //  }

      memset(data, 0, sizeof(data));
      
     
      Serial.println("");

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1262] RSSI: "));
      Serial.print(radio.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1262] SNR: "));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }

    radio.startReceive();
    enableInterrupt = true;
  }

}

-------------------------------------------------------------------------

I tried the examples provided but I still get -2 and -20 errors.

Until today, everything worked. I tried to reinstall v.5.3 but still the problem exists. 
I can't believe that my modules may be burnt or something.

Also the AES library is from here: https://github.com/DavyLandman/AESLib

jgromes, 2022-10-05:
Please post the information listed in the template, mainly the Arduino platform you are using (Uno, ESP32 etc.).

Also, which version of the library is the debug dump from? There was an error in 5.4.0 that behaved similar to what you are describing, but the debug dump does not match 5.4.0/5.4.1.

EliasA97, 2022-10-05:
I'm experiencing the same problem on arduino mega and arduino uno. 

The error code I posted above was with these parameters:

Ebyte SX1262 (E22 - 400M30S)

Arduino IDE 1.8.19
Library version 5.3.0

I installed 5.4.1 version and I got these:

`[SX1262] Initializing ... M	SX126x
CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
[... 1044 lines cut ...]
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	80	

CMD	C0	
DATR	0	0	0	0	

CMD	80	
DATW	0	0	

CMD	80	
DATW	0	0	

failed, code -2
`

jgromes, 2022-10-06:
It seems there is almost nothing on the SPI bus. Are you sure there was no change in the hardware? Also, since you're using 5V-logic Arduino boards, do you have a voltage level converter between that and the Ebyte module?

EliasA97, 2022-10-06:
I've made a replica of your 1W groundstation, but instead of ESP32, I've made arduino uno and mega versions. Everything worked fine until 2 days ago. Also the SPI bus is only used by the module. Should I try #include SPI?

jgromes, 2022-10-06:
As far as I can thell from the datasheet, CD74HC4050 can only do down conversion to Vcc, so if your Vcc is connected to 3.3V then that *should* be fine for converting signals from Arduino to the SX1262 (not the other way round). I have not seen it being used in this way before though.

Adding #include SPI is not needed, the library includes it on its own.

Unfortunately I am not able to replicate the issues you are seeing, with my Arduino Uno, MAX3001 level shifter and SX1262. While the timing is suspicious, from the debug mode output it really looks like a hardware issue - the SPI bus is always at 0. I would try to examine the communication directly, with a signal analyzer or an oscilloscope.

closed 2022-10-06 by jgromes

EliasA97, 2022-10-06:
Yes, the CD74 gets 3.3V in order to convert the 5V signals. The 3.3V signals returning from Lora to Arduino are not level shifted. After testing I found that my mega board works fine, while the uno board doesn't recognise the lora module.  I can program the atmega328 through USBasp via SPI. That means the SPI works. Unfortunatelly I conclude that the Lora module may be burnt somehow.

This part is from the datasheet. Maybe those chances aren't so slim after all. 

[image]