jgromes/RadioLib #589: Unable to send int data.

VASAPOL, 2022-10-15:
Unable to sent int data

```
#include <RadioLib.h>

SX1278 radio = new Module(10, 2, 9, 3);
// create AFSK client instance using the FSK module
// pino D1 do ESP8266 conectado ao pino DIO2 do SX1278
AFSKClient audio(&radio, 5);
// create AX.25 client instance using the AFSK instance
AX25Client ax25(&audio);

void setup() {
  Serial.begin(9600);
  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing … "));
  int state = radio.beginFSK();
  state = radio.setFrequency(433.5);
  state = radio.setFrequencyDeviation(10.0);
  state = radio.setRxBandwidth(50.0);
  state = radio.setOutputPower(2.0);
  state = radio.setCurrentLimit(100);
  state = radio.setBitRate(1.2);
  //state = radio.setBitRate(600.0);
  //state = radio.setCRC(true,false);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  // initialize AX.25 client
  Serial.print(F("[AX.25] Initializing … "));
  state = ax25.begin("HS0AW");
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  // send AX.25 unnumbered information frame
  Serial.print(F("[AX.25] Sending UI frame … "));
  // destination station callsign:     "NJ7P"
  // destination station SSID:         0
  int state = ax25.transmit(int(15), "HS0AW");

  
  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));
  } else {
    // some error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }
  delay(10000);
}

```

from my SoundModem its shown
```
1:Fm HS0AW To HS0AW <UI R Pid=F0 Len=1> [20:26:29R] [+++]

```

MCU: Arduino UNO with SX1278

jgromes, 2022-10-15:
What led you to the conclusion that this is the correct way to send an integer?

Soundmodem only shows strings (at least the default 1200 baud AFSK/AX.25 version). Therefore, you need to take your integer, and convert it to string. That's a very common task, solution to which is one google search away. Only then it makes sense to pass the string to the transmit method.

Also, why did you feel the need to try and explicitly configure the settings like bitrate? `AX25::begin()` does it all automatically, as the examples show.

TL;DR please follow the examples and when in doubt, check the API documentation linked from the readme.

closed 2022-10-15 by jgromes

VASAPOL, 2022-10-15:
I already convert it to String but still not working
```
int state = ax25.transmit(String(15), "HS0AW");
```

src/main.cpp:61:48: error: no matching function for call to 'AX25Client::transmit(String, const char [6])'

jgromes, 2022-10-15:
That's not a string, that's a *String* - an important distinction:

* string (or a C-string) is an array of characters terminated by a NULL
* String is an Arduin container class that attempts to simply string operations for beginners, and fails horribly at it.

Personally, I would suggest to use the `snprintf()` C function. You can use the String class (provided you don't care about it possibly corrupting the memory) e.g. like this:

```
String str = String(15);
ax25.transmit(str.c_str(), "HS0AW");
```