jgromes/RadioLib #595: Missing useRfo mode in setOutputPower method for select SX127x, RFM9x chips

lesykm, 2022-10-25:
Like in https://github.com/jgromes/RadioLib/issues/385 (which was fixed in https://github.com/jgromes/RadioLib/commit/9d3807168f0cfed6b2e7e18644b13ee8c549be38) useRfo is needed in other Lora chips which have them.
There are plenty boards (notably ExpressLRS but there are lot others, like FRSky R9M, etc) on the market which use aforementioned chip + RFO pin + external power amplifier.

Affected chips:
SX1276
SX1277
SX1279

RFM95
RFM96
RFM97
RFM98

Essentially all SX127x and RFM9x should support RFO mode.

Please refer for a datasheets to find the mention of RFO pin in all these chips:

https://cdn.sparkfun.com/assets/learn_tutorials/8/0/4/RFM95_96_97_98W.pdf
https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf

ExpressLRS uses own driver, you can find how they set RFO settings here:
https://github.com/ExpressLRS/ExpressLRS/blob/26ce80fcddccb0dc20e2fc80b8cf940ead4fcdbc/src/lib/SX127xDriver/SX127x.cpp#L178

Thanks!

jgromes, 2022-10-25:
Have you tried the `setOutputPower(int8_t power, bool useRfo)` method? It's available for all SX127x modules.

lesykm, 2022-10-25:
I apologize for the noise, did not notice inheritance order.
I'm having power issues with setting on/of of RFO mode but then this is not related to this issue.

closed 2022-10-25 by lesykm