jgromes/RadioLib #596: Timing issues causing decoding issues for protocols where the timing is controlled by the CPU

KevWal, 2022-10-27:
**Is your feature request related to a problem? Please describe.**
When using slow CPU speeds (to save Solar / Battery power) the timing of (micros() - start < _bitDuration) gets too far out and causes significant issues decoding.  For example, this type of delay in a 50 Baud FSK4 packet causes decodes to fail:

[image]

**Describe the solution you'd like**
Moving from while(mod->micros() - start < _bitDuration) to Interupt based timing for time sensitive protocols - RTTY & FSK4 for  example.  In testing I see 50baud at 1.8432Mhz generating timing that is within 4us, compared to regualr differences of 200us with micros()

[image]

**Describe alternatives you've considered**
Unable to run the processor at a faster speed due to low voltages

**Additional context**
I am sure this is far from perfect, but the code I have used:

```
void isr_timer1_start() {
  noInterrupts();                                
  TCCR1A = 0;                                    // Set entire TCCR1A register to 0 = Disconnect interrupt output pins, sets normal waveform mode. We're just using Timer1 as a counter.
  TCNT1  = 0;                                    // Initialize counter value to 0.
  TCCR1B = (1 << CS11) |                         // Set CS11 bit to set prescale to /8
    (1 << WGM12);                                // turn on CTC mode.
  //OCR1A = 4607;                                  // Set up interrupt trigger count = (mhz / (prescaler * baud)) - 1
  OCR1A = (F_CPU / (8 * FSK4_BAUD)) - 1;          // Set up interrupt trigger count = (mhz / (prescaler * baud)) - 1
  TIMSK1 = (1 << OCIE1A);                        // Enable timer compare interrupt.
  interrupts();                                  // Re-enable interrupts.
}

// Disable Timer1
void isr_timer1_stop() {
  noInterrupts();
  TCCR1B = 0x00;
  interrupts();
}

// Timer interrupt vector. This toggles the variable we use to gate each column of output to ensure accurate timing. 
// Called whenever Timer1 hits the count set in interruptSetup().
ISR(TIMER1_COMPA_vect){
  proceed = true;
} 

void fsk4_tone(PhysicalLayer* phy, uint8_t i) {
  proceed = false; while (!proceed) { yield(); } // Wait for the right time to start the next tone
#ifdef DEVTIMING
  PINB = 0b00010000; //digitalWrite(DEVTIMING, !digitalRead(DEVTIMING));
#endif
  fsk4_transmitDirect(phy, fsk4_base + fsk4_tones[i]);
}

void fsk4_standby(PhysicalLayer* phy){
    proceed = false; while (!proceed) {yield();} // Make sure the last tone lasts the required amount of time.
#ifdef DEVTIMING
  PINB = 0b00010000; //digitalWrite(DEVTIMING, !digitalRead(DEVTIMING));
#endif
    phy->standby();
}
```

jgromes, 2022-10-27:
While a dedicated timer would be a lot more accurate than the current while/micros combination, it's not feasible to implement it on every single platform RadioLib supports - there's about 20 of them.

I can see at leat two workarounds:

1. Adding a manual correction coefficient, like for AX.25 or SSTV. This would allow you to tune the 4-FSK tone length according to your exact setup. This could also be helpful to others, so I would prefer this solution. Of course, it would have to be tested this does actually resolve your problem.

2. Overriding `Module::micros()` from user code. Since you seem to have a bit of a special case with low voltage, this could be an option for you, to implement something that is either more precise than the Arduino micros(), or just waits for the `proceed` flag.

KevWal, 2022-10-29:
Thanks Jan for the quick reply!

The manual correction would certainly be helpful, the majority of tones were too long, so for rtty it gets out of sync after a few characters and then never recovers.  Not sure if it would correct it enough, as a few were too short and it would make those worse of course.

The option to create my own timer toggling a flag, and the option in radiolib to use that flag would be great - my interrupt based flag routine was very acurate and would remove the complexity of the library having to cope with all the different platforms?

Thanks very much
Kev

jgromes, 2022-10-29:
@KevWal could you check what happens if instead of 50 baud you try to manually tune the baudrate on the transmitter (e.g. 49 baud) so that it is realiably decoded? If this works, then correction is possible, if not, some other approach will be needed.

KevWal, 2022-10-30:
So testing again with RTTY, set at 50 Baud, from a random selection of tones I see 20.299ms, 20.214ms, 20.181ms, 20.075ms:
[image]
and I get very little decoded: 16:07:34 [ERROR]  Couldn't decode ASCII - 'ascii' codec can't decode byte 0x97 in position 5: ordinal not in range(128) - b'\x00$$$$\x97\xaeZ\x84\x86\ [...]  or 16:08:31 [ERROR]  Couldn't decode ASCII - 'ascii' codec can't decode byte 0xda in position 14: ordinal not in range(128) - b'$$$$KW-BCKbXbf\xda\xc1\xb1\xc1\xc1\xd3\x83\x83\xd3 [...]

If I go to 51 baud, from a random selection of tones I see: 19.713ms, 19.840ms, 19.867ms:
[image]
and I still get very little decoded:  16:13:21 [ERROR]  Couldn't decode ASCII - 'ascii' codec can't decode byte 0xb1 in position 10: ordinal not in range(128) - b'\x00$KW-BCK1,\xb1\x99\x9b\x9b\x96\xcc\xcc\xce [...] or 16:13:34 [ERROR]  Couldn't decode ASCII - 'ascii' codec can't decode byte 0x99 in position 14: ordinal not in range(128) - b'\x00$$$$KW-BCK1,1\x99\x9b\x9c\x96\xc8\xcc\xce\xcc\xcc\ [...]

So it looks like Baud is too coarse, and as an Int I can't go to 50.5 :)

Thanks very much
Kev

KevWal, 2022-11-05:
I am sure my coding is probably not the best, but I have it coded as:

Adding to BuildOpt.h:
```
/*
 * KW Uncomment to enable interrupt based timing for the RTTY protocol.
 */
#if !defined(RADIOLIB_INTERRUPT_TIMING)
  #define RADIOLIB_INTERRUPT_TIMING
#endif
```
Adding to RTTY.h:
```
// KW Used for interrupt based timing
#if defined(RADIOLIB_INTERRUPT_TIMING)
	extern volatile bool proceed;
#endif
```
and replacing these 3 routines in RTTY.cpp:
```
// KW
void RTTYClient::mark() {
  Module* mod = _phy->getMod();
  #if defined(RADIOLIB_INTERRUPT_TIMING)
  proceed = false; while (!proceed) { mod->yield(); }
  #else
  uint32_t start = mod->micros();
  #endif
  transmitDirect(_base + _shift, _baseHz + _shiftHz);
  #if !defined(RADIOLIB_INTERRUPT_TIMING)
  while(mod->micros() - start < _bitDuration) {
    mod->yield();
  }
  #endif
}

// KW
void RTTYClient::space() {
  Module* mod = _phy->getMod();
  #if defined(RADIOLIB_INTERRUPT_TIMING)
  proceed = false; while (!proceed) { mod->yield(); }
  #else
  uint32_t start = mod->micros();
  #endif
  transmitDirect(_base, _baseHz);
  #if !defined(RADIOLIB_INTERRUPT_TIMING)
  while(mod->micros() - start < _bitDuration) {
    mod->yield();
  }
  #endif
}

// KW
int16_t RTTYClient::standby() {
  #if defined(RADIOLIB_INTERRUPT_TIMING)
  // Make sure we transmit our last tone for long enough before standby
  Module* mod = _phy->getMod();
  proceed = false; while (!proceed) { mod->yield(); }
  #endif
  #if !defined(RADIOLIB_EXCLUDE_AFSK)
  if(_audio != nullptr) {
    return(_audio->noTone());
  }
  #endif
  return(_phy->standby());
}
```

I haven't got the hardware with me to test it this weekend - will test and come back soon.

jgromes, 2022-11-06:
This looks pretty good and could be implented to all protocols that need precise timing (e.g. SSTV). Couple of thoughts:

1. Instead of rewriting each instance of the "while(micros)" loop, I would create a common method with that functionality. e.g:

```c++
void Module::waitForMicroseconds(uint32_t start, uint32_t len) {
  #if defined(RADIOLIB_INTERRUPT_TIMING)
  (void)start;
  // TODO: maybe instead of voiding len, call some "interrupt setup" callback here?
  (void)len;
  proceed = false;
  while (!proceed) { mod->yield(); }
  #else
   while(this->micros() - start < len) {
    this->yield();
  }
  #endif
}
```

2. I would call the flag something less generic than `proceed`, e.g. `RadioLibTimerFlag`.

3. Some example of usage could be useful for e.g. Arduino Uno, but I can take care of that.

If this can be implemented well, I will acceptit as a PR. There have been instances like https://github.com/jgromes/RadioLib/issues/553#issuecomment-1226023501, where it seems inconsisntent timing of micros() is leading to distortions that cannot be fixed by linear corrections.

KevWal, 2022-11-07:
Just to confirm my changes certanily seem to sort the timing issues for me:  
[image]

And I can now decode my transmission:
13:48:08 [INFO]  Habitat - Uploaded sentence: $$KW-BCK1,1842,00:00:00,0.00000,0.00000,0,0*65FB

jgromes, 2022-11-07:
Nice, I think that validates the approach. Would you be willing to implement the changes more generically (as outlined in my previous comment)? If so I would gladly accept the PR.

And then maybe you can transmit SSTV from your HAB ;)

jgromes, 2022-11-16:
@KevWal if you don't have the time then I can do it, just didn't want to take the credit for your idea ;)

jgromes, 2022-11-18:
@KevWal Had some time today, so I added this in a more generic fashion - see https://github.com/jgromes/RadioLib/wiki/Interrupt-Based-Timing

closed 2022-11-18 by jgromes

KevWal, 2022-11-18:
Fantastic, thankyou, and you did it with code much better than I could have achieved :)

P.s. FSK4 is another one that is timing based so also has timing issues resolved by this.

P.p.s I do.wonder if the timing of this is as good as possible - here I think you start the tone, you then wait for tone length, you then spend some cpu cycles deciding what tone is needed next, and then you set the next tone.

In my version I make the 'what tone next' decision within the tone transmission time.  Not sure how significant that is?

Cheers
Kev

jgromes, 2022-11-18:
It is going to be off by a couple of cycles, though I did not test by how much - just that all of the digital modes still decode while using the interrupt-based timing.

In either case, you can still adjust the tone length by correction factors (because it is the library that calculates the duration) , so if the difference is significant, it can be tuned manually.