jgromes/RadioLib #597: Lambda62(SX1262): interrupt not generating

BlackFingerz, 2022-10-28:
**Describe the bug**
Hi,

Hi, I’m using a RF solutions Lambda62 arduino shield to develop an LoRa RC Application.
The shield uses the Semtech SX1262. I am using an arduino Uno R3 as the MCU for both transmitter and receiver.  

I had success in configuring the chip using the SPI interface directly, ie without using a library. I have used this setup to send and receive Packets in FSK mode but was unable to implement The loRa modem. 

At that point I switched to using a library and after several unsuccessful options settled on RadioLib. (Massive thanks for your hard work on this. I've found it both comprehensive and intuitive.)

I have configured the 'sx126x_Transmit.ino' and 'sx126x_Receive.ino' examples to work with the Lambda62 module and they function as expected.

I moved on to use the 'Transmit_interrupt' and 'Receive_Interrupt' examples I have a problem. 
**Namley that the Interrupt is not triggered. It looks like the sx1262 is not triggering the interrupt pin (pin 7).** 

The output from debug mode also suggests the interrupt on DIO 1 is configured correctly,  ie  IRQ 'RxDone' unmasked and enabled on DIO1.
QUESTION: should all interrupts on DIO2 and DIO3 be disabled?

In my own code I have used the interrupt on pin 7 sucessfully suggesting this is not a hardware issue.

When using 'sx126x_Receive_Interrupt'  if I force the 'receiveFlag' to true 'a' packet is received. (the data in the string is sometimes scrambled in the output written to the serial port.)  

I have also tried the same sketch with the 'spi slowdown' option enabled in Buildopt.h and still encountered the same problem.
OTHER NOTES: the LAMBDA62 module is preset at 915Mhz, 
it uses DIO2 as an RF switch and does not use a TCXO. 

This issue may be related to issue #271. I tried the work around suggested within that issue (although it was not recommended/verified here) but it had no effect.  

Any help or suggestions you can offer on this would be very much appreciated! 

Thx

**To Reproduce**
Attached here as txt files are:
Configured transmit.ino:    Radiolib_Lambda62_SX126x_Transmit
Configured Receive Interrupt.ino:    RadioLib_Lambda62_SX126x_Receive_Interrupt
and Verbose Debug Output File from Transmit_Inrerrupt: 

[Debug Output Radiolib_Lambda62_SX126x_Receive_Inrerrupt.txt](https://github.com/jgromes/RadioLib/files/9887077/Debug.Output.Radiolib_Lambda62_SX126x_Receive_Inrerrupt.txt)
[Radiolib_Lambda62_SX126x_Receive_Inrerrupt.txt](https://github.com/jgromes/RadioLib/files/9887078/Radiolib_Lambda62_SX126x_Receive_Inrerrupt.txt)
[Radiolib_Lambda62_SX126x_Transmit.txt](https://github.com/jgromes/RadioLib/files/9887079/Radiolib_Lambda62_SX126x_Transmit.txt)

**Expected behavior**
Using 'sx126x_Transmit.ino' on the transmit device and 'sx126x_Receive_Interrupt' on the Receiver I expected to see the reception of the 'hello world' message once every second with no receive timeouts 

**Additional info (please complete):**
 - MCU: Arduino Uno R 3
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type: RF Solutions: Lambda62 (https://www.rfsolutions.co.uk/downloads/6245d785a8ca8974DS-LAMBDA62-ARD-1.pdf)
 - Arduino IDE version: 1.8.19, (Windows store 1.8.57.0)
 - Library version  5.4.1

jgromes, 2022-10-28:
> In my own code I have used the interrupt on pin 7 sucessfully suggesting this is not a hardware issue

How? Arduino Uno does not have an external interrupt on pin 7, only on pins 2 and 3. This is a limitation of the microcontroller. So whoever designed the has very little knowledge about Arduino Uno.

jgromes, 2022-10-28:
Also, that shield does not seem to have any sort of voltage level conversion between the LoRa module (a 3.3V logic device) and the Arduino Uno (a 5V logic device). So it is going to slowly die if you use it. There is no dedicated power source for the LoRa module, relying on the LDO onboard the Arduino. During transmissions, expect problems, instability, microcontroller resets etc. The Rx/Tx switch seems suspicious at best.

As an electrical engineer, it makes me physically sick that someone has the balls to sell this as a "product" - I would struggle to call it a high-school weekend project.

For 37 pounds, it seems to be a lot worse than 95% of LoRa modules you can get for 5 bucks on Ebay.

BlackFingerz, 2022-10-28:
Thanks for your response, and such a prompt one!

So, No interrupt on Pin 7. I was completely oncorrect to say "In my own code I have used the interrupt on pin 7 sucessfully suggesting this is not a hardware issue."

Reviewing my own origial code I see that I was using DIO1 to pull pin 7 low then using 'digital read' to trigger a receive routine. So not using an interrupt at all.

I have now strapped DIO 1 to pin 2 and reconfigured. Receive_interrupt is functioning correctly.

BlackFingerz, 2022-10-28:
Thanks also for your feedback on the Lambda62ARD arduino shield. Dulely noted re the absence of a dedicated power supply and good to know it may be generating problems.
 
Reviewing this it is intended as a development board which incorporates a more comprehensive module
(https://www.rfsolutions.co.uk/radio-modules-c10/dip-lora-transceiver-module-20km-868mhz-p807)
 (I have found RF Solutions to be a reliable supplier of components in the past.) I hope the module itself is more comprehensive than the dev board.

jgromes, 2022-10-28:
Glad you got it working.

However, I stand by my assesment - selling that board at its current (or any, really) price is a rip-off. Not what I would call a behavior of "a reliable supplier".

closed 2022-10-28 by jgromes