jgromes/RadioLib #600: Enable CRC for SX1276 and possibly other SX127x radios

nseidle, 2022-11-02:
We are seeing byte errors during transmissions of 100s to 1000s of bytes between two SX1276 based radios. I had assumed CRC was enabled but now I believe the library is not enabling CRC by default.

setCRC() looks available for SX1272 and SX1278 but not exposed for SX1276 (shown below):

[image]

By default, CRC is disabled (shown below):
[image]

I think I can add setCRC() and the accompanying receive check, but I'm not familiar with radios outside SX1276. If I create a PR to add this functionality for the SX1276, would that be helpful? Is there anything else that I should consider adding?

jgromes, 2022-11-02:
Seems like CRC is only enabled for FSK, so I guess you're talking about LoRa.

However, the methods are available already, since SX1276 inherits from SX1278. So it should be just a matter of putting it into the begin() method.

nseidle, 2022-11-03:
Sorry, yes, I'm working in LoRa mode.

Ah! I missed the inheritance. I've got CRC working with setCRC(true), and I'm seeing failures where I was not previously (below) so I'm confident it's enabled:
[image]

> So it should be just a matter of putting it into the begin() method.

Indeed, but I leave it to your judgement if adding setCRC(true) to begin() be will break existing code or not; I'm worried it would catch users off guard. I wonder if it would cause problems with users operating at spread factor 6 (explicit header mode).

I'm all set for my particular application (thanks!) so I'll close. Feel free reopen if needed.