jgromes/RadioLib #602: [SX1278] Stream receive mode works incorrect

notsleep196, 2022-11-09:
Hello! 
I already tried to use SX1278 and Radiolib to receive long packets in FSK #399  and i'm glad to see the implementation of "Stream mode" in it. I tried to use your [example](https://github.com/jgromes/RadioLib/blob/master/examples/Stream/Stream_Receive/Stream_Receive.ino) and watch how fifoGet() functionality works.
This functionality works unexpectedly. To simplify I sent messages of 60 bytes payload (+1 byte packet length) from another SX1278 module in variableLengthPacketMode. I think it is valid scenario to check how FIFO threshold functionality works, because default FIFO threshold now is 31.
I used [example](https://github.com/jgromes/RadioLib/blob/master/examples/Stream/Stream_Receive/Stream_Receive.ino) for stream receive (with small changes, see below):

- totalLength value changed to 60
- added interruptTriggeredFlag to check if FIFO threshold interrupt was triggered
- my fifoGet interrupt handler: 

```cpp
void IRAM_ATTR fifoGet() {
    // check if the interrupt is enabled
    if(!enableInterrupt) {
        return;
    }

    // set the flag when we receive the full packet
    receivedFlag = radioRx.fifoGet(rxBuffer, totalLength, &receivedLength);
    interruptTriggeredFlag = true;
}
```
- in main loop added some debug info:
```cpp
if (interruptTriggeredFlag) {
    Serial.println("FifoThresh interrupt triggered");
    Serial.print("ReceivedLength: ");
    Serial.println(receivedLength);
    interruptTriggeredFlag = false;
}
```
and 
```cpp
 // print data of the packet
Serial.print(F("[SX1278] Data: "));
for (auto i = 0; i < totalLength; i++) {
    Serial.print(rxBuffer[i]);
    Serial.print(' ');
}
Serial.println();
```
**Steps to reproduce**
Sending next data: 3C 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 
Result: 
```
[SX1278] Initializing Rx... success!
[SX1278] Starting to listen ... success!
FifoThresh interrupt triggered
ReceivedLength: 30
```
So i couldn't see the reception of the whole packet and as we can see from additional info FIFO threshold interrupt triggered only once and only 30 bytes of data were read. If i send same data again i can see next result:
```
[SX1278] Initializing Rx... success!
[SX1278] Starting to listen ... success!
FifoThresh interrupt triggered
ReceivedLength: 30
FifoThresh interrupt triggered
ReceivedLength: 60
[SX1278] Received packet!
[SX1278] Data: 60 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 60 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 
```
Only first 30 bytes of both packets are in result buffer.
**Proposed solution**
I think the problem is in `fifoGet()` method in RadioLib. If i comment ` clearIRQFlags();` string in `fifoGet()` method FIFO threshold interrupt triggers twice (as expected) and we can see packet reception. Output for this patch:
```
[SX1278] Initializing Rx... success!
[SX1278] Starting to listen ... success!
FifoThresh interrupt triggered
ReceivedLength: 30
FifoThresh interrupt triggered
ReceivedLength: 60
[SX1278] Received packet!
[SX1278] Data: 60 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 
```
So, we can receive long packet, FIFO threshold interrupt works as expected. 

The last thing: this two strings in `fifoGet()` method:
```
// dump the padding byte
  _mod->SPIreadRegister(RADIOLIB_SX127X_REG_FIFO);
```
I saw this hack with padding byte for TX, but for RX it looks strange, as you can see above, one value from initial byte sequence  (1E hex, 30 decimal) was missed on RX side. So i think this "dump padding byte" is also a mistake. If i comment this, i see valid byte sequence as i sent.

 - MCU: Wemos D1 mini ESP32
 - Wireless module type SX1278
 - Library version 5.4.1
Looking forward for your reply!

jgromes, 2022-11-09:
Hmm, I'm not actually sure whether I tested with transmission of less than 256 bytes (since that was the original use case). I will try to reproduce this and let you know.

jgromes, 2022-11-11:
I spent a little while thinking about this, before I realized it doesn't actually make sense - the stream mode is entered only if the packet length is higher than the SX1278 internal FSK buffer size (64 bytes). So if the packet is shorter than this threshold, it will use the normal packet transfer. The problem is that there's now also an interrupt service tied to the DIO lines, which attempts to handle FIFO on its own. This limitation probably should be documented somewhere.

A qcuick test with Arduino Uno and SX1278 confirms the behavior that was reported.

Could you please try if the problem persists for messages longer than 64 bytes?

notsleep196, 2022-11-17:
Thank you for reply! Yes, i'll try with messages longer than 64 bytes later this or next week.

notsleep196, 2022-11-30:
Hello!
I changed totalLength value to 208
**Buffer to send (208 bytes):** CF 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF B0 B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF C0 C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF 

So the result is completely the same. FIFO threshold interrupt triggers only once every time i send packet.

jgromes, 2022-12-02:
Could you please provide the exact code you are using now?