jgromes/RadioLib #603: ESP32+SX1280  send/receive

nerodu, 2022-11-09:
I test the SX128x Transmit Example with ESP32+SX1280, but always get the error log: 

Initializing ... success!
[2022-11-09_17:41:35:597][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:36:707][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:37:848][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:38:993][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:40:137][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:41:288][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:42:420][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:43:558][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:44:705][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:45:855][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:46:989][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:48:143][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:49:285][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:50:436][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:51:574][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:52:718][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:53:858][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:54:989][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:56:133][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:57:283][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:58:428][SX1280] Transmitting packet ... failed, code -5
[2022-11-09_17:41:59:564][SX1280] Transmitting packet ... failed, code -5

I just modif the pin in radio = new Module() with mine.... What may cause this problems? Myy PCB is below
[image]

jgromes, 2022-11-09:
Could you please fill in the information required by the[ template](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md)? Especially the debug output.

nerodu, 2022-11-10:
Hi details as below:

    MCU: ESP32
    Wireless module type：SX1280, we design as the semtech datasheet.
    Arduino IDE version: 2.0.1
    Library version: 5.4.1
    Schematic diagram: 
    
[image]

   Examples: SX128x Transmit Example
   Output:
   [2022-11-10_09:21:31:212][SX1280] Initializing ... 	SX128x
[2022-11-10_09:21:32:286]CMD	80	
[2022-11-10_09:21:32:286]DATW	0	43	

[2022-11-10_09:21:32:318]CMD	C0	
[2022-11-10_09:21:32:318]DATR	0	40	0	43	

[2022-11-10_09:21:32:318]CMD	80	
[2022-11-10_09:21:32:349]DATW	0	43	

[2022-11-10_09:21:32:349]CMD	C0	
[2022-11-10_09:21:32:349]DATR	0	40	0	43	

[2022-11-10_09:21:32:381]CMD	8F	
[2022-11-10_09:21:32:381]DATW	0	43	0	43	

[2022-11-10_09:21:32:413]CMD	C0	
[2022-11-10_09:21:32:413]DATR	0	40	0	43	

[2022-11-10_09:21:32:444]CMD	8A	
[2022-11-10_09:21:32:444]DATW	1	43	

[2022-11-10_09:21:32:444]CMD	C0	
[2022-11-10_09:21:32:479]DATR	0	40	0	43	

[2022-11-10_09:21:32:479]CMD	88	
[2022-11-10_09:21:32:513]DATW	60	43	

[2022-11-10_09:21:32:513]CMD	C0	
[2022-11-10_09:21:32:513]DATR	0	40	0	43	

[2022-11-10_09:21:32:545]CMD	96	
[2022-11-10_09:21:32:545]DATW	1	43	

[2022-11-10_09:21:32:576]CMD	C0	
[2022-11-10_09:21:32:576]DATR	0	40	0	43	
[2022-11-10_09:21:32:609]
[2022-11-10_09:21:32:609]CMD	86	
[2022-11-10_09:21:32:609]DATW	B8	43	9D	43	89	43	

[2022-11-10_09:21:32:644]CMD	C0	
[2022-11-10_09:21:32:644]DATR	0	40	0	43	
[2022-11-10_09:21:32:676]
[2022-11-10_09:21:32:676]CMD	3	
[2022-11-10_09:21:32:676]DATR	0	43	0	1	

[2022-11-10_09:21:32:676]CMD	C0	
[2022-11-10_09:21:32:708]DATR	0	40	0	43	

[2022-11-10_09:21:32:708]CMD	8B	
[2022-11-10_09:21:32:740]DATW	90	43	18	43	3	43	

[2022-11-10_09:21:32:740]CMD	C0	
[2022-11-10_09:21:32:772]DATR	0	40	0	43	

[2022-11-10_09:21:32:772]CMD	3	
[2022-11-10_09:21:32:772]DATR	0	43	0	1	

[... 673 lines cut ...]
[2022-11-10_09:21:47:397]CMD	97	
[2022-11-10_09:21:47:397]DATW	FF	43	FF	43	

[2022-11-10_09:21:47:397]CMD	C0	
[2022-11-10_09:21:47:429]DATR	0	40	0	43	

[2022-11-10_09:21:47:429]CMD	83	
[2022-11-10_09:21:47:461]DATW	0	43	0	43	0	43	

[2022-11-10_09:21:47:461]CMD	C0	
[2022-11-10_09:21:47:493]DATR	0	C0	0	C3	

[2022-11-10_09:21:47:589]CMD	97	
[2022-11-10_09:21:47:589]DATW	FF	43	FF	43	

[2022-11-10_09:21:47:589]CMD	C0	
[2022-11-10_09:21:47:622]DATR	0	40	0	43	

[2022-11-10_09:21:47:622]CMD	80	
[2022-11-10_09:21:47:654]DATW	0	43	

[2022-11-10_09:21:47:654]CMD	C0	
[2022-11-10_09:21:47:654]DATR	0	40	0	43	

[2022-11-10_09:21:47:687]failed, code -5

jgromes, 2022-11-11:
I tried to repdoruce this with SX1280 and Arduino Pro Mini and it seems to be working. The debug output looks good, are you sure about your wiring? It looks like the ESP32 does not receive the "Tx done" signal from DIO1.

nerodu, 2022-11-17:
I use DIO2 as the interrupt pin. And according to the datasheet, DIO1, DIO2, DIO3 all can be used for the interrupt. So I use like this:  SX1280 radio = new Module(SS, DIO2, NRST, BUSY); Is this way right?

nerodu, 2022-11-17:
After move the interrupt from DIO2 tp DIO1, it works now..... Maybe it is good for user to define the interupt in code. Just a suggestion. Thanks for support.

closed 2022-11-17 by nerodu

jgromes, 2022-11-17:
> So I use like this: SX1280 radio = new Module(SS, DIO2, NRST, BUSY); Is this way right

The library asked for connection to DIO1, and you connected it to DIO2, to me it seems quite obvious that is not going to work.