jgromes/RadioLib #620: LLCC68.h TCXO wrong default value

tripla-m, 2022-11-23:
**Describe the bug**
Using the default values ​​for the LLCC68 chip returns code error -707 when the radio module is booted up. This is caused by an
incorrect TCXO value on [LLCC68.h](https://github.com/jgromes/RadioLib/blob/master/src/modules/LLCC68/LLCC68.h) file.
I've seen other users complain about this, for example here: #353 

**To Reproduce**
For reproduce the problem i used SX126x_Trasmit example with only one change
Substitute this
```C++
SX1262 radio = new Module(10, 2, 3, 9);
```
with this
```C++
LLCC68 radio = new Module(10, 2, 3, 9);
```
When I upload the code to the board (Arduino DUE), I read this from the serial monitor: `[SX1262] Initializing ... failed, code -707
`

**Expected behavior**
If the default parameters are used 
```C++
int state = radio.begin();
``` 
I expect everything to work correctly.

There are two ways to fix this:
1) set TCXO = 0, like this:
```C++
...

#define FREQ 434
#define BDWH 125
#define SPFC 9
#define CRTD 7
#define SCWD RADIOLIB_SX126X_SYNC_WORD_PRIVATE
#define POWR 10
#define PRLH 8
#define TCXO 0                //TCXO, default 1.6
#define RLDO false

....

int state = radio.begin(FREQ, BDWH, SPFC, CRTD, SCWD, POWR, PRLH, TCXO, RLDO);

...
```

2) change the default tcxoVoltage value on [LLCC68.h file](https://github.com/jgromes/RadioLib/blob/master/src/modules/LLCC68/LLCC68.h) from this (row 46, look directly `float tcxoVoltage = 1.6`):
```C++
int16_t begin(float freq = 434.0, float bw = 125.0, uint8_t sf = 9, uint8_t cr = 7, uint8_t syncWord = RADIOLIB_SX126X_SYNC_WORD_PRIVATE, int8_t power = 10, uint16_t preambleLength = 8, float tcxoVoltage = 1.6, bool useRegulatorLDO = false);
```
to this (`float tcxoVoltage = 0`):
```C++
int16_t begin(float freq = 434.0, float bw = 125.0, uint8_t sf = 9, uint8_t cr = 7, uint8_t syncWord = RADIOLIB_SX126X_SYNC_WORD_PRIVATE, int8_t power = 10, uint16_t preambleLength = 8, float tcxoVoltage = 0, bool useRegulatorLDO = false);
```
This modification allows you to use radio.begin(); without setting TCXO = 0

**Additional info:**
 - MCU: Arduino DUE
 - Wireless module type: Ebyte E220-400M30S (LLCC68)
 - Arduino IDE version  1.8.19
 - Library version 5.5.0
 - I could not verify if the change I applied causes further errors in radiolib

jgromes, 2022-11-23:
The presence of TCXO or XTAL is a property of your Ebyte module, not of LLCC68 chip inside it. There are other manufacturers that have an LLCC68-based module (e.g. NiceRF, RadioControlli, and maybe others), so unless it can be shown that XTAL-based modules are the most common ones, I do not see a need to change the default value.

tripla-m, 2022-11-23:
Sorry, I hadn't thought that other modules might have a TCXO crystal. I have to stop working at night.

Thank you.

closed 2022-11-23 by tripla-m

jgromes, 2022-11-23:
No worries, I'm fine with changing the default value, but it needs to be justified. Currently it's 1.6 because it seems that most SX16x modules use TCXO. But if that stops being the case, it can be adjusted.