jgromes/RadioLib #625: Enable GFSK on SI4432

Dantali0n, 2022-12-01:
**Is your feature request related to a problem? Please describe.**
The GFSK feature in the SI4432 is present all be it poorly documented. It has a fixed BT of 0.5 hence you can't find how to configure it from the documentation. But having to manually set the registers to configure it is a bit tedious.

**Describe the solution you'd like**
Expose regular methods and parameters to enable GFSK as opposed to FSK with similar interfaces as newer chips supported by RadioLib.

**Describe alternatives you've considered**
Manually setting the registers

**Additional context**
The GFSK shaper does not work in directmode so ideally the methods exposed by RadioLIB should account for this, potentially disabling GFSK in direct mode as it leads to undefined behavior or it might outright return an error. If so this error should be caught and exposed.

All I had to do to manually enable it is use:
```c
...->SPIsetRegValue(RADIOLIB_SI443X_REG_MODULATION_MODE_CONTROL_2, RADIOLIB_SI443X_MODULATION_GFSK, 1, 0);
```

Below I provide to image of spectrum both from an SI4432 where it is clearly visible GFSK shaping is working
First is FSK, second GFSK

[image]
g

[image]

jgromes, 2022-12-01:
I think this can be implemented easily by extending the existing `setDataShaping()` method, I will take a look at it once I have a bit more time.

jgromes, 2022-12-04:
GFSK support was added, you can now use `setDataShaping(RADIOLIB_SHAPING_0_5)` to enable GFSK.

closed 2022-12-04 by jgromes