jgromes/RadioLib #630: SPI_CMD_INVALID on SetTx() for SX1262

X-yl, 2022-12-01:
Hey, I'm unable to use the [SX126x_Transmit.ino](https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Transmit/SX126x_Transmit.ino) sketch. It all goes well until I try to actually transmit something, at which point I get a `RADIOLIB_ERR_SPI_CMD_INVALID`

Here's my code:

```c++
#include <RadioLib.h>

// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  9
// BUSY pin:  8

SX1262 radio = new Module(10, 2, 9, 8);

void setup() {
  Serial.begin(9600);
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(868.0, 125.0, 9, 7, 0x1424, 10, 8, 0, false);
  radio.setDio2AsRfSwitch(false);
  
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));
  int state = radio.transmit("Hello World!");

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));

    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    Serial.println(F("timeout!"));

  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  while (1);
}
```

This is the debug log

<details>
  <summary>Click me</summary>
  
```
[SX1262] Initializing ... M	SX126x
CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8F	
DATW	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8A	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	93	
DATW	20	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	2	
DATW	43	A2	FF	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0A2	

CMD	C0	
DATR	0	22	0	22	

CMD	89	
DATW	7F	A2	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

Symbol length: 1.02 ms
CMD	8B	
DATW	9	A2	6	A2	3	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	D	7	40	
DATW	24	A2	44	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	7	36	
DATR	0	A2	0	D	

CMD	C0	
DATR	0	22	0	22	

CMD	D	7	36	
DATW	D	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	D	8	E7	
DATW	18	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	9D	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	96	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

Symbol length: 1.02 ms
CMD	8B	
DATW	9	A2	6	A2	3	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

Symbol length: 4.10 ms
CMD	8B	
DATW	9	A2	4	A2	3	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	98	
DATW	D7	A2	DB	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	86	
DATW	36	A2	40	A2	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	8	E7	
DATR	0	A2	0	18	

CMD	C0	
DATR	0	22	0	22	

CMD	95	
DATW	4	A2	7	A2	0	A2	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8E	
DATW	A	A2	4	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	D	8	E7	
DATW	18	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	8	D8	
DATR	0	A2	0	E8	

CMD	C0	
DATR	0	22	0	22	

CMD	D	8	D8	
DATW	FE	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	9D	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

success!
[SX1262] Transmitting packet ... CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

Timeout in 253440 us
CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	7	36	
DATR	0	A2	0	D	

CMD	C0	
DATR	0	22	0	22	

CMD	D	7	36	
DATW	D	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8C	
DATW	0	A2	8	A2	0	A2	C	A2	1	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8	
DATW	2	A2	1	A2	0	A2	1	A2	0	A2	0	A2	0	A2	0A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8F	
DATW	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	E	0	
DATW	48	A2	65	A2	6C	A2	6C	A2	6F	A2	20	A2	57	A2	6F	A2	72	A2	6C	A2	64	A2	21	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	2	
DATW	43	A2	FF	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	8	89	
DATR	0	A2	0	4	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	D	8	89	
DATW	4	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	83	
DATW	0	A2	0	A2	0	A2	

CMD	C0	
DATR	0	B8	0	0	

failed, code -706
```

</details>

I'm not certain about the Timeout message? Right at the end though, we send `83 00 00 00`. Which according to [the datasheet](https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R000000Un7F/yT.fKdAr9ZAo3cJLc4F2cBdUsMftpT2vsOICP7NmvMo), is the correct format for SetTx(). However the radio responds with 0xB8, i.e command processing error.

I'm not really sure what could be wrong here so help would be greatly appreciated.

jgromes, 2022-12-01:
I appreciate the debug log, but I'm missing the other information from [issue template](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md) namely what sort of module are you using, and what is your platform? Arduino, ESP32, ...?

X-yl, 2022-12-01:
Apologies, here you go:

  MCU: Arduino Uno
  Link to Arduino core: https://github.com/arduino/ArduinoCore-avr
  Wireless module type: SX1262
  Arduino IDE version: 2.0.2
  Library version: 5.5.0

jgromes, 2022-12-01:
And what sort of module are you using? Is it something from Ebyte like E22? Or Dorji DRF1262? etc.

X-yl, 2022-12-07:
Hey thanks for being so prompt (.. and sorry for the delay getting back to you). It was the RF Solutions LAMBDA62. I had a quick search for LAMBDA62 and came across issue #153, which kindly explains what I did wrong: didn't actually set the RF switch to enable TX (+ many other helpful tips).

I'm closing this as solved. Thanks for the help!

closed 2022-12-07 by X-yl