jgromes/RadioLib #637: SX127x::getPacketLength() in LoRa mode for SX1276/77/78/79

IhorNehrutsa, 2022-12-07:
https://github.com/jgromes/RadioLib/blob/b57e7275fd2e939c4884340d04b4e000d9d870f7/src/modules/SX127x/SX127x.cpp#L1111-L1137

I can not find any difference for Spreading factor SF6 opposite to SF7 - SF12
Address: 0x13 Register Name: RegRxNbBytes  Description:Number of received bytes

See in:
https://semtech.my.salesforce.com/sfc/p/#E0000000JelG/a/2R0000001Rbr/6EfVZUorrpoKFfvaF_Fkpgp5kzjiNyiAbqcpqh9qSjE

It look like right code is:
```
 if(modem == RADIOLIB_SX127X_LORA) { 
   return(_mod->SPIreadRegister(RADIOLIB_SX127X_REG_RX_NB_BYTES)); 
  
 } else if(modem == RADIOLIB_SX127X_FSK_OOK) { 
```

jgromes, 2022-12-07:
In SF6, the only possible header mode is implicit header, in which case the receiver should have no way of knowing the packet length without the user specifying it. Have you tested this change works?

IhorNehrutsa, 2022-12-07:
_packetLength is 0, not setted

closed 2022-12-07 by IhorNehrutsa