jgromes/RadioLib #639: [SX1276] E19-915M30S with Teensy 4.1 not hopping

nealmh, 2022-12-08:
Hello,
I'm attempting to get the basic FHSS transmit working with the E19-915M30S (a Sparfun SparkX SPX1276 board, to be specific) with a Teensy 4.1 controller. Currently the "hopping" never appears to occur, with the debug output showing 0 hops completed and transmission finished.  Based on the test message and radio settings applied I expect there to be 35 hops per packet. I've altered the SX127x_Transmit_FHSS.ino supplied as shown at the end of this post. All code is being complied with PlatformIO.

No errors are generated, and enabling the RadioLib debug (uncommenting #define RADIOLIB_DEBUG in BuildOpt.h) does not provide any real information. Attaching LEDs to DIO0 and DIO1 on the module shows DIO0 blinking (tx complete) but, DIO1 remains off (FHSS hop does not latch high). The first few lines of the serial output are shown below for reference. I have tried with two different radio radios and two different Teensy 4.1s with the same behavior.

Any guidance on finding the issue would be greatly appreciated. Thanks for your time.

```
[SX1276] Initial Settings
--------------------------------------------------------------------------------
RadioLib version is 5.3.0.0
RadioLib MCU Platform is Teensy
Bandwith = 125.00
Frequency Range = 902.00MHz to 928.00MHz
Number of Channels = 6
Max Dwell Time = 400msec
Sync Word = 18
Spread Factor = 9
Coding Rate = 6
Preamble Length = 8
M       SX127x
Symbol length: 0.01 ms
Symbol length: 4.10 ms
[SX1276] Initializing Success!
Symbol Time = 4.10msec
Hopping Period = 97
[SX1276] Hopping period set!
Message Air Time = 1312msec
Expected Number of Hops = 35
[SX1276] Sending first packet ... transmission finished!
[SX1276] Radio is on channel: 0
[SX1276] Hops completed: 0
[SX1276] Sending another packet ... transmission finished!
[SX1276] Radio is on channel: 0
[SX1276] Hops completed: 0
[SX1276] Sending another packet ... transmission finished!
[SX1276] Radio is on channel: 0
[SX1276] Hops completed: 0
```

```
/*
   RadioLib SX127x Transmit with Frequency Hopping Example
   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)
   Other modules from SX127x/RFM9x family can also be used.
   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem
   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
   SX127x supports FHSS or Frequency Hopping Spread Spectrum.
   Once a hopping period is set and a transmission is started, the radio
   will begin triggering interrupts every hop period where the radio frequency
   is changed to the next channel.
*/

#include <RadioLib.h>

const float BANDWIDTH = 125.0;
const float FREQUENCY_MIN = 902.0;
const float FREQUENCY_MAX = 928.0;
const uint8_t NUMBER_OF_CHANNELS = 6;
const uint16_t MAX_DWELL_TIME = 400; 
const uint8_t RADIO_SYNC_WORD = 0x12; 
#define MAX_PACKET_SIZE 255 //Limited by SX127x 
const uint8_t SPREAD_FACTOR = 9; //6 to 12. Use higher factor for longer range.
const uint8_t CODING_RATE = 6; //5 to 8. 6 was chosen to allow higher spread factor. Higher coding rates ensure less packets dropped.
[... 181 lines cut ...]
    Serial.print(F("[SX1276] Sending another packet ... "));
    String packet = longPacket + packetCounter;
    transmissionState = radio.startTransmit(packet);
  }

  // check if we need to do another frequency hop
  if (fhssChangeFlag == true) {
    // we do, change it now
    int state = radio.setFrequency(channels[radio.getFHSSChannel() % numberOfChannels]);
    if (state != RADIOLIB_ERR_NONE) {
      Serial.print(F("[SX1276] Failed to change frequency, code "));
      Serial.println(state);
    }

    // increment the counter
    hopsCompleted++;

    // clear the FHSS interrupt
    radio.clearFHSSInt();

    // we're ready to do another hop, clear the flag
    fhssChangeFlag = false;
  }
}
```

nealmh, 2022-12-08:
I've done some further digging, and I may have found the answer. The defines for RADIOLIB_SX127X_DIO1_LORA* appear to be  incorrect in SX127x.h. The current values need to be right shifted by two places. So, currently they are 
```
#define RADIOLIB_SX127X_DIO1_LORA_RX_TIMEOUT                   0b00000000  //  5     4
#define RADIOLIB_SX127X_DIO1_LORA_FHSS_CHANGE_CHANNEL          0b01000000  //  5     4
#define RADIOLIB_SX127X_DIO1_LORA_CAD_DETECTED                 0b10000000  //  5     4
```

This should be updated to 

```
#define RADIOLIB_SX127X_DIO1_LORA_RX_TIMEOUT                   0b00000000  //  5     4
#define RADIOLIB_SX127X_DIO1_LORA_FHSS_CHANGE_CHANNEL          0b00010000  //  5     4
#define RADIOLIB_SX127X_DIO1_LORA_CAD_DETECTED                 0b00100000  //  5     4
```

The adjacent defines look to be correct. I would like to confirm that this is the appropriate place to fix this issue, though.

nealmh, 2022-12-09:
And, I'm going to close this out because my project was pegged to an old version and this problem has been fixed in new versions of the library. Sorry for that, file this one under PEBKEC.

closed 2022-12-09 by nealmh

jgromes, 2022-12-09:
Glad you were able to solve it ;)