jgromes/RadioLib #641: POCSAG - no decodes/reception with pager_receive example

ManoDaSilva, 2022-12-12:
Hi! 

I've been playing around with the lib, and getting the transmit part to work quite effectively with a variety of pagers, on VHF and UHF. Just started using it for reception, but for some reason, I can't get any message to "pass" through.

Here's my setup:

- Transmitter: MMDVM hotspot on DAPNET mode (Rpi Zero + Zumspot), 100mW, tested working with a multitude of pagers. Frequency has been adjusted to a few 10s of Hz (against a TCXO SDR itself referenced against a GPS-disciplined base station)
- Receiver: LORA32-TTGO Hotspot, running the latest radiolib lib, frequency corrected as well. Previously tested and working on Tx. 

I'm manually "keying" the POCSAG transmitter to send an alpha page to address 100, and using the page_receive example (adapted to my frequency/offset/pinout):
`
float offset = 0.0044; //in MHz. VHF: 0.0014 UHF: 0.0044
  float frequency = 439.98750;
  state = pager.begin(frequency+offset, 1200);
`
and 

`  state = pager.startReceive(pin, 100);
`

I might be missing something out, but would that be related to the polarity invert pin issue we had previously?

Cheers!
Manoel

jgromes, 2022-12-12:
Please share the code, just so that it's posted for reference.

I tried reception on T-Beam, although just when transmitting from RadioLib, not from actual pagers. One thing that might be woth trying is to disable address checking, i.e. setting mask to 0 in `startReceive`.

ManoDaSilva, 2022-12-12:
Mask to 0, do you mean something like this? 

`state = pager.startReceive(pin, 100);`

```
// include the library
#include <RadioLib.h>

#define LORA_SCK        5
#define LORA_MISO       19
#define LORA_MOSI       27
#define LORA_SS         18
#define LORA_DIO0       26
#define LORA_DIO1       33
#define LORA_DIO2       32
#define LORA_RST        23

SX1278 radio = new Module(LORA_SS, LORA_DIO0, LORA_RST, LORA_DIO1);
const int pin = 32;

// create Pager client instance using the FSK module
PagerClient pager(&radio);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;
void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // initialize Pager client
  Serial.print(F("[Pager] Initializing ... "));
  float offset = 0.0044; // device specific, in MHz. VHF: 0.0014 UHF: 0.0044
  float frequency = 439.98750;
  state = pager.begin(frequency+offset, 1200);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // start receiving POCSAG messages
  Serial.print(F("[Pager] Starting to listen ... "));
  // address of this "pager":     1234567
  state = pager.startReceive(pin, 100);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

}

void loop() {
  // the number of batches to wait for
  // 2 batches will usually be enough to fit short and medium messages
  if (pager.available() >= 2) {
    Serial.print(F("[Pager] Received pager data, decoding ... "));

    // you can read the data as an Arduino String
    String str;
    int state = pager.readData(str);

    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));

      // print the received data
      Serial.print(F("[Pager] Data:\t"));
      Serial.println(str);

    } else {
      // some error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }
  }
}
```

jgromes, 2022-12-12:
No, I meant this: `startReceive(pin, 100, 0);`

startReceive has optional third argument, which lets you set the bits of address thet you want to check. So if you set it to 0, it will not check any address bits, acting essentially as a sniffer.

ManoDaSilva, 2022-12-12:
Still no decoding. I'll chuck a scope on the DIO2 pin to make sure the hardware is operating properly :)

jgromes, 2022-12-12:
Looking at the pinout, maybe check the DIO1 and DIO2 pins? They don't seem to be exposed on LORA32-TTGO.

ManoDaSilva, 2022-12-12:
Indeed! Wired DIO2 to GPIO34, and I can confirm data is being received by the SX127X. So there's something going on with the ESP32, I'll dig a bit further :)

ManoDaSilva, 2022-12-12:
yep, phase inversion issues. When I change the phase on the transmitter's side, it decodes successfully :)

jgromes, 2022-12-12:
Well, in that case it's a pretty easy fix, just enable inversion in the begin method ;)

ManoDaSilva, 2022-12-12:
Sorry, tried it, and it doesn't help, inversion seems to only affect TX, not RX

jgromes, 2022-12-12:
Ah, you're right - it's not implemented for reception. That's something I should probably fix ...

jgromes, 2022-12-12:
OK, should be fixed now. Can you try?

ManoDaSilva, 2022-12-12:
Just tested, still not decoding.... 

I tried a veeeeeery ugly hack on /modules/SX127X/SX127X.cpp, in the function readBit:
`  updateDirectBuffer((uint8_t)!digitalRead(pin));`
That works, albeit being really ugly.

ManoDaSilva, 2022-12-15:
Yes, that did it, cheers! 
Perhaps I'd invert the overall logic in Rx, as when I initialize pager.begin, by default Rx is inverted whereas Tx isn't (compared against 2 pagers and one transmitter where none are inverted).

I hope I'm not too confusing :)

ManoDaSilva, 2022-12-15:
Okay, took a step back, and proceeded as follows:
-Test transmitter, **non**-inverted
-Test pager,  **non**-inverted

-ESP32-TTGO, Radiolib, transmits a message to the test pager at boot time, then switches to Rx and waits for messages from the test transmitter

All of them accurate within 100Hz.

If the lib is initialized as such:
`pager.begin(frequency + offset, 1200,false);`
The test pager doesn't receive the message, and the TTGO doesn't receive anything from the test transmitter

If I invert with:
`pager.begin(frequency + offset, 1200,true);`
The test pager recejves the message, and the TTGO receives everything from the test transmitter

So that means that both RX and TX's logic must be inverted (inv ==> !inv and vice-versa)

Cheers!

jgromes, 2022-12-22:
You're right, although it took me a while to realize why - it's because the modeuls like SX127x assume high frequency to be 1 and low one to be 0, which is the opposite of the POCSAG specification.

It should be fixed now.

closed 2022-12-22 by jgromes