jgromes/RadioLib #643: Raspberry Pi Pico compilation error

orhanyor, 2022-12-13:
I tried to compile [SX128x_Transmit_Interrupt.ino] example but having the following error.

c:\Users\MainPC\Documents\Arduino\libraries\RadioLib\src\Module.cpp:7:10: fatal error: mbed.h: No such file or directory
    7 | #include "mbed.h"
      |          ^~~~~~~~
compilation terminated.

I'm using the latest version of arduino and this library.
As for the core i'm using this repo --> https://github.com/earlephilhower/arduino-pico
Is it possible to get around this problem or am i doing something wrong? Thank you

jgromes, 2022-12-13:
RadioLib supports the official Arduino mbed core, which also contains support for RP2040: https://github.com/arduino/ArduinoCore-mbed

orhanyor, 2022-12-13:
> RadioLib supports the official Arduino mbed core, which also contains support for RP2040: https://github.com/arduino/ArduinoCore-mbed

Yes, i tried it after i posted. Too bad official one is not as good and lacks features and flexibility that are present in earlephilhower's version. Probably thats why it's more popular than the official one.

orhanyor, 2022-12-13:
I think problem arises #define RADIOLIB_MBED_TONE_OVERRIDE in BuildOpt.h is defined which later on calls for mbed.h file. I just commented #define RADIOLIB_MBED_TONE_OVERRIDE under #elif defined(ARDUINO_ARCH_RP2040)
and code compiled without an error but unfortunately my sx1280 modules are still on the way so i couldnt check if the code works that way.

jgromes, 2022-12-14:
Unfortunately that overrid is needed for the mbed core. Better approach would be to add support for the unofficial RP2040 core, similarly to how it is done for STM32.

orhanyor, 2022-12-14:
Ok so I tried adding a flag 
#elif defined(ARDUINO_RASPBERRY_PI_PICO) just before #elif defined(ARDUINO_ARCH_RP2040) and copied the contents to it except these 2 lines 
#define RADIOLIB_TONE_UNSUPPORTED
#define RADIOLIB_MBED_TONE_OVERRIDE

code compiled fine but I need the modules to see if its working.

jgromes, 2022-12-14:
It's a little bit more complicated than that, however, I added support for the "unofficial" core, and CI is now crunching all the examples. Everything should work unless the platform has some weird behavior at the low level of e.g. SPI transactions.

closed 2022-12-14 by jgromes

orhanyor, 2022-12-14:
Thank you very much, appreciate your work. Most of the libraries have no problem with it so i guess it should be ok. I will test it as soon as i get the modules.