jgromes/RadioLib #645: esp8266 llcc68 chips send data was out of time, and get error response -706

chcmq, 2022-12-15:
code
===

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "lora_service.h"
#include "config.h"
#include <RadioLib.h>
// SX1262 has the following connections:
// NSS pin:   15
// DIO1 pin:  4
// NRST pin:  10
// BUSY pin:  5
LLCC68 radio = new Module(15, 4, 10, 5);

lora_service::lora_service() {}
lora_service::~lora_service() {}

/* =============================icache functions========================= */
void ICACHE_FLASH_ATTR lora_service::setup() {
// initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin(433.0, 125.0, 9, 7, 0x12, 17, 8, 0, false);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
     // Some SX126x modules use DIO2 as RF switch. To enable
  // this feature, the following method can be used.
  // NOTE: As long as DIO2 is configured to control RF switch,
  //       it can't be used as interrupt pin!
  if (radio.setDio2AsRfSwitch() != RADIOLIB_ERR_NONE) {
    Serial.println(F("Failed to set DIO2 as RF switch!"));
    while (true);
  }

  Serial.println(F("All settings succesfully changed!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
   
  Serial.println("begin LoRa radio init success");

}// end io setup

void ICACHE_FLASH_ATTR lora_service::loop()
{

  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // Don't load RF module with continous transmit
  delay(3000);
  
 
}//end loop

/* ===========================end iram functions=========================== */
lora_service lora;

```
Debug

```
DATW    48      A2      65      A2      6C      A2      6C      A2      6F      A2  20       A2      57      A2      6F      A2      72      A2      6C      A2      64  A2       21      A2

CMD     C0
DATR    0       22      0       22

CMD     2
DATW    43      A2      FF      A2

CMD     C0
DATR    0       22      0       22

CMD     1D      8       89
DATR    0       A2      0       4

CMD     C0
DATR    0       22      0       22

CMD     11
DATR    0       A2      0       0

CMD     C0
DATR    0       22      0       22

CMD     D       8       89
DATW    4       A2

CMD     C0
DATR    0       22      0       22

CMD     83
DATW    0       A2      0       A2      0       A2

CMD     C0
DATR    0       B8      0       0

failed, code -706
```

jgromes, 2022-12-15:
1. Issue title is supposed to sum up the problem, it's not the first paragraph.
2. Please follow the "Module not working" issue template - that includes a **reproducible, minimum** code example - what you have posted is neither reproducible, nor minimal. Start with the examples.
3. You have not posted the entire debug log, but it seems to me that the failure occurs just as the LLCC68 is about to start transmitting. That may be caused by inappropriate power source - and that's exactly the reason why the issue template is asking for a hardware description.

jgromes, 2022-12-16:
2.7 V seems very low - I'm guessing you're powering it from the ESP8266 onboard LDO regulator, it should be 3.3V. If it's dropping by 0.6V in standby, I think it's safe to say it is not going to be able to handle the power needed for transmission. You should use an external power source for the LLCC68, e.g. a dedicated LDO.

jgromes, 2022-12-30:
Closed due to inactivity.

closed 2022-12-30 by jgromes