jgromes/RadioLib #650: SX1262.begin() fails after ESP32 deep sleep wakeup

spleenware, 2022-12-21:
**Describe the bug**
I have successfully used the SX1278 RadioLib class in an ESP32 project, where the DIO0 (packet ready) pin wakes up the ESP32 from deep sleep. The SX1278.begin() seems to work fine when the processor is waking up, and effectively *resuming* (ie. where esp_reset_reason() == ESP_RST_DEEPSLEEP)

However, when using a different dev board, which has an SX1262 LoRa module,  the same resuming from deep sleep (but with the DIO1 pin being packet ready), the SX1262.begin() returns status of -2 (chipset not found)

**To Reproduce**
My sketch is way too big to paste.

**Expected behavior**
Expect the .begin() to return 0 (ERR_NONE) when re-initialising the SX1262 radio.

**Additional info (please complete):**
 - MCU: ESP32-S3
 - Wireless module type: SX1262
 - IDE version : PlatformIO
 - Library version: 4.4.2

jgromes, 2022-12-21:
> My sketch is way too big to paste.

Then your sketch is way too big to debug. Try to create a minimum example, from your description it should be quite straightforward - just intitializing the SX126x, putting it to receive mode and then going to deep sleep with wakeup on DIO1. Still, Also, please include verbose debug output.

spleenware, 2022-12-22:
OK, have two sketches that reproduce the problem:

modified SX126x_Transmit.ino:
-----------
// include the library
#include <RadioLib.h>

// Heltec V3 pins
#define  P_LORA_DIO_1  14    // DIO1
#define  P_LORA_NSS     8
#define  P_LORA_RESET  12
#define  P_LORA_BUSY   13

#define  P_LORA_SCLK    9
#define  P_LORA_MISO   11
#define  P_LORA_MOSI   10

SPIClass v3_spi;

SX1262 *radio;

void setup() {
  Serial.begin(9600);

  v3_spi.begin(P_LORA_SCLK, P_LORA_MISO, P_LORA_MOSI, P_LORA_NSS);
  radio = new SX1262(new Module(P_LORA_NSS, P_LORA_DIO_1, P_LORA_RESET, P_LORA_BUSY, v3_spi));

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio->begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */

  pinMode(LED, OUTPUT);
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  digitalWrite(LED, HIGH);
  int state = radio->transmit("Hello World!");
  digitalWrite(LED, LOW);

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio->getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for two seconds before transmitting again
  delay(2000);
}

spleenware, 2022-12-22:
OK, here's a modified SX126x_Transmit sketch:
`// include the library
#include <RadioLib.h>

// Heltec V3 pins
#define  P_LORA_DIO_1  14    // DIO1
#define  P_LORA_NSS     8
#define  P_LORA_RESET  12
#define  P_LORA_BUSY   13

#define  P_LORA_SCLK    9
#define  P_LORA_MISO   11
#define  P_LORA_MOSI   10

SPIClass v3_spi;

SX1262 *radio;

void setup() {
  Serial.begin(9600);

  v3_spi.begin(P_LORA_SCLK, P_LORA_MISO, P_LORA_MOSI, P_LORA_NSS);
  radio = new SX1262(new Module(P_LORA_NSS, P_LORA_DIO_1, P_LORA_RESET, P_LORA_BUSY, v3_spi));

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio->begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */

  pinMode(LED, OUTPUT);
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  digitalWrite(LED, HIGH);
  int state = radio->transmit("Hello World!");
  digitalWrite(LED, LOW);

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio->getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for two seconds before transmitting again
  delay(2000);
}`

spleenware, 2022-12-22:
OK, here are two sketches which reproduce the problem, modified from the SX126x_Transmit and _Receive_Interrupt examples.
[sx1262.zip](https://github.com/jgromes/RadioLib/files/10282945/sx1262.zip)

spleenware, 2022-12-22:
BTW, I just tried these two sketches with the latest RadioLib (5.5.0) and still get same behaviour.

jgromes, 2022-12-22:
I'm still missing the debug mode output.

Looking at the Rx sketch, the one thing that strikes me as odd is that the interrupt service routine `setFlag` is missing the `ICACHE_RAM_ATTR` attribute, which is required on ESP32. Triggering ISR wihtout this attribute could cause the ESP32 to halt or crash. But the better question is - why do you have `setFlag` in the first place? You aren't using the flag for anything. I would suggest removing `setFlag` and the call to `setDio1Action`.

spleenware, 2022-12-23:
OK, here is the verbose debug output. (Compiled against latest lib ver 5.5.0)
Plus I commented out the ->setDio1Action() as it's not needed.
[verbose_debug_log.txt.zip](https://github.com/jgromes/RadioLib/files/10293452/verbose_debug_log.txt.zip)

jgromes, 2022-12-23:
Thanks, it looks like after the reset, the SPI no longer returns any data, which is strange, but I don't think it has anything to do with the library.

One thing I realized looking at your code is that you are calling the begin method after waking up - this will reset the SX1262, deleting any data it may have received. So you should make sure the begin method runs only once.

jgromes, 2023-01-08:
Closed due to inactivity.

closed 2023-01-08 by jgromes