jgromes/RadioLib #654: SX126x Class Reference states incorrect step size for setCurrentLimit API

MrSniffer, 2023-01-03:
In two places, the SX126x Class Reference says the setCurrentLimit step size is 0.25mA; it should state 2.5mA step size.

jgromes, 2023-01-03:
Thanks for being so attentive! I found one of the instances (in SX126x.h), where is the second one?

MrSniffer, 2023-01-03:
I was referring to the PDF document that describes the APIs. Perhaps both references are built from the .h file?

jgromes, 2023-01-04:
I'm not sure which PDF you're referring to, there's the API reference on GitHub pages (https://jgromes.github.io/RadioLib/), which is built from the .h files by Doxygen.

MrSniffer, 2023-01-04:
My bad; I was referring to my own PDF of the GitHub pages.
While you are at it, perhaps you should point out that the values must be >=0 and <= 140 mA.
Thanks for the edit you already made.

jgromes, 2023-01-04:
True, most other configuration methods list the range, so it's added now.

closed 2023-01-04 by jgromes