jgromes/RadioLib #655: Adding support for the RF4463F30 radio module

Liponchik, 2023-01-03:
This is a good module with si4463 chip and support for 4GFSK modulation, 1w big power.  Might be a cool addition to Radiolib.  in the repository https://github.com/Borchevkin/driver_si4463

jgromes, 2023-01-03:
I'm aware of the Si4463, though don't have much experience with it, so if anyone would like to have a major contribution, that would be welcome.

MesutErdemir, 2023-07-27:
Any updates about Si4463 module? Is anyone working on it?

jgromes, 2023-07-28:
@MesutErdemir unfortunately I'm not working on this, I don't think anyone is.

maxsrobotics, 2024-01-07:
This document shows migration from the Si443x to Si446x. This library already support the 3x, so this may be a helpful resource. https://www.silabs.com/documents/public/application-notes/AN799.pdf. This would also support both this issue and issue 920.

maxsrobotics, 2024-01-07:
I'll create a fork of this and fiddle around with the transition. If I get the module experimentally working, I'll create a PR and explain the changes. using AN799 above, I should be able to port the Si443x to Si4x6x. If I get stuck, I'll update here.

-Max

jgromes, 2024-01-07:
@maxsrobotics that's amazing, looking forward to the PR :)

maxsrobotics, 2024-01-10:
I've started changing command addresses, and I'll start with a PR for the basic commands (START_TX, GET_CHIP_STATUS etc.). Then, someone else can pick up for the more advanced commands, or I could do it if I have time.

maxsrobotics, 2024-01-10:
Expect a PR in ~1-2 weeks

MesutErdemir, 2024-01-29:
> Expect a PR in ~1-2 weeks

I am looking forward your PR.
Thanks.

ellisgl, 2024-04-07:
> Expect a PR in ~1-2 weeks

It's been a while, any issues getting the PR together?

maxsrobotics, 2024-04-07:
Hey, sorry, I've pretty severely underestimated how much work would go into this, and m Si4063 breakout board was delayed far. I created a separate driver for the Si, bt it's not integrated into RadioLib. Feel free to use as much as you want, but it's not done. I'll link to it in a bit.

-Max

maxsrobotics, 2024-04-07:
https://github.com/maxsrobotics/si4063_driver/