jgromes/RadioLib #656: Playing with SX1280

lw2dtz, 2023-01-03:
I started working with the SX1280 and an arduino 8 MHZ internal crystal,in 2.4Ghz. The examples work perfect, but when I want to use it with the MORSE example, I have to change the line to:

int state = radio.beginGFSK();

for the IDE 1.8.13 to compile it. Once loaded it reports:

[SX1280] Initializing ... failed, code -705

TIA

Gustavo.

jgromes, 2023-01-04:
You have to use the `beginGFSK()` method since the default `begin()` method initializes the LoRa modem, whereas the GFSK modem is needed for Morse.

To get some more information however, please fill in the rest of the issue template. Mainly the output from [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode) (although the other points such as the wiring are important as well).

lw2dtz, 2023-01-13:
I found the error in the configuration of the pins.
 Runs OK.

[image]

 Thanks.

jgromes, 2023-01-13:
Glad you were able to fix it!

closed 2023-01-13 by lw2dtz