jgromes/RadioLib #658: RFM69HCW and RP2040 (RP2040 disconnecting)

GamendeCat, 2023-01-07:
**Sketch that is causing the module fail**
```c++
paste the sketch here, even if it is an unmodified example code
//
// RASPBERRY PI PICO / GROUND STATION
//
#include <RadioLib.h>

// RF69 has the following connections:
#define RST_PIN  9
#define CS_PIN   17
#define MOSI_PIN 15
#define MISO_PIN 11
#define SCK_PIN  14
#define INT_PIN  10

RF69 radio = new Module(CS_PIN, INT_PIN, RST_PIN); // Module(cs, int, rst, gpio, &spi, spiSettings)

// or using RadioShield
// https://github.com/jgromes/RadioShield
//RF69 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);
  
  // NOTE: some RF69 modules use high power output,
  //       those are usually marked RF69H(C/CW).
  //       To configure RadioLib for these modules,
  //       you must call setOutputPower() with
  //       second argument set to true.

  Serial.print(F("[RF69] Setting high power module ... "));
  int state = radio.setOutputPower(20, true);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[RF69] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to 64 characters long
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 64 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 64 bytes
    Serial.println(F("too long!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

**Hardware setup**
[image]

**Debug mode output**
It just disconnects so I can't really give any debugging error here. I am receiving this windows error "A request for the USB device descriptor failed.". When uploading other programs, I just get a COM7 port where I can look at Serial Monitor. But here after uploading it just disconnects my Raspberry Pi Pico and the light on the pi keeps blinking

**Additional info (please complete):**
 - Using Raspberry Pi Pico RP2040 and RFM69HCW
 - https://github.com/stm32duino/Arduino_Core_STM32
 - Arduino IDE version 2.0.3 (last version)
 - Library version: Last version

GamendeCat, 2023-01-07:
Hello So I managed to make it stop disconnecting by removing: 
```
  // NOTE: some RF69 modules use high power output,
  //       those are usually marked RF69H(C/CW).
  //       To configure RadioLib for these modules,
  //       you must call setOutputPower() with
  //       second argument set to true.

  Serial.print(F("[RF69] Setting high power module ... "));
  int state = radio.setOutputPower(20, true);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```

And just making it default: 

```
  // initialize RF69 with default settings
  Serial.print(F("[RF69] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```

But now I just receive nothing in my Serial Monitor, I'll try to debug in a second

GamendeCat, 2023-01-07:
Oh nevermind, I'm not able to debug on the raspberry pi Pico
[image]

JulesPruvost, 2023-01-08:
Added an delay of 3 seconds and now I receive some output in the serial monitor:
06:36:56.911 -> [RF69] Initializing ... failed, code -2

jgromes, 2023-01-08:
In your first example, you cannot just call `setOutputPower` - first you have to initialize the module with `begin`, only after that you can call other methods. See the wiki and all of the examples.

Next, from the photo, it seems you are powering the RFM69 directly from some on-board power source for the RP2040. Make sure that source is sufficient to power the RFM69 during transmissions. If it is not, I would expect resets/brownouts of the RP2040.

Finally, the code -2 means that the module was not found, which usually signals some problem in the wiring. You did not specify any SPI bus, which means RadioLib will use the default `SPI`. Googling Raspberry Pi Pico pinout, it seems you have connected the module to `SPI1` - I don't know if that is correct, you should double-check.

JulesPruvost, 2023-01-08:
Hello, Thanks for the quick reply. 
Is there any way I could set the SPI pins in the library?

jgromes, 2023-01-08:
See the [Basics wiki page](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup) - you can, but your platform has to support it.

JulesPruvost, 2023-01-08:
Not sure If I can ask this here but so I am using this Arduiuno core : https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/SPI/SPI.cpp. And I have a constructor here
```
arduino::MbedSPI::MbedSPI(int miso, int mosi, int sck) :
  _miso(digitalPinToPinName(miso)), _mosi(digitalPinToPinName(mosi)), _sck(digitalPinToPinName(sck)) {
}
```

But whenever I run my code with this:
```
MBedSPI spi(MOSI_PIN, MISO_PIN, SCK_PIN);
SPISettings spiSettings(2000000, MSBFIRST, SPI_MODE0);
RF69 radio = new Module(CS_PIN, INT_PIN, RST_PIN, spi, spiSettings);  // Module(cs, int, rst, gpio, &spi, spiSettings)
```

I Receive this error: 
```
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:15:1: error: 'MBedSPI' does not name a type
 MBedSPI spi(MOSI_PIN, MISO_PIN, SCK_PIN);
 ^~~~~~~
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:17:51: error: 'spi' was not declared in this scope
 RF69 radio = new Module(CS_PIN, INT_PIN, RST_PIN, spi, spiSettings);  // Module(cs, int, rst, gpio, &spi, spiSettings)
                                                   ^~~
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:17:51: note: suggested alternative: 'spi0'
 RF69 radio = new Module(CS_PIN, INT_PIN, RST_PIN, spi, spiSettings);  // Module(cs, int, rst, gpio, &spi, spiSettings)
                                                   ^~~
                                                   spi0
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino: In function 'void setup()':
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:27:3: error: 'spi' was not declared in this scope
   spi.begin();
   ^~~
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:27:3: note: suggested alternative: 'spi0'
   spi.begin();
   ^~~
   spi0
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:54:5: error: expected primary-expression before '/' token
   * /
     ^
C:\Users\User\Documents\Arduino\pico-file.ino\pico-file.ino.ino:55:1: error: expected primary-expression before '}' token
 }
 ^

exit status 1

Compilation error: 'MBedSPI' does not name a type
```

jgromes, 2023-01-08:
You have to use `SPIClass`, not `MbedSPI` - that's what RadioLib expects. If the Arduino-mbed core doesn't work, RadioLib also supports this one: https://github.com/earlephilhower/arduino-pico

Apparently it's supposed to be better than the official.

JulesPruvost, 2023-01-08:
I've tried that other library before and it seemed to be not working with my pico. Whenever I uploaded code with that core my arduino just disconnected like I said before and I couldn't see anything on my serial monitor because I had no serial monitor so I'd like to keep that for last case scenario.

Now when I reverenced referenced SPIClass with the nmbed core. It pointed me to a class HardwareSPI with a constructor with 0 params. But when I look at the github of the mbed. I can't seem to find this HardwareSPI.h class

jgromes, 2023-01-08:
From what I could tell, HardwareSPI is just an API abstraction, you cannot create an instance of it (notice how all of its methods are pure virtual).

You can try to run the sketch in debug verbose mode (https://github.com/jgromes/RadioLib/wiki/Debug-mode), though I doubt much useful info will be shown.

JulesPruvost, 2023-01-08:
```
20:01:00.336 -> [RF69] Initializing ... R	10	0	
20:01:00.336 -> RF69 not found! (1 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.378 -> R	10	0	
20:01:00.378 -> RF69 not found! (2 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.378 -> R	10	0	
20:01:00.378 -> RF69 not found! (3 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.413 -> R	10	0	
20:01:00.413 -> RF69 not found! (4 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.413 -> R	10	0	
20:01:00.413 -> RF69 not found! (5 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.459 -> R	10	0	
20:01:00.459 -> RF69 not found! (6 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.459 -> R	10	0	
20:01:00.459 -> RF69 not found! (7 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.504 -> R	10	0	
20:01:00.504 -> RF69 not found! (8 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.504 -> R	10	0	
20:01:00.504 -> RF69 not found! (9 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.552 -> R	10	0	
20:01:00.552 -> RF69 not found! (10 of 10 tries) RADIOLIB_RF69_REG_VERSION == 0x0000, expected 0x0024
20:01:00.552 -> No RF69 found!
20:01:00.552 -> failed, code -2
```

JulesPruvost, 2023-01-08:
Hey So I managed to get it working with the SPI bus: but I am getting the error -16 now

JulesPruvost, 2023-01-08:
Hey so I managed to get it working for some reason. I had to use GPx number for pins and not the actual pins on the board. Yeah idk so thanks anyway for the help.

closed 2023-01-08 by JulesPruvost

jgromes, 2023-01-08:
Glad you were able to get it working, thanks for letting me know!