jgromes/RadioLib #662: Support for ASR6601SE SoC (such as LA66 Module)

hahalolha, 2023-01-13:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
N/A. This issue is a request for hardware compatibility.

**Describe the solution you'd like**
Some form of Implementation, unsure.

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or screenshots about the feature request here.

Pardon me for my lack of knowledge in this field. I am a beginner just starting experimentation with LoRa, I stumbled across this module by Dragino named [LA66](http://wiki.dragino.com/xwiki/bin/view/Main/User%20Manual%20for%20LoRaWAN%20End%20Nodes/LA66%20LoRaWAN%20Module/) viewable below.
[image]
I was particularly interested in their [USB Devboard](http://wiki.dragino.com/xwiki/bin/view/Main/User%20Manual%20for%20LoRaWAN%20End%20Nodes/LA66%20USB%20LoRaWAN%20Adapter%20User%20Manual/) which is essentially the previously mentioned LA66 module with an integrated CP2102 USB to TTL and other devboard nice-to-haves.
[image]
The LA66's source files, drivers and more are viewable [here](https://github.com/dragino/LA66) and [here](https://www.dropbox.com/sh/z6daguv1gach8ce/AAAw_FjksNXD91bE3aJOOmnta?dl=0) 

Digging a bit further, I found their FCC listing which included extra info. These listings can be viewed on [fcc.gov](https://gov.fccid.io/ZHZLA66) or [fccid.io](https://fcc.id/ZHZLA66) or on [archive.org](http://web.archive.org/web/20230113111701/https://fccid.io/ZHZLA66).
The main item which peaked my interest was the [internal photos](https://fccid.io/ZHZLA66/Internal-Photos/Internal-Photos-6197734) which showed this module utilises the ASR6601SE SoC from [ASR Microelectronics](http://www.asrmicro.com/en/goods/proinfo/23.html). A datasheet for this product can be found [here](https://www.ebyte.com/pdf-down.aspx?id=2347) and [here](http://web.archive.org/web/20230113112948/https://www.ebyte.com/pdf-down.aspx?id=2347)
A brief excerpt is repeated below:
>  ​ASR6601 is a general LPWAN Wireless Communication SoC, with an integrated RF Transceiver, Modem, and a 32-bit RISC MCU. The MCU uses ARM Cortex M4, with a 48 MHz operating frequency. The RF Transceiver has continuous frequency coverage from 150 MHz to 960 MHz. The Modem supports LoRa modulation for LPWAN use cases and (G)FSK modulation for legacy use cases. The Modem also supports BPSK modulation in TX and (G)MSK modulation in TX and RX. The LPWAN Wireless Communication Module designed with ASR6601 provides ultra-long range and ultra-low-power communication for LPWAN applications. ASR6601 can achieve a high sensitivity to -148 dBm and the maximum transmit power is up to +22 dBm. This makes the chip suitable to be used in long-range LPWAN and has high efficiency. 

A [hackster.io article](https://www.hackster.io/akarsh98/lora-microcontroller-solution-asr6601-tutorial-lorawan-8f2a00) featured this SoC.

I believe this module/SoC is quite powerful, outputting +22 dBm means it is in the higher bracket AFAIK when it comes to LoRa Transceivers and has quite excellent(ish) documentation from the manufacturers'. The only issue I forsee is control over the SoC as it appears to be controlled via AT commands, possibily enforcing restriction upon what may be TX/RX.

[The resaler I was planning to purchase from](https://www.iot-store.com.au/products/la66-usb-lorawan-adapter)

If any links die I have backups I can make available upon request for most documents

Finally on a more personal note (I apologise if this section is a bit far offtopic); Would this be a decent begginner board? I have a bit of experience in RF and think I could probably contribute to integrating this into this project, or should I go with the standard SX12XX modules which have been embraced by the OSS community? My plan is to (attempt to) contribute meaningfully with whichever board I ultimately decide. (Unfortunately I'm a dumbass who happens to have an internet connection so no guarantees)

jgromes, 2023-01-13:
RadioLib already supports ASR650x series in the CubeCell modules made by Heltec: https://heltec.org/project/htcc-ab01-v2/
The biggest hurdle to overcome there was their ... let's say questionable Arduino core.

That same Arduino core (https://github.com/HelTecAutomation/CubeCell-Arduino) also supports the ASR660x series, so it should already work - however, this only applies to running the code directly on the SoC.

For the board you linked, I'm not sure if that is an option since it seems the code on the SoC is fixed and you can only feed it AT commands from a higher system (such as a PC or a different microcontroller via the UART interface). RadioLib doesn't support that mode of operation, since the AT interfaces are typically quite limited in functionality compared to direct SPI access.

However, if you don't require ASR660x, you could use the Heltec board linked above. It has a different SoC (ASR6501), but it seems to be working like any other Arduino-type devboard. I have no personal experience with either option.

jgromes, 2023-01-27:
Closed due to inactivity.

closed 2023-01-27 by jgromes