jgromes/RadioLib #664: Support of chips connected via UART

GitYaSome, 2023-01-17:
There is a thread going on in the Meshtastic Discord where a bunch of people want to connect devices via UART / RS485 / Serial in general, and the devs there mentioned that they can't do it because they use RadioLib and RadioLib does not support it.  Do you have that on any type of timeline?  Can you let me know one way or the other?  Thanks!

jgromes, 2023-01-17:
The Meshtastic devs aren't wrong, RadioLib does not support modules with Serial interface. Support for these kinds of modules came up a couple times (e.g. #467 or #565).

The problem is that there are several manufacturers (Ebyte, Dorji etc.) with different interfaces for different modules, so it would be quite resource-intensive to support everything. In addition, it is not guaranteed the Serial interface will support everything that RadioLib needs to fully function (i.e. access to all registers). Not mentioning all the hacks required to support ham radio modes...

The only "simple" solution I can think of is to create some sort of an interface/template "module" that would have Serial instead of SPI. It could inherit all methods of `PhysicalLayer` to make it compatible with all SPI-based modules, however, it would be left up to the higher layer to provide the implementations. So basically, implementation of the Serial interface would be outside of RadioLib scope - RadioLib would only define which methods must be implemented. That doesn't really solve the issue though, only shifts it one layer higher :P

cc @caveman99 @geeksville

jgromes, 2023-01-17:
Good analogy :) Time is definitely the limiting factor on these projects.

@GitYaSome is there some "killer feature" that these modules provide, that cannot be found on the SPI-based ones? If so, we can discuss further. If not, then I remain unconvinced it's worth supporting these.

geeksville, 2023-01-17:
yep  - I'm not programming much lately (though fortunately we have a large group of friendly meshtastic devs) but I agree.  These "internally spi based radios that add some closed source firmware, a  UART and some AT commandish serial interface" don't seem well suited to apps that are optimized for lora.  Way better to buy a different module IMO.

I think their main market is for people who want to do 1-1 "I'd like to drop in a serial tunnel between two nodes".

jgromes, 2023-01-29:
Closed due to inactivity - perhaps yet another reason to just not bother with UART devices.

MisterRager, 2023-02-09:
I'd chatted about this - it'll take writing one's own firmware for the remote devices and implementing a different [RadioInterface](https://github.com/meshtastic/firmware/blob/master/src/mesh/RadioInterface.h). RadioLib isn't really the right place to put this kind of functionality.