jgromes/RadioLib #665: Pager receive example error for speed = 512

brooksbUWO, 2023-01-17:
In the pager receive example,  pager.startReceive returns a status_code = -101 (RADIOLIB_ERR_INVALID_BIT_RATE) when using 512 for speed in pager.begin. 

Is it possible to change https://github.com/jgromes/RadioLib/blob/7a25b27c3183ffe3627c5c221243c378d8951da7/src/modules/SX127x/SX127x.cpp#L829
to `RADIOLIB_CHECK_RANGE(br, 0.512, 300.0, RADIOLIB_ERR_INVALID_BIT_RATE);`?

<br>
<br>
Just for reference, here's the code on the TTGO-LoRa32-V2.1 T3_V1.6.1.

```
#include <RadioLib.h>

// board = ttgo-lora32-v21
#define LORA_SCK        5
#define LORA_MISO       19
#define LORA_MOSI       27
#define LORA_SS         18
#define LORA_DIO0       26
#define LORA_DIO1       33
#define LORA_DIO2       32
#define LORA_RST        23

SX1276 radio = new Module(LORA_CS, LORA_IRQ, LORA_RST, LORA_DIO1);
PagerClient pager(&radio);
String pagerMsg = "";	  // Pager recieved data
int pagerAdr = 800828;    // Pager address

void setup() {
  Serial.begin(115200);

  // Initialize SX1276
  Serial.print(F("[SX1276] Initializing ... "));
  int state = radio.beginFSK();

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // Initalize Pager client
  Serial.print(F("[Pager] Initializing ... "));
  state = pager.begin(462.45, 512, true);	// 462.45 MHz, 512 bps, invert
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // start receiving POCSAG messages
  Serial.print(F("[Pager] Starting to listen ... "));
  state = pager.startReceive(LORA_DIO2, pagerAdr, 0);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

}

void loop() {
  // the number of batches to wait for
  // 2 batches will usually be enough to fit short and medium messages
  if (pager.available() >= 2) {
    Serial.print(F("[Pager] Received pager data, decoding ... "));
    int state = pager.readData(pagerMsg);

    if (state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));

      // print the received data
      Serial.print(F("[Pager] Data:\t"));
      Serial.println(str);

    } else {
      // some error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }
  }
}
```

jgromes, 2023-01-17:
The slowest data rate SX127x supports is 1200 bps, so no, this cannot be changed - it's a hardware limitation of the module.

brooksbUWO, 2023-01-17:
> The slowest data rate SX127x supports is 1200 bps, so no, this cannot be changed - it's a hardware limitation of the module.

The datasheet only has a few examples of "Classical modem baud rates (multiples of 1.2kbps" and it does not show 512 bps in their example. If we can set RegBitRateMsb=0xD0 and RegBitRateLsb=0x56, then 512 bps will work.

[image]

jgromes, 2023-01-18:
There is a section in the datasheet that does actually specify the support bit rate range - in Table 7, minimum bit rate is 1.2 kbps.

However, I tested the 512 bps POCSAG and somewhat surprisingly, it does work and decodes in PDW. I usually use the parameter ranges from the datasheet, but in this case, it seems reasonable to make an exception. I lowered the minimum data rate for SX127x and RF69 (since it has the same FSK modem) to 500 bps, so now it should work without issues.

Thanks for pointing this out!

closed 2023-01-18 by jgromes