jgromes/RadioLib #671: Missing SubGhz.h

ZaynHaque, 2023-01-23:
Hello, 

When I try to compile the STM32WLx_Transmit example I run into this error

`c:\Users\zayn\Documents\Arduino\libraries\RadioLib-master\src\modules\SX126x\STM32WLx.cpp:12:10: fatal error: SubGhz.h: No such file or directory
   12 | #include <SubGhz.h>
      |          ^~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1`

Where can I find this file?

Using Arduino IDE 2.0.3, running latest version of radiolib

jgromes, 2023-01-23:
It seems very likely you are attempting to build the STM32WLx example on a platform that is ... well, not STM32WLx. Hard to guess though, since you have not provided the information requested by the issue template.

ZaynHaque, 2023-01-23:
I am using the Cicerone by Move-x that uses STM32WLEx. 

Link to Arduino core: https://github.com/Move-X/Move-Xduino

Using the following sketch 

```/*
   RadioLib STM32WLx Transmit Example

   This example transmits packets using STM32WL MCU with integrated
   (SX126x) LoRa radio.

   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)
   
   This example assumes Nucleo WL55JC1 is used. For other Nucleo boards
   or standalone STM32WL, some configuration such as TCXO voltage and
   RF switch control may have to be adjusted.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// no need to configure pins, signals are routed to the radio internally
STM32WLx radio = new STM32WLx_Module();

// set RF switch configuration for Nucleo WL55JC1
// NOTE: other boards may be different!
static const RADIOLIB_PIN_TYPE rfswitch_pins[] =
                         {PC3,  PC4,  PC5};
static const Module::RfSwitchMode_t rfswitch_table[] = {
  {STM32WLx::MODE_IDLE,  {LOW,  LOW,  LOW}},
  {STM32WLx::MODE_RX,    {HIGH, HIGH, LOW}},
  {STM32WLx::MODE_TX_LP, {HIGH, HIGH, HIGH}},
  {STM32WLx::MODE_TX_HP, {HIGH, LOW,  HIGH}},
  STM32WLx::END_OF_MODE_TABLE,
};

void setup() {
  Serial.begin(9600);

  // set RF switch control configuration
  // this has to be done prior to calling begin()
  radio.setRfSwitchTable(rfswitch_pins, rfswitch_table);

  // initialize STM32WL with default settings, except frequency
  Serial.print(F("[STM32WL] Initializing ... "));
  int state = radio.begin(868.0);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set appropriate TCXO voltage for Nucleo WL55JC1
  state = radio.setTCXO(1.7);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[STM32WL] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example STM32WLx_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[STM32WL] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

jgromes, 2023-01-23:
My guess is that the Move-X core is using an outdated STM32Duino, without support for the built-in SX126x module, as that was added quite recently (i.e., mid-December last year).

You're going to have to take this to the maintainers of the Move-X core. I'm not sure why you would use that instead of the STM32 core directly, it doesn't seem to be doing anything extra over STM32Duino.

closed 2023-01-23 by jgromes