jgromes/RadioLib #679: Add support for LR1110 / LR1120 radios

caveman99, 2023-02-09:
Semtech introduced a new range of wideband radios. The LR1110 is meant for Sub-GHz bands while the LR1120 is a full band chip ranging up to 2.4 GHz. In addition the chips can capture raw GPS data for post processing.

I have a LR1120 dev board here and will be looking into basic support next week, jst opening the issue for others to chime in.

jgromes, 2023-02-09:
Thanks for the heads-up - those are some really interesting chips, especially the LR1120. I will have to see if I can get my hands on a development kit (Mouser seems unwilling to sell me one ...).

On the first glance, it loooks like the SPI interface is the same as for the SX126x/8x series, so it should be pretty straight-forward to get thee up and running.

jgromes, 2023-02-19:
@caveman99 FYI in recent commits, I moved parts of the SX126x/8x SPI interface the the Module class, so that could be useful to you should you decide to add the support.

jgamage91, 2023-08-28:
Also, some helpful src is available here https://github.com/Lora-net/SWDR001/tree/master

bradjc, 2023-11-30:
> On the first glance, it loooks like the SPI interface is the same as for the SX126x/8x series, so it should be pretty straight-forward to get thee up and running.

I just want to note that the LR1110 uses an entirely different SPI interface from the SX12xx chips. The [user manual](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/3n000000vBxN/WCHgxGocDfTi4znxyeqdTUXzHRbb1O0WO9RHIZ1MGng) documents the interface.

jgromes, 2023-12-01:
@caveman99 I recently got my hands on LR1110 dev board, did you start working on this at some point?

caveman99, 2023-12-01:
@jgromes ibdidbstart with the wio-wm1110 dev kit but didn't get to the Radio part. The bootloader and base Firmware is almost working ;-)

jgromes, 2023-12-29:
@caveman99 I had a bit of time to work on this over the holidays, I have a basic SPI interface to the LR1110 and can read the version info:

```
11:42:59.592 -> [LR1110] Initializing ... 
11:42:59.592 -> RadioLib Debug Info
11:42:59.592 -> Version:  6.3.0.0
11:42:59.592 -> Platform: Adafruit nRF52
11:42:59.592 -> Compiled: Dec 29 2023 10:06:54
11:42:59.592 -> 
11:42:59.776 -> CMDW	1	1	
11:42:59.776 -> SI			
11:42:59.776 -> SO	5	13	
11:42:59.776 -> CMDR	
11:42:59.776 -> SI	0	0	0	0	0	
11:42:59.776 -> SO	7	22	1	3	7	
11:42:59.776 -> Found LR11x0: RADIOLIB_LR11X0_CMD_GET_VERSION = 0x01
11:42:59.776 -> M	LR11x0
```

So I guess now it's time to start implement all the commands.

gc9n, 2024-03-20:
[image]

LR1121 it's ready to rock but I cannot talk T her. Any news on the support

jgromes, 2024-03-20:
@gc9n I got a little bit stuck with my hardware (and somewhat sidetracked with other stuff). I did get the driver to transmit LoRa, but not to receive yet. So still in progress and not ready to be shared yet ;)

gc9n, 2024-03-20:
Perfect, I guess I cant use other driver only for recieving rssi noise , ( my current project is to receive rssi noise of the current frequency

gc9n, 2024-03-21:
Hey @caveman99   did you manage to make your board work?
i mean i dont want to mess with SEMTECH github examples etc , is a nightmare , i need a simple Arduino project , i need a library to talk into LR1121  , my board is connected to a ESP32 , and is ready ,

gc9n, 2024-03-21:
[image]
[image]

gc9n, 2024-03-21:
> @gc9n I got a little bit stuck with my hardware (and somewhat sidetracked with other stuff). I did get the driver to transmit LoRa, but not to receive yet. So still in progress and not ready to be shared yet ;)

Thanks I am looking forward for you to release something, remember , that there is no library out there for this chip

jgromes, 2024-04-14:
More than a year later, but basic LR11x0 support has been added - see https://github.com/jgromes/RadioLib/tree/master/examples/LR11x0

Currently, LoRa transmit and receive is implemented in both blocking as well as interrupt mode. I would also like to implement GFSK support before closing this issue.

Additional features of the LR11x0 like the WiFi and GNSS scanning will be added after that, as I don't yet have a clear idea of what the public API should look like.

gc9n, 2024-04-14:
FSK is very very important!
Good job

jgromes, 2024-04-20:
In the last couple of commits support for GFSK and LoRa CAD was added. With that I think the basic support for LR11x0 can be considered complete. As mentioned above, support for the features specific to LR11x0 will be added gradually, though hopefully before the next update (6.6.0).

closed 2024-04-20 by jgromes

Tipetofficial, 2024-07-24:
@gc9n Were you able to get the Semtech LR1121 Devkit running and can share some code?