jgromes/RadioLib #685: Dorji RDF1268T Radio lib examples on ESP32 not working.

Rustie0125, 2023-03-01:
Hey Strange error. 

This worked before but its been a few weeks and updated Radio lib atleast twice since then, the simple receive and transmitt example schetches return -2 or -706 error codes. 

This module has an external SW pin on the module. and I see the library allows for external RX or TX switch, this is unconnected but i dont think it has anything todo with error codes returned.

jgromes, 2023-03-01:
As per the "Module not working" issue template - please show at the very least the [debug and verbose mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

Rustie0125, 2023-03-01:
> As per the "Module not working" issue template - please show at the very least the [debug and verbose mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

DId think it was needed for simple example but this is result trying now.

Rustie0125, 2023-03-01:
**Compile log**
`C:\Users\J's PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\J's PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\J's PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Sketch uses 237662 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13952 bytes (4%) of dynamic memory, leaving 313728 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM6
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:5a:1b:60:bd:30
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 1638.4 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.2 seconds (effective 857.7 kbit/s)...
Hash of data verified.
Compressed 246752 bytes to 117204...
Writing at 0x00010000... (12 %)
Writing at 0x00014000... (25 %)
Writing at 0x00018000... (37 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (62 %)
Writing at 0x00024000... (75 %)
Writing at 0x00028000... (87 %)
Writing at 0x0002c000... (100 %)
Wrote 246752 bytes (117204 compressed) at 0x00010000 in 2.6 seconds (effective 763.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 346.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`
Output 
`15:01:46.702 -> 	 B'I�x��	���h�[SX1262] Initializing ... failed, code -2
15:02:00.108 -> ets Jul 29 2019 12:21:46
15:02:00.154 -> 
15:02:00.154 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:02:00.154 -> configsip: 0, SPIWP:0xee
15:02:00.154 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:02:00.154 -> mode:DIO, clock div:1
15:02:00.154 -> load:0x3fff0018,len:4
15:02:00.154 -> load:0x3fff001c,len:1216
15:02:00.154 -> ho 0 tail 12 room 4
15:02:00.154 -> load:0x40078000,len:10944
15:02:00.154 -> load:0x40080400,len:6388
15:02:00.154 -> entry 0x400806b4
15:02:00.251 -> �.r����f��p�1�e��[SX1262] Initializing ... failed, code -2
15:02:31.775 -> '%!$]��QP���rH��[SX1262] Initializing ... failed, code -2`

Rustie0125, 2023-03-01:
Will load the library debug log now

Rustie0125, 2023-03-01:
Weird, with debug enabled I get compilation errors
`In file included from C:\Users\J's PC\AppData\Local\Temp\.arduinoIDE-unsaved202321-26020-ovo7m.eko6xf\SX126x_Channel_Activity_Detection\SX126x_Channel_Activity_Detection.ino:19:0:
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:53:61: note: #pragma message: RADIOLIB_PLATFORM="ESP32"
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_PLATFORM))
                                                             ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:54:66: note: #pragma message: RADIOLIB_VERSION_MAJOR=(0x05)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_MAJOR))
                                                                  ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:55:66: note: #pragma message: RADIOLIB_VERSION_MINOR=(0x06)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_MINOR))
                                                                  ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:56:66: note: #pragma message: RADIOLIB_VERSION_PATCH=(0x00)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_PATCH))
                                                                  ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:57:66: note: #pragma message: RADIOLIB_VERSION_EXTRA=(0x00)
   #pragma message(RADIOLIB_VAR_NAME_VALUE(RADIOLIB_VERSION_EXTRA))
                                                                  ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src\Module.cpp: In static member function 'static void Module::hexdump(uint8_t*, size_t, uint32_t, uint8_t, bool)':
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src\Module.cpp:720:37: error: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Werror=format=]
     sprintf(str, "%07lx  ", i+offset);
                                     ^
c:\Users\J's PC\Documents\Arduino\libraries\RadioLib-master\src\Module.cpp:720:37: error: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Werror=format=]
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1`

Rustie0125, 2023-03-01:
Im confused.. deleted the latest library now, and installed a build from 2023/02/07 and that builds and loads fine, the latest library gives the errors above. will run debugger now and see verbose output but it will be on old library.

Rustie0125, 2023-03-01:
**Other common output **
`15:40:48.912 -> CMD	80	
15:40:48.912 -> DATW	0	FF	
15:40:48.912 -> 
15:40:48.912 -> CMD	80	
15:40:48.944 -> DATW	0	FF	
15:40:48.944 -> 
15:40:48.944 -> CMD	80	
15:40:48.944 -> DATW	0	FF	
15:40:48.976 -> 
15:40:48.976 -> CMD	80	
15:40:48.976 -> DATW	0	FF	
15:40:48.976 -> 
15:40:49.023 -> failed, code -2`

GUVWAF, 2023-03-01:
Your module seems to be an SX1268, so probably you would need to select that instead of SX1262. The error code -2 means `RADIOLIB_ERR_CHIP_NOT_FOUND`. This check was added recently. I am curious whether your chip SX1268 correctly reports it's an SX1268, as in #683, my supposed to be SX1262 module is detected as a SX1261.

I didn't have problems enabling debug mode on the latest library, seems compiler dependent. Probably removing the `-Werror` flag if it was set or adding `Wno-error` 'fixes' the build.

Rustie0125, 2023-03-01:
> Your module seems to be an SX1268, so probably you would need to select that instead of SX1262. The error code -2 means `RADIOLIB_ERR_CHIP_NOT_FOUND`. This check was added recently. I am curious [...]

Ya I saw your issue, and seems similar. am I suppose to set chip manually or is this autodetected ? Im coming over from the TinyGS community and this issues seems to be preset there aswell. none of the sx126 varient latest builds (atleast manually compiled wants to work) error code 2 the whole time.

jgromes, 2023-03-01:
@Rustie0125 slow down please ... There was a warning that got converted to error by the ESP32 compiler settings. It was fixed in the latest commit, so feel free to try it. There's also not much reason copying the compilation output, I'm interested in the Serial output when running with both debug and verbose enabled - this will show all the SPI transcations and will provide hints as to what exactly is going on.

> Did not think it was needed for simple example but this is result trying again now.

Had you followed the template, you could have had the solution already ;)

Also, as @GUVWAF suggested, it seems likely the module type you have in code doesn't match what you have in reality - but we'll see once you post the serial output.

Rustie0125, 2023-03-01:
``

> @Rustie0125 slow down please ... There was a warning that got converted to error by the ESP32 compiler settings. It was fixed in the latest commit, so feel free to try it. There's also not much [...]

Okay new version compiles fine now. here is the serial output, 

Here is the serial output. 

`16:07:00.339 -> DATW	0	FF	
16:07:00.371 -> 
16:07:00.371 -> CMD	80	
16:07:00.371 -> DATW	0	FF	
16:07:00.371 -> 
16:07:00.371 -> CMD	80	
16:07:00.403 -> DATW	0	FF	
16:07:00.403 -> 
16:07:00.403 -> CMD	80	
16:07:00.403 -> DATW	0	FF	
16:07:00.435 -> 
16:07:00.435 -> CMD	80	
16:07:00.435 -> DATW	0	FF	
16:07:00.467 -> 
16:07:00.467 -> CMD	80	
16:07:00.467 -> DATW	0	FF	
16:07:00.467 -> 
16:07:00.467 -> CMD	80	
16:07:00.499 -> DATW	0	FF	
16:07:00.499 -> 
16:07:00.499 -> CMD	1D	3	20	
16:07:00.531 -> DATR	0	FF	0	0	0	0	0	0	0	0	0	E1	0	19	0	D	0	80	0	20	0	1F	0	FB	0	3F	0	78	0	FE	0	FB	0	3F	
16:07:00.595 -> 
16:07:00.595 -> SX126x not found! (10 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
16:07:00.691 -> 0000320 00 00 00 00 e1 19 0d 80 20 1f fb 3f 78 fe fb 3f | ........ ..?x..?
16:07:00.755 -> 
16:07:00.755 -> No SX126x found!
16:07:00.793 -> failed, code -2`

jgromes, 2023-03-01:
It seems that the module is consistently returning 0xFF, which usually indicates bad SPI wiring and/or incorrect pinout. So I would say the -2 error code (or even -705 `RADIOLIB_ERR_SPI_CMD_TIMEOUT`) is a pretty accurate result.

> Lol I would hope if they laser engrave 1268T on the module that its indeed a 1268 chip.

You think if Dorji engraves "1268T" on the shield of the module, then that guarantees what is under the shield? Though I happen to have a DRF1268T as well and that reports as SX1268, but maybe I just got lucky ;)

Rustie0125, 2023-03-01:
Found it! Holly crap. 

Iv been using pin 17 for NSS but seems like you cannot randomly assign it, put it on Pin 5 as per defualt and its working. 

@GUVWAF  I think this is the line you are interested in. 

`
16:44:23.184 -> Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
16:44:23.249 -> 0000320 53 58 31 32 36 38 20 56 32 46 20 32 46 30 32 00 | SX1268 V2F 2F02.
`

Thank you! Sometimes you just need know absolute thats its wiring vs software this helped me take another look!

jgromes, 2023-03-01:
@Rustie0125 glad you were able to get it working!

closed 2023-03-01 by jgromes