jgromes/RadioLib #689: [SX1262] Fails RADIOLIB_ERR_SPI_CMD_FAILED on setup.

OK9UWU, 2023-03-04:
I am using RadioLib 5.6.0, latest Platformio, genuine Arduino Nano board (new bootloader) and a 2W SX1262 LoRa module from China.
From the debug transaction it looks like RadioLib can see the board, identify the chip, issue commands etc. but fails after a second or so.

I am using a snippet of example code "SX126x_Settings.ino" with only the most required code left in it, but at some point the `.begin()` function just fails with error code -707.

main.cpp

`#include <Arduino.h>
#include <RadioLib.h>

SX1262 radio1 = new Module(10, 2, 3, 9);

void setup() {
  Serial.begin(9600);

  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio1.begin();
    if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}
void loop() {
}`

Debug printout

`[SX1262] Initializing ... M        SX126x
CMD     80
DATW    0       A2

CMD     80
DATW    0       A2

CMD     8F
DATW    0       A2      0       A2

CMD     8A
DATW    1       A2

CMD     93
DATW    20      A2

CMD     88
DATW    3       A2      16      A2      A       A2      0       A2      0       A2      0       A2      0       A2

CMD     2
DATW    43      A2      FF      A2

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2

CMD     89
DATW    7F      A2

CMD     80
DATW    0       A2

CMD     17
DATR    0       A2      0       0       0       0

CMD     97
DATW    0       A2      0       A2      1       A2      40      A2

CMD     11
DATR    0       A2      0       1

Symbol length: 1.02 ms
CMD     8B
DATW    9       A2      6       A2      3       A2      0       A2

CMD     11
DATR    0       A2      0       1

CMD     D       7       40
DATW    14      A2      24      A2

CMD     11
DATR    0       A2      0       1

CMD     1D      7       36
DATR    0       A2      0       D

CMD     D       7       36
DATW    D       A2

CMD     8C
DATW    0       A2      8       A2      0       A2      FF      A2      1       A2      0       A2

CMD     D       8       E7
DATW    18      A2

CMD     9D
DATW    1       A2

CMD     96
DATW    1       A2

CMD     11
DATR    0       A2      0       1

Symbol length: 1.02 ms
CMD     8B
DATW    9       A2      6       A2      3       A2      0       A2

CMD     11
DATR    0       A2      0       1

Symbol length: 4.10 ms
CMD     8B
DATW    9       A2      4       A2      3       A2      0       A2

CMD     98
DATW    6B      A2      6F      A2

CMD     86
DATW    1B      AA      20      D6      0       8       0       F7

failed, code -707`

Thx upfront!

jgromes, 2023-03-04:
Looks like it fails just after command 0x98, `RADIOLIB_SX126X_CMD_CALIBRATE_IMAGE` - usually that would mean either incorrect frequency, or TCXO vs XTAL issues.

The default frequency is 434 MHz, which should be fine for both SX1262/21 and SX1268, so I guess that's not the problem. The library also assumes TCXO is used by default, maybe your module doesn't have one?

OK9UWU, 2023-03-04:
The module according to the datasheet has a TCXO, also confirmed by looking under the shield.
Module in question: <https://www.nicerf.com/products/detail/sx1262-lora-module-lora1262f30.html>

jgromes, 2023-03-04:
It could be that the TCXO expects a different reference voltage - it can be set by one of the arguments of `begin` and is 1.6V by default. I can't get to the examples on their site (because it's password-protected for some reason), but maybe the correct voltage will be in there?

OK9UWU, 2023-03-04:
Got my hand on an older version of what seems to be the same file (apparently leaked, whoops),  the internal TCXO seems to run at 3.3V, cant confirm or deny, haven't got it to run yet.

Internally, hardware seems weird, the TCXO has grounds diagonally and DIO3 seems like it wasnt even hooked up!?!
Anyway, added a blob of solder where i suspect usually would be a 0ohm resistor, still to no avail.

Gonna discuss it with someone who knows the hardware a bit more so he can confirm or deny my suspicion, perhaps leave this issue open until then. I guess this should be a red flag not to buy their modules :/

jgromes, 2023-03-04:
Thanks, I will keep this open - let me know!

jgromes, 2023-03-05:
@OK9UWU FYI I added a bit more verbose checking of the calibration result in recent commits, so that might help.

GUVWAF, 2023-03-05:
@jgromes I’m sorry to interrupt here, but might it be that for module with a TCXO the calibration result will return with error `RADIOLIB_SX126X_XOSC_START_ERR` because `setTCXO()` is only called after `config()`? I’m experiencing this now:

> CMD	89	
> DATW	7F	A2	
> 
> CMD	C0	
> DATR	0	2A	0	0	
> 
> CMD	80	
> DATW	0	AA	
> 
> CMD	17	
> DATR	0	A2	0	0	0	20	
> 
> CMD	C0	
> DATR	0	22	0	22	
> 
> Calibration failed, device errors: 0x20

Due to this new check, initialization fails.
I see that this error is actually cleared in `setTCXO()`.

jgromes, 2023-03-05:
@GUVWAF Ah, you're correct - that's a mistake on my part. Serves me right for not testing these changes with both module types ...

Come to think of it, it's really weird to first do config including calibration and only then select the crystal type. Swapping this order within the begin method fixes this problem.

jgromes, 2023-03-30:
@OK9UWU any updates on this?

jgromes, 2023-04-10:
Seeing as there is no changes I'm closing this for now. Definitely looks more like an issue with particular hardware than anything in the library.

closed 2023-04-10 by jgromes

Saba-abiri, 2025-03-04:
Hi [OK9UWU](https://github.com/OK9UWU)

I had this issue too and I fixed it!
first of all before any command do it :

set NRESET_LORA to Low
wait for about 200 milli seconds
set NRESET_LORA to High
wait for about 500 milli seconds

then send this command

// SetDIO3asTCXOCtrl SPI Transaction
cmnd[0] = 0x97; // 0x97 is SetDIO3asTCXOCtrl
cmnd[1] = 0x00; // DIO3 outputs 1.6 V to supply the TCXO
cmnd[2] = 0x00;
cmnd[3] = 0x00;

let me know if you fix it 👍

(base on my repository you can see more setting : https://github.com/Saba-abiri/LORA1262/blob/main/LORA1262/Libs/Src/SX1262.c)

jgromes, 2025-03-04:
@Saba-abiri this thread is nearly two years old at this point, and there has been a lot of development in the meantime. There is no need to hack around with pulling reset pin and sending SPI transactions, as simply setting the appropriate TCXO reference voltage in the `begin` method or using `radio.XTAL=true;` if you don'. have a TCXO will work, as described in many places, for example [here](https://github.com/jgromes/RadioLib/wiki/Troubleshooting-Guide#sx126xsx128x-incorrect-oscillator).

Saba-abiri, 2025-03-05:
thank you [jgromes](https://github.com/jgromes) for kindly describe this for me 🌸