jgromes/RadioLib #690: PlatformIO Library Dependency Finder problem: SubGhz.h in generic board def

omegat, 2023-03-04:
Hi

when compiling RadioLib 5.6.0 with my *very* generic board using stm32duino I get problems with SubGhz.h, especially with the (obviously) not defined SubGhz SPI peripheral used in these STM32WL-Chips including a radio:
```
[.........]
Compiling .pio\build\windows\liba3d\RadioLib\modules\SX126x\STM32WLx.cpp.o
In file included from C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:15:
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.h:22:4: error: #error "This board does not seem to have a SubGhz radio module, cannot use this library."
   22 |   #error "This board does not seem to have a SubGhz radio module, cannot use this library."
      |    ^~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.h:98:5: error: 'SUBGHZSPIClass' does not name a type
   98 |     SUBGHZSPIClass SPI;
      |     ^~~~~~~~~~~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::attachInterrupt(SubGhzClass::callback_function_t)':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:36:28: error: 'SUBGHZ_Radio_IRQn' was not declared in this scope; did you mean 'SUBGHZ_Radio_IRQHandler'?
   36 |   HAL_NVIC_ClearPendingIRQ(SUBGHZ_Radio_IRQn);
      |                            ^~~~~~~~~~~~~~~~~
      |                            SUBGHZ_Radio_IRQHandler
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::enableInterrupt()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:53:22: error: 'SUBGHZ_Radio_IRQn' was not declared in this scope; did you mean 'SUBGHZ_Radio_IRQHandler'?
   53 |   HAL_NVIC_EnableIRQ(SUBGHZ_Radio_IRQn);
      |                      ^~~~~~~~~~~~~~~~~
      |                      SUBGHZ_Radio_IRQHandler
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::disableInterrupt()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:58:23: error: 'SUBGHZ_Radio_IRQn' was not declared in this scope; did you mean 'SUBGHZ_Radio_IRQHandler'?
   58 |   HAL_NVIC_DisableIRQ(SUBGHZ_Radio_IRQn);
      |                       ^~~~~~~~~~~~~~~~~
      |                       SUBGHZ_Radio_IRQHandler
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::clearPendingInterrupt()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:63:28: error: 'SUBGHZ_Radio_IRQn' was not declared in this scope; did you mean 'SUBGHZ_Radio_IRQHandler'?
   63 |   HAL_NVIC_ClearPendingIRQ(SUBGHZ_Radio_IRQn);
      |                            ^~~~~~~~~~~~~~~~~
      |                            SUBGHZ_Radio_IRQHandler
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'bool SubGhzClass::isInterruptPending()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:68:33: error: 'SUBGHZ_Radio_IRQn' was not declared in this scope; did you mean 'SUBGHZ_Radio_IRQHandler'?
   68 |   return HAL_NVIC_GetPendingIRQ(SUBGHZ_Radio_IRQn);
      |                                 ^~~~~~~~~~~~~~~~~
      |                                 SUBGHZ_Radio_IRQHandler
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::setNssActive(bool)':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:74:5: error: 'LL_PWR_SelectSUBGHZSPI_NSS' was not declared in this scope
   74 |     LL_PWR_SelectSUBGHZSPI_NSS();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:76:5: error: 'LL_PWR_UnselectSUBGHZSPI_NSS' was not declared in this scope
   76 |     LL_PWR_UnselectSUBGHZSPI_NSS();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'bool SubGhzClass::isNssActive()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:82:10: error: 'LL_PWR_IsSUBGHZSPI_NSS_Selected' was not declared in this scope
   82 |   return LL_PWR_IsSUBGHZSPI_NSS_Selected();
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'void SubGhzClass::setResetActive(bool)':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:88:5: error: 'LL_RCC_RF_EnableReset' was not declared in this scope
   88 |     LL_RCC_RF_EnableReset();
      |     ^~~~~~~~~~~~~~~~~~~~~
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:90:5: error: 'LL_RCC_RF_DisableReset' was not declared in this scope; did you mean 'LL_RTC_TS_Disable'?
   90 |     LL_RCC_RF_DisableReset();
      |     ^~~~~~~~~~~~~~~~~~~~~~
      |     LL_RTC_TS_Disable
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'bool SubGhzClass::isResetActive()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:96:10: error: 'LL_RCC_RF_IsEnabledReset' was not declared in this scope; did you mean 'LL_TIM_OC_IsEnabledFast'?
   96 |   return LL_RCC_RF_IsEnabledReset();
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
      |          LL_TIM_OC_IsEnabledFast
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp: In member function 'bool SubGhzClass::isBusy()':
C:\Users\tobia\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.cpp:101:11: error: 'LL_PWR_IsActiveFlag_RFBUSYS' was not declared in this scope; did you mean 'LL_RTC_IsActiveFlag_RS'?
  101 |   return (LL_PWR_IsActiveFlag_RFBUSYS());
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |           LL_RTC_IsActiveFlag_RS
*** [.pio\build\windows\lib14a\SubGhz\SubGhz.cpp.o] Error 1
```

The fun part is, the problem occurs even though `RADIOLIB_EXCLUDE_STM32WLX` is defined. The preprocessor shouldn't include the file, and in fact it doesn't. Still, commenting the `#include <SubGhz.h>` in `STM32WLx.cpp` and `STM32WLx.h` fixes all issues and the code compiles fine. So to reiterate (with STM32WLx.cpp (starting with line 9) as an example):
```
#include "STM32WLx.h"
#if !defined(RADIOLIB_EXCLUDE_STM32WLX)
#error SubGhz include called!
#include <SubGhz.h>
[...]
```
Does NOT produce a compiler error but fails with SubGhz.h complaining that `This board does not seem to have a SubGhz radio module` (even though I soldered the SX1276 myself ;) )  -> b/c the SPI peripheral isn't defined [...].

```
#include "STM32WLx.h"
#if !defined(RADIOLIB_EXCLUDE_STM32WLX)
#error SubGhz include called!
//#include <SubGhz.h>
[...]
```
Does NOT produce a compiler error and compiles fine.

So after an afternoon of investigating I found out that PlatformIO scanned the file with the Library Dependency Finder and wants to make the Stm32Duino SubGhz library into a shared 'SubGhz.a' library (or at least tries and fails to do so).

Ways to mitigate:
1. Switch the PlatformIO LDF to 'chain+' mode which lets it also evaluate preprocessor statements in platformio.ini:  `lib_ldf_mode = chain+`
2. Tell the LDF to ignore the SubGhz-lib alltogether in platformio.ini: `lib_ignore = SubGhz`

I can't figure a way to mitigate the issue in source at this point. Posting this as it seems that noone had the issue before....

Cheers and 73

jgromes, 2023-03-04:
According to this (https://docs.platformio.org/en/stable/librarymanager/ldf.html) the LDF should evaluate C preprocessor directives including conditionals - maybe it's struggling with `#if !defined()` as opposed to `#ifndef()` or `#if ... #else`?

Another think to try might be to add a `library.json` file to RadioLib, maybe that would allow to fix the LDF behavior. I don't use PlatformIO, so I don't really know how that would work.

Other than that, I don't think much can be done from RadioLib source, the only other thing I can think of is definining some macros that would make PlatformIO think it can compile SubGHz (e.g. by undefining stuff specific STM32Lxx and defining STM32WLx). Not sure if that would work though.

omegat, 2023-03-04:
I tried both, `#ifndef()` and `#if ... #else` using 'chain' (default) as the LDF mode. Unfortunately no joy.

It could be a workaround to define an instance of this SubGhz SPI, but you would also have to define an IRQ handler and NSS pin (as far as I can see from reading the SubGhz lib). The main problem probably is that PIO compiles SubGhz into a shared lib seperate from RadioLib. So getting the fake hardware instances over when the shared lib is compiled is a whole other can of worms. I totally agree that it is probably not 'fixable' in the RadioLib source. 

I really like PlatformIO though. It reduces all the fiddling with the build chain to an acceptable minimum...

jgromes, 2023-03-04:
> It could be a workaround to define an instance of this SubGhz SPI

That's not really what I meant - the main thing that's failing is the following check in the SubGHz library, as well as some HAL `LL_xxx()` functions.

```
#if !defined(SUBGHZSPI_BASE)
  #error "This board does not seem to have a SubGhz radio module, cannot use this library."
#endif
```

So if you can define `SUBGHZSPI_BASE`, and define all the `LL_xxx` functions as macros that do nothing, the library can be compiled. Hacky, but it should work.

omegat, 2023-03-04:
Hm. You're right, that would work fine. 
But... still you wohld have to somehow define `SUBGHZSPI_BASE` bevore / while the SubGhz shared lib is being compiled. I fear that just defining it in RadioLib somewhere won't cut it, b/c PIO tends to build its shared libraries (especially framework related) first and everything else second (which makes sense considering you want to use all those libs in your code...).
The funcitons probably are no problem (unless they are also somehow out of scope when compiling SubGhz.a (missing board/chip headers??).

jgromes, 2023-03-04:
> But... still you wohld have to somehow define SUBGHZSPI_BASE bevore / while the SubGhz shared lib is being compiled.

Correct, unfortuantely I'm not familiar with PlatformIO's build system, so I don't know how to achieve that.

> The funcitons probably are no problem (unless they are also somehow out of scope when compiling SubGhz.a (missing board/chip headers??).

I would suspect that STM32L HAL/LL is probably not going to contain things like `LL_PWR_SelectSUBGHZSPI_NSS`, since it's specific to STM32WL, hence the need for dummy macros.

All that was a bit of a wild guess - maybe adding `library.json` to better control the build process is the correct way to go?

omegat, 2023-03-04:
> Correct, unfortuantely I'm not familiar with PlatformIO's build system, so I don't know how to achieve that.

You could either define 'global' compiler parameters (i.e. `build_flags = -DSUBGHZSPI_BASE`) in the board .json file or in platformio.ini. But if you have to edit those files you could - more easily - change the LDF mode to chain+ .

> I would suspect that STM32L HAL/LL is probably not going to contain things like `LL_PWR_SelectSUBGHZSPI_NSS`, since it's specific to STM32WL, hence the need for dummy macros.

Exactly. This is why I think the compiler will be annoyed of all those `LL_xxx` functions being not defined ad compile time (even though they have carefully crafted makros somewhere else).

> maybe adding `library.json` to better control the build process is the correct way to go?

HM. That's a bit out of my scope... ;) - I think it can work because you can define some build flags. But I doubt that they are used 'globally' on the whole project. (examples for library.json are here [https://docs.platformio.org/en/latest/manifests/library-json/fields/build/builder.html#examples](https://docs.platformio.org/en/latest/manifests/library-json/fields/build/builder.html#examples). Plus all those workwrounds would have to painlessly go away when the "board does seem to have a SubGhz radio module"... 

It's a bit of a pain really.

jgromes, 2023-03-05:
I was able to reproduce this issue by building on Platformio for Nucleo L452RE. After a little bit of searching, I found out that LDF mode can be specified library-wide by using the `library.json` file. Adding that with LDF mode set to `chain+`, it started correctly compiling without SubGHz.

closed 2023-03-05 by jgromes