jgromes/RadioLib #691: SX1268 failed, code -707 RadioLib SX126x Transmit Example

safe-power, 2023-03-05:
I'm having trouble to initialize new modules. 2 modules SX1268 and 1 module LLCC68 have been tried and give error 707
Other libraries such as 
[Arduino-LoRa-Ra01S](https://github.com/nopnop2002/Arduino-LoRa-Ra01S) and from Stuarts projects give similar init issues. 

I've found the other reports here with error 707 but I do not understand the solutions. Any suggestions what to try next would be great. 

The module is connected and connections have been triple checked as: 
TXEN 5,
 RXEN 4, 
MOSI 11,
 MISO 12,
 SCK 13 
 NSS pin:   10
 DIO1 pin:  2
NRST pin:  3
 BUSY pin:  9

release: 5.6.0
Arduino 1.8.19
SX1268 Ai-thinker Ra-01S module
Arduino Pro-Mini- M 328, 16MHz, running on 3.3V
 The module fails to initialize with   failed, code -707

RadioLib SX126x Transmit Example
https://github.com/jgromes/RadioLib/tree/master/examples/SX126x/SX126x_Transmit

**Sketch that is causing the module fail**
`#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
//SX1262 radio = new Module(10, 2, 3, 9);
SX1268 radio = new Module(10, 2, 3, 9); // changed to SX1268

//TXEN 5, RXEN 4, MOSI 11, MISO 12, SCK 13  are also connected

void setup() {
  Serial.begin(9600);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1268] Init... "));
    //int state = radio.begin(); -tried  failed, code -707

   int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 1.6, false);
    
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }`  ... sketch truncated 

--------------------------------- debug output-------------------

[SX1262] Initializing ... M	SX126x
CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8F	
DATW	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8A	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	93	
DATW	20	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	2	
DATW	43	A2	FF	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	89	
DATW	7F	A2	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	17	
DATR	0	A2	0	0	0	0	

CMD	C0	
DATR	0	22	0	22	

CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

Symbol length: 1.02 ms
CMD	8B	
DATW	9	A2	6	A2	3	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	D	7	40	
DATW	14	A2	24	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	11	
DATR	0	A2	0	1	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	7	36	
DATR	0	A2	0	D	

CMD	C0	
DATR	0	22	0	22	

CMD	D	7	36	
DATW	D	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8C	
DATW	0	A2	8	A2	0	A2	FF	A2	1	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	D	8	E7	
DATW	18	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	9D	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	96	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	98	
DATW	6B	A2	6F	A2	

CMD	C0	
DATR	0	2A	0	0	

failed, code -707

jgromes, 2023-03-05:
Since the initialization failed after image calibration (command 0x98), that would suggest either bad frequency or TCXO vs XTAL issues. 434 MHz should be fine for all SX126x devices, so the next suspect is the crystal. You set your TCXO reference voltage to 1.6V - are you sure that is correct? And does your device really have a TCXO, or just plain XTAL?

safe-power, 2023-03-05:
> And does your device really have a TCXO, or just plain XTAL?  

I'm not sure how to find this out. It's a new Aliexpress module. I will connect the LLCC68 now to another Arduino to see if it fails at the same point as the SX1268. 

The second radio.begin line is copied from the API example.

I would like to try out a few other settings for XTAL. Could you post an example or point me towards example radio.begin statements ?

int state = radio.begin(); -tried  failed, code -707
 int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 1.6, false);  code -707

jgromes, 2023-03-05:
> I'm not sure how to find this out

Step 1 is to check the documentation of the module, if there is any. If not, then trial and error - set the TCXO reference voltage to 0 and reset/power cycle the module.

```
int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 0, false); 
```

Also, `begin()` without arguments expands to `begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 1.6, false)` since those are the default values.

safe-power, 2023-03-05:
Thank you so much. Fixed!
`int state = radio.begin(434.0, 125.0, 9, 7, 0x12, 10, 8, 0, false);`
did the trick. 
Both SX1268 and  LLCC68 
work with this setting.
2 days of my life gone, but you fixed it within 1 hour.

jgromes, 2023-03-05:
Glad you got it working - don't hesitate to ask sooner than after two days ;)

closed 2023-03-05 by jgromes