jgromes/RadioLib #694: [SX1262] Not sending packets after updating to v5.7

timmbogner, 2023-03-07:
**Describe the bug**
Under RadioLib 5.6, the following sketch successfully sends via a Heltec LoRa32 v3 SX1262. It is received successfully via a TTGO OLED LoRa v1 (SX1276) but the RSSI is ~120. When updated to RadioLib 5.7, the sender responds identically, but no packet is received by the TTGO.  I'm using Arduino 2.0.4 in Windows 10.

I found out while making this issue that enabling the RadioLib debug options causes packets to be recieved under RL 5.7. When this occurs the RSSI drops to 30. This is my first time working with the SX1262, do I need to do something with the RF switch?

Sorry if this is something I've overlooked!

**To Reproduce**
This sketch includes all of the settings I use to configure for different LoRa setups, I hope you don't mind. Other than that it is pretty much the Transmit_Interrupt sketch:
``` cpp

#define RADIOLIB_MODULE SX1262  // ESP32 SX1262 (HELTEC)
#define LORA_SS 8
#define LORA_RST 12
#define LORA_DIO 14
#define USE_SX126X

#define FDRS_LORA_TXPWR 17

#define FDRS_LORA_FREQUENCY 915.0
#define FDRS_LORA_BANDWIDTH 125.0
#define FDRS_LORA_SF 7
#define FDRS_LORA_CR 5
#define FDRS_LORA_SYNCWORD 0x12

#include <RadioLib.h>

#ifdef CUSTOM_SPI
#ifdef ESP32
SPIClass SPI1(HSPI);
#endif  // ESP32
RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, -1, SPI1);
#else
RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, -1);
#endif  // CUSTOM_SPI

void setup() {
  Serial.begin(115200);
  //  delay(5000);
  Serial.println("Hello, world!");

#ifdef CUSTOM_SPI
#ifdef ESP32
  SPI1.begin(LORA_SPI_SCK, LORA_SPI_MISO, LORA_SPI_MOSI);
#endif  // ESP32
#ifdef ARDUINO_ARCH_RP2040
  SPI1.setRX(LORA_SPI_MISO);
  SPI1.setTX(LORA_SPI_MOSI);
  SPI1.setSCK(LORA_SPI_SCK);
  SPI1.begin(false);
#endif  //ARDUINO_ARCH_RP2040
#endif  // CUSTOM_SPI

#ifdef USE_SX126X
  int state = radio.begin(FDRS_LORA_FREQUENCY, FDRS_LORA_BANDWIDTH, FDRS_LORA_SF, FDRS_LORA_CR, FDRS_LORA_SYNCWORD, FDRS_LORA_TXPWR, 8, 1.6, false);
#else
  int state = radio.begin(FDRS_LORA_FREQUENCY, FDRS_LORA_BANDWIDTH, FDRS_LORA_SF, FDRS_LORA_CR, FDRS_LORA_SYNCWORD, FDRS_LORA_TXPWR, 8, 0);
#endif

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println("RadioLib initialization successful!");
  } else {
    Serial.println("RadioLib initialization failed, code " + String(state));
  }

}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  int state = radio.startTransmit("Hello World!");

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}

```

jgromes, 2023-03-08:
A couple observations:

1. You're using the `startTransmit` method, which is only the first half of interrupt-driven transmission. You should also call `finishTransmit` after this is done. Or, just use the blocking `transmit` method.
2. `startTransmit` doesn't measure data rate, only blocking transmission does.

> When this occurs the RSSI drops to 30

I'm assuming the actual value is -30 dBm, which is a lot more than the original -120 dBm. So the signal seems to be several orders of magnitude stronger now.

> do I need to do something with the RF switch?

That depends if your module has an RF switch. If not, then you don't need to control it. If it does, and if it connected to DIO2, then the library can take care of it automatically. If the switch has more control pins (e.g. TXEN and RXEN), then you need to use the `setRfSwitchPins` method.

Next, can you also post the received code and the debug mode output?

timmbogner, 2023-03-08:
Sure thing. I added finishTransmit and it doesn't change behavior, and also switched to the blocking transmit to no success. I should mention that the actual problem is happening in my larger FDRS project, and I'm doing my best at re-creating the problem on a small scale with these examples. I also only have one SX1262 to test with, so the configs help me test it against SX127X modules. In this post will show the **sender sketch, its serial output with debugging, and then the receiving code.** 
_When debugging is enabled on the sender, the problem still goes away._ 
Sender:
``` cpp

#define RADIOLIB_MODULE SX1262  // ESP32 SX1262 (HELTEC)
#define LORA_SS 8
#define LORA_RST 12
#define LORA_DIO 14
#define USE_SX126X

#define FDRS_LORA_TXPWR 17

#define FDRS_LORA_FREQUENCY 915.0
#define FDRS_LORA_BANDWIDTH 125.0
#define FDRS_LORA_SF 7
#define FDRS_LORA_CR 5
#define FDRS_LORA_SYNCWORD 0x12
#define RADIOLIB_DEBUG
#define RADIOLIB_VERBOSE

#include <RadioLib.h>

#ifdef CUSTOM_SPI
#ifdef ESP32
SPIClass SPI1(HSPI);
#endif  // ESP32
RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, -1, SPI1);
#else
RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, -1);
#endif  // CUSTOM_SPI

void setup() {
  Serial.begin(115200);
  //  delay(5000);
  Serial.println("Hello, world!");

#ifdef CUSTOM_SPI
#ifdef ESP32
  SPI1.begin(LORA_SPI_SCK, LORA_SPI_MISO, LORA_SPI_MOSI);
#endif  // ESP32
#ifdef ARDUINO_ARCH_RP2040
  SPI1.setRX(LORA_SPI_MISO);
  SPI1.setTX(LORA_SPI_MOSI);
  SPI1.setSCK(LORA_SPI_SCK);
  SPI1.begin(false);
#endif  //ARDUINO_ARCH_RP2040
#endif  // CUSTOM_SPI

#ifdef USE_SX126X
  int state = radio.begin(FDRS_LORA_FREQUENCY, FDRS_LORA_BANDWIDTH, FDRS_LORA_SF, FDRS_LORA_CR, FDRS_LORA_SYNCWORD, FDRS_LORA_TXPWR, 8, 1.6, false);
#else
  int state = radio.begin(FDRS_LORA_FREQUENCY, FDRS_LORA_BANDWIDTH, FDRS_LORA_SF, FDRS_LORA_CR, FDRS_LORA_SYNCWORD, FDRS_LORA_TXPWR, 8, 0);
#endif

  if (state == RADIOLIB_ERR_NONE) {
    Serial.println("RadioLib initialization successful!");
  } else {
    Serial.println("RadioLib initialization failed, code " + String(state));
  }
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));
  int state = radio.transmit("Hello World!");
  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));
    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
[... 463 lines cut ...]
      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1278] RSSI:\t\t"));
      Serial.print(radio.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1278] SNR:\t\t"));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));

    } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("[SX1278] CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("[SX1278] Failed, code "));
      Serial.println(state);
    }

    // put module back to listen mode
    radio.startReceive();
  }
}
```

jgromes, 2023-03-08:
TL;DR: On SX1262, you're missing the BUSY pin.

Long version:

For a while, I was unable to reproduce this. I even pulled out some breadboards to replicate the setup as close as I could, with ESP32/SX1262 transmitter, and ESP32/SX1278 receiver. I was unable to get it to break with my default configuration, so I started replicating yours. Even setting the same frequency etc. had no effect, but then I noticed this line in your examples:

```c++
RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, -1);
```

In all cases, you are setting the last pin of the Module constructor to -1 (BTW the correct RadioLib alias for unconnected pin is `RADIOLIB_NC`). For SX127x, this is fine as the second DIO pin (DIO1 in this case) is optional and only needed for some functionality.

However, in the case of SX126x, the last pin is BUSY, and it's not optional. It lets RadioLib know when the chip is ... well, busy. Since this signal is used to control timing of SPI transactions  as well as calibrations, setting it to -1 (which prevents RadioLib from using it) will introduce timing issues: for example, RadioLib might try to initiate an SPI transaction before the previous one was executed.

However, in debug+verbose, there's the additional overhead of printing everything to Serial. It seems like this slowdown is sufficient for the SPI transactions to get executed.

Why wasn't this broken pre-5.7.0, I'm not sure. There was a slight rework of the SPI for SX126x in 5.7.0, so it is possible there was an unrelated issue that was having a similar effect without debug.

PS: It looks like SX1262 modules will always report as SX1261.

jgromes, 2023-03-08:
No worries, I'm glad you were able to get it working!