jgromes/RadioLib #696: STM32 with RFM95 not compiling

phires, 2023-03-12:
**Describe the bug**
Compiling for STM32 Bluepill with RM95 using Platform IO (VSCode)results in this error: 

`C:\Users\phili\.platformio\packages\framework-arduinoststm32\libraries\SubGhz\src\SubGhz.h:22:4: error: #error "This board does not seem to have a SubGhz radio module, cannot use this library."
   22 |   #error "This board does not seem to have a SubGhz radio module, cannot use this library."`

**To Reproduce**
_platform.ini_
`[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
framework = arduino
upload_protocol = stlink
monitor_port = COM7
lib_deps = 
	jgromes/RadioLib@^5.7.0`

_main.cpp_
`#define LORA_FREQ 868.0
#define LORA_BANDWIDTH 250.0
#define LORA_SPREADING_FACTOR 10
#define LORA_CODING_RATE 6
#define LORA_SYNC_WORD 0x24
#define LORA_OUTPUT_POWER 10
#define LORA_CURRENT_LIMIT 80
#define LORA_PREAMBLE_LENGTH 15
#define LORA_GAIN 0
#define LORA_NSS PA4
#define LORA_DIO0 PA10
#define LORA_RST PA9
RFM95 radio1 = new Module(LORA_NSS, LORA_DIO0, LORA_RST);

void setup() {
	delay(1000);

	initialize_lora();
	// delay to wait for settings
	delay(5000);
}

void initialize_lora() {
	DBG_PRINT(F("Initializing LoRa... "));
	#ifdef RADIO_LIB
	int state = radio1.begin(
		LORA_FREQ, 
		LORA_BANDWIDTH, 
		LORA_SPREADING_FACTOR, 
		LORA_CODING_RATE, 
		LORA_SYNC_WORD, 
		LORA_OUTPUT_POWER, 
		LORA_PREAMBLE_LENGTH,
		LORA_GAIN);

	if (state == RADIOLIB_ERR_NONE) {
		DBG_PRINTLN(F("success!"));
	} else {
		DBG_PRINT(F("failed, code "));
		DBG_PRINTLN(state);
	}
}

`

**Expected behavior**
Compiling succuessful.

**Additional info (please complete):**
 - MCU: Bluepill STM32 F103C8
 - Link to Arduino core: https://github.com/stm32duino/Arduino_Core_STM32
 - Wireless module type: RM95W
 - Arduino IDE version Platform IO 3.0.0 with framework-arduinostm32 @ 4.20400.0
 - Library version 5.7.0

jgromes, 2023-03-12:
Looks like the same issue as was reported in #690. I was able to replicate it using your `.ini` configuration.

As a workaround, you can add `lib_ldf_mode = chain+` to your `.ini` file, however, that's not really fixing the root cause - I was under the impression that having a library.json in the repository would force Platform.IO to apply the settings within that file, but apparently not ... I will have to investigate further.

jgromes, 2023-03-27:
@phires seems like this was an issue in Platformio, as discussed here: https://community.platformio.org/t/library-ldf-mode-ignored-for-lib-deps/32639

It should be fixed now, could you try to clean and rebuild your project?

derekmpeterson, 2023-03-30:
@jgromes FWIW I was able to successfully build for a STM32WLx that does use SubGhz prior to this change, and am now getting this error:

```
12 | #include <SubGhz.h>
      |          ^~~~~~~~~~
compilation terminated.
*** [.pio/build/lora_e5/lib909/RadioLib/modules/SX126x/STM32WLx_Module.cpp.o] Error 1
```

jgromes, 2023-03-31:
@derekmpeterson That's odd. @matthijskooijman any idea why this is the case? Checking the build process, `SUBGHZSPI_BASE` is defined, but despite that PlatformIO doesn't find the SubGhz library.

matthijskooijman, 2023-03-31:
I had a look at this, but I'm quite unsure what happens. I also have little experience with pio, so that's not really helpful.

@derekmpeterson, it seems that you are seeing a different issue than the subject of this issue (you *are* using an STM32WL board, and the issue is about using another STM board). Maybe it would be helpful to create a separate issue for this (since *this* issue seems already fixed, just @phires needs to confirm it). If you create a new issue, could you include the entire error message? Seems you just pasted the context, not the error message itself.

jgromes, 2023-04-02:
Seems the original was resovled, as I can no longer reproduce it. I added a separate issue for the new problem.

closed 2023-04-02 by jgromes