jgromes/RadioLib #699: v5.7 bugs

EliasA97, 2023-03-15:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please use Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jgromes, 2023-03-15:
This is not very helpful.

1. If you have issues with some chips, post the debug output. It is possible they are being misidentified, or they are nto actually what you think they are.

2. What platform are you using? How significant is the increase? Again, you gave me almost no information here ...

EliasA97, 2023-03-15:
Sorry my bad...

I use both arduino uno and mega as transceivers with ebyte E22 400m30s. 

The uno specifically used about 89% of program space. After the update it got up to 99%. That raised a flag. 
When I updated to 5.7 yesterday suddenly the lora wasn't recognised in both boards. That got me thinking it may be a bug  which explains the memory increase(?), so I downgraded to 5.6 and everything continue working. 

Did you changed anything command-wise?

If not I'll post you later a debug output.

Thank's

jgromes, 2023-03-15:
In 5.7 a check was introduced to make sure the module you have connected is actually what it claims to be. This is displayed in the debug output, so please post it.

The report template exists for a reason ;)

EliasA97, 2023-03-16:
Sorry Jan, I'm in a bit of a rush and my mind is slipping away. 

This is the output of your SX126x-Settings example (v5.7)

```
[SX1262] Initializing ... CMD	80	
DATW	0	AA	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	3	20	
DATR	0	A2	0	53	0	58	0	31	0	32	0	36	0	38	0	20	0	56	0	32	0	46	0	20	0	32	0	46	0	30	0	32	0	0	

SX126x not found! (1 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 38 20 56 32 46 20 32 46 30 32 00 | SX1268 V2F 2F02.
Expected string: SX1261
CMD	80	
DATW	0	AA	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	3	20	
DATR	0	A2	0	53	0	58	0	31	0	32	0	36	0	38	0	20	0	56	0	32	0	46	0	20	0	32	0	46	0	30	0	32	0	0	

SX126x not found! (2 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 38 20 56 32 46 20 32 46 30 32 00 | SX1268 V2F 2F02.
Expected string: SX1261
CMD	80	
DATW	0	AA	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	1D	3	20	
DATR	0	A2	0	53	0	58	0	31	0	32	0	36	0	38	0	20	0	56	0	32	0	46	0	20	0	32	0	46	0	30	0	32	0	0	

SX126x not found! (3 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 38 20 56 32 46 20 32 46 30 32 00 | SX1268 V2F 2F02.
Expected string: SX1261
CMD	80	
DATW	0	AA	

```

This is the same example but with v.5.6

```
[SX1262] Initializing ... M	SX126x
CMD	80	
DATW	0	AA	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	80	
DATW	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8F	
DATW	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	8A	
DATW	1	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	93	
DATW	20	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	88	
DATW	3	A2	16	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	C0	
DATR	0	22	0	22	

CMD	2	
DATW	43	A2	FF	A2	

CMD	C0
```	

Note that I only have SX1262 and I didn't erase the second instance.

Now, on arduino uno with v5.6 I get 54% used program space and 50% ram. With v5.7 I get 62% program and 49% ram.
As you can see there's not a problem with my setup because it works fine when using v5.6

jgromes, 2023-03-16:
```
SX126x not found! (5 of 10 tries) RADIOLIB_SX126X_REG_VERSION_STRING:
0000320 53 58 31 32 36 38 20 56 32 46 20 32 46 30 32 00 | SX1268 V2F 2F02.
Expected string: SX1261
```

The module clearly identifies itself as SX1268. The [Ebyte page for E22 400M30S](https://www.ebyte.com/en/product-view-news.html?id=454) also states that there's an SX1268 inside it.

> Note that I only have SX1262

You don't have an SX1262 - you have an SX1268. This check was introduced in 5.7.0, so before that there was nothing stoppping you from using an incorrect module type.

Regarding the increased memory footprint, I will check if there's something that can be done about it.

EliasA97, 2023-03-16:
That's odd. I thought I had a SX1262 version all that time. 
Changing SX1262 to SX1268 seems to fix the problem.

closed 2023-03-16 by EliasA97