jgromes/RadioLib #703: SX126x IRQ handling does not follow the normal operation mode

antirez, 2023-03-18:
Dear owner of the repository,

I'm using this library to interact with an SX1262, and I'm the author of another library that works with SX1276. The chip, by default, is put in continuous receiving mode:

    int16_t startReceive(uint32_t timeout = RADIOLIB_SX126X_RX_TIMEOUT_INF...

When in this mode, if we associate an interrupt to a set of actions, the chip does not need to be put back in receiving mode each time the interrupt is called. The normal operations is a follows:

1. Put the chip in receive mode once, at the start.
2. Use radio.setDio1Action() to register an IRQ handler.
3. When the IRQ handler is called, we check the IRQ state to see what action triggered our interrupt.
4. At this point we need just to clear the IRQ flags, without putting back the radio in receive mode (it already is), and our interrupt will be called again.

With this library, the method to clear the IRQ flags is private, so it looks like that even when the radio is in continuous receive mode, we need to activate every time the mode again, otherwise there is no way our interrupt will be called again. Or I'm missing something? In such case, any help would be appreciated. This is a bit of background about *why* I can't put the radio back in receive mode each time.

1. One reason I don't want to do it, is conceptual. The radio is designed to stay in this mode indefinitely, so no need to write the command again. This creates a small window when the packets are no longer sensed, too.
2. The other reason is very practical: I want the IRQ to be called when the preamble is detected and later when a valid header is detected. If I can't set the IRQ flags, I'll not get the next calls to the IRQ.

Thanks,
Salvatore

jgromes, 2023-03-18:
In the current SX126x driver implementation, the device will be put to standby to read the data (in the `SX126x::readData` method). The reason why it's implemented this way is that SX126x documentation to the best of my knowledge doesn't specify whether you can read a packet while in Rx mode, and what happens if another packet is received while you are reading the previous one. Calling standby before reading ensures this cannot happen, which to me is the safer approach (since I don't know the limitations of the sceond one).

Although this could be investigated - I imagine you could have two devices, one transmitting with high SF and low bandwidth (so that the transmission takes a long time) and one that receives and reads a packet slowly. This could be used to induce those collisions (receiving while reading data).

What is the source of the "normal operations" flow you have described? And does it specify what happens when two apckets collide?

> This creates a small window when the packets are no longer sensed, too

In theory yes, though I don't think in real-world applications, the size of that window is significant.

> I want the IRQ to be called when the preamble is detected and later when a valid header is detected. If I can't set the IRQ flags, I'll not get the next calls to the IRQ

You can pass custom IRQ flags to `startReceive`. Although you still need to have the ability to clear the IRQ flags - for that, I think the `clearIrqStatus` method can be made public.

antirez, 2023-03-18:
Thank you @jgromes, that puts the current operations much better in perspective. I enabled "God mode" (thanks for providing this!) so that I can investigate better if it is totally safe to perform things in different ways. I'll report my findings, and you can later choose depending on the findings if you want me to send a PR, or perform other tests, or just leave things as they are, and in such case I can write documentation (and some example) on how to implement in-app, and in God mode, the "stay in RX" mode. Basically I'm finding the SX1262 to be a less linear and clear chip than the SX1276. For instance, the SX1276 has a status register that can be used much more comfortably to implement "listen before talk" while in RX mode, without resorting to IRQ flags for preamble start, header start, timeouts, ...

Anyway, I really appreciate your reply both for details and timing. Thank you.

jgromes, 2023-03-19:
Thanks for looking into this, I would defnitely like to avoid using godmode in production, so if there's any way to improve the SX1262 driver that doesn't interfere with existing functionality, I will gladly accept a PR for it.

GUVWAF, 2023-03-19:
> Basically I'm finding the SX1262 to be a less linear and clear chip than the SX1276. For instance, the SX1276 has a status register that can be used much more comfortably to implement "listen before [...]

I agree on this. 
Recently I adapted the [“isActivelyReceiving” function](https://github.com/meshtastic/firmware/blob/master/src/mesh/SX126xInterface.cpp#L249) in the Meshtastic firmware for SX126x radios by setting both the `PREAMBLE_DETECTED` and `HEADER_VALID` IRQ flags. When using only `HEADER_VALID`, I encountered in practice that it sometimes doesn’t detect it is actively receiving with a 16-symbol preamble. However, preamble detection is susceptible to false detections, and the IRQ will not be cleared until it receives a real packet. Therefore, I implemented a timer to ignore the `PREAMBLE_DETECTED` flag when we should have received the `HEADER_VALID` IRQ. This works well, but it is only checked when this function is called; it would be better to always clear the flag based on a timer, because now we transmit later than necessary when we determined there was a false detection somewhere before.
It seems that the Signal Detected modem status of SX127x radios automatically gets cleared when the signal is not synchronized (modem in lock) when it should have based on the preamble length. I’m wondering if a similar functionality would be worth implementing for SX126x radios in RadioLib. That would be even more user-friendly than making `clearIrqStatus` public (although I agree that would be nice as well).

antirez, 2023-03-19:
So, a bit of background about what I believe is the source of my findings:

1. While the SX1262 data sheet is definitely not well written, there are many hints about the fact this is the intended more of operations: A) In no place there is ever written that we should go to standby mode to read the data. This would be a crucial information they would not fail to mention. B) There is an explicit "stay in RX mode" in both SX1276 and SX1262. That would be of no use at all if we couldn't read packets while in RX mode. Note that this mode is not just called "infinite timeout", but continuous receiving mode. C) The buffer implementation looks conceived on purpose in order to support this mode of operations: the command GetRxBufferStatus() atomically reads the length *and the base address* of the received packet. So even fi the radio would receive part of the next packet, because the IRQ is extremely slow for any reason (kernel scheduler?), yet we would read the right length and would fetch from the right address. Even if, I believe, right now RadioLib is not using this capability, but will just discard the offset and get the packet len, and then assume that the base of the buffer contains the packet.
2. I tried to read the packet without going into standby mode, and now everything works well: I get the preamble detected, then the valid header (in case is not just a spurious preamble detection -- otherwise I use a timer as 
@GUVWAF mentioned), then if there is some actual data, I read it like thata:

```c
        size_t len = radio.getPacketLength();
        int state = radio.readBuffer(packet,len);
        float rssi = radio.getRSSI();
        int bad_crc = (status & RADIOLIB_SX126X_IRQ_CRC_ERR) != 0;

        packetsQueueAdd(RXQueue,packet,len,rssi,bad_crc);

        /* Reset the SX1262 state to receive the next packet: note that
         * it will stay in RX mode, since we initialized the chip with
         * "infinite" timeout. */
        radio.setBufferBaseAddress();
        radio.clearIrqStatus();
```

Yet I agree that at this point it is useful to understand what happens with a radio continuously transmitting in a low bit rate mode, and introduce some busy look in the IRQ handler to make sure there is a collision. I believe that nothing will happen because the TX buffer continues to be populated *after* the old packet. The problem is only going to happen if we wait a very large amount of time, to the point the other packet is totally received. I believe that since right now RadioLib is not using the base address of the last received packet, that could be obtained via GetRxBufferStatus(), what happens is that we read from offset 0 but read a packet length that is the one of the last packet received.

However, let's consider the current implementation. Even if we switch to standby, this race condition is still there: the IRQ may be slow to put the radio in standby mode, so anyway the next packet will be received and we will be in the exact same condition. What you want, if you want to avoid this problem, is to avoid setting the radio in continuous mode *at all*, and instead just use the single RX way (no 0xffff... timeout when RX mode is enabled). In this way once the packet is received, the radio does immediately return back to standby, avoiding any problem. However the IRQ needs to handle the timeout in this case, and put the radio back in RX mode if the timeout is triggered.

Given that the current situation AFAIK does not fix the race, I propose the following set of changes (but I may change idea after more testing):

1. Don't go to standby when reading packets. It does not improve the current situation: the race is still possible.
2. Read the packet from the right base address, as reported by the atomic last packet length/offset function of the chip.
3. Document that in case the IRQ may be slow, and the user wants to avoid at any cost a read race condition or half-packets (even if we implement "2" the buffer left for the next packet may not be enough), a non continuous mode should be used, but instead the radio should go in standby automatically after each packet is received and put back online when needed.
4. Alternatively to all this, I'm super happy with just using God mode if there would be a way to suppress the warning (an additional define to test, like `RADIOLIB_GOD_MODE_I_KNOW_WHAT_IM_DOING`).

I'm going to test it and report back. Thanks!

jgromes, 2023-03-19:
I'm OK with implementing the first 3 points. Regarding 4th, God mode has an annoying warning on purpose, I don't want people to use this to workaround issues or bad design decisions in the library itself.

Regarding point number 5, I don't think it's all that advantageous to have a separate `startReceiveSingle` method. Instead, I would propose to document the difference in behavior with 0xFFFFFF timeout and actual timeout value.

vtjballeng, 2023-03-26:
This is very exciting work @antirez! We have an application that could benefit from continuous receive mode without a need to go into standby when receiving a message.

jgromes, 2023-03-30:
@antirez any progress on this? If not I can implement your proposal, but I don't want to be redoing it if you're already working on this.

jgromes, 2023-04-10:
Seeing as there was no reaction, I implemeted the proposed changes in the previous commit. The major one is that standby is no longer called prior to reading data, I also added some more explanation of the different Rx modes. Tested on a pair of SX1268 devices and there seem to be no functional change.

I also found that the same change can be done on SX127x.

closed 2023-04-10 by jgromes